esp8266/Arduino #7507: The infrared sensor outputs 1 level change, but the Arduino reads multiple

tanghui0102, 2020-08-06:
Hello everyone, novice ask for advice.
Thinking of the arduino infrared obstacle avoidance module to recognize the white sticker and realize the speed measurement function.
code show as below:
`int LED = 13; // Use the onboard Uno
volatile float n = 0;
float dt;
float Speed;
float begin_time;
const byte interruptPin = 2;  //用2号引脚作为中断触发引脚

void setup() {
  pinMode(LED, OUTPUT);
  pinMode(interruptPin, INPUT);
  attachInterrupt(digitalPinToInterrupt(interruptPin),count_pulse,FALLING);
  begin_time=millis();
  Serial.begin(9600);
}

void count_pulse(){
  Serial.println("I am being called");
  n += 1;
}

void loop() {
  dt=millis()-begin_time;
  /*每隔500ms输出一次转速，并重新开始计算下一个500ms的转速 */
  if(dt > 1000.0 ){
    //每一圈有6个脉冲，下面返回的值为rpm
    //delayMicroseconds(50);
    Serial.print("脉冲个数：");
    Serial.println(n);
    Serial.print("dt: ");
    Serial.println(dt);
    Serial.print("转速：");
    Speed =  (n/6) / (dt/60000);
    Serial.println(Speed);
    n = 0;
    begin_time = millis();
    Serial.flush();  // 确定都已经打印出去

  }
}`

Below is the output：
`0:08:40.398 -> dt: 1001.00
10:08:40.398 -> 转速：0.00
10:08:41.358 -> 脉冲个数：0.00
10:08:41.398 -> dt: 1001.00
10:08:41.398 -> 转速：0.00
10:08:41.518 -> I am being called
10:08:41.518 -> I am being called
10:08:41.558 -> I am being called
10:08:41.558 -> I am being called
10:08:41.598 -> I am being called
10:08:41.638 -> I am being called
10:08:41.678 -> I am being called
10:08:41.678 -> I am being called
10:08:41.718 -> I am being called
10:08:41.718 -> I am being called
10:08:41.758 -> I am being called
10:08:41.758 -> I am being called
10:08:41.877 -> I am being called
10:08:41.917 -> I am being called
10:08:41.917 -> I am being called
10:08:41.957 -> I am being called
10:08:41.957 -> I am being called
10:08:42.198 -> I am being called
10:08:42.198 -> I am being called
10:08:42.198 -> I am being called
10:08:42.238 -> I am being called
10:08:42.238 -> I am being called
10:08:42.358 -> I am being called
10:08:42.398 -> I am being called
10:08:42.478 -> 脉冲个数：24.00
10:08:42.478 -> dt: 1001.00
10:08:42.518 -> 转速：239.76
10:08:42.557 -> I am being called
10:08:42.557 -> I am being called
10:08:42.597 -> I am being called
10:08:42.597 -> I am being called
10:08:42.637 -> I am being called
10:08:42.637 -> I am being called
10:08:42.677 -> I am being called
10:08:42.677 -> I am being called
10:08:42.717 -> I am being called
10:08:43.118 -> I am being called
10:08:43.158 -> I am being called
10:08:43.158 -> I am being called
10:08:43.198 -> I am being called
10:08:43.198 -> I am being called
10:08:43.238 -> I am being called
10:08:43.238 -> I am being called
10:08:43.278 -> I am being called
10:08:43.358 -> I am being called
10:08:43.398 -> I am being called
10:08:43.398 -> I am being called
10:08:43.438 -> I am being called
10:08:43.438 -> I am being called
10:08:43.478 -> I am being called
10:08:43.478 -> I am being called
10:08:43.518 -> I am being called
10:08:43.598 -> 脉冲个数：25.00
10:08:43.638 -> dt: 1001.00
10:08:43.638 -> 转速：249.75
10:08:43.678 -> I am being called
10:08:43.678 -> I am being called
10:08:43.718 -> I am being called
10:08:43.718 -> I am being called
10:08:43.758 -> I am being called
10:08:44.638 -> 脉冲个数：5.00
10:08:44.638 -> dt: 1001.00
10:08:44.678 -> 转速：49.95
10:08:45.638 -> 脉冲个数：0.00
10:08:45.678 -> dt: 1001.00
10:08:45.678 -> 转速：0.00
10:08:46.638 -> 脉冲个数：0.00`

The number of output pulses is obviously wrong, and when the white sticker is recognized, pulses are continuously output.
However, the waveforms viewed by the data acquisition card, the 6 white paper stickers identified after one revolution of the wheel, exactly output 6 pulses. The number of pulses output by the arduino is as shown above, which is not the change of high and low levels but adds 1
I am very grateful to ask the great master for advice!

tanghui0102, 2020-08-06:
[image]

earlephilhower, 2020-08-06:
The template was ignored, closing.

https://gitter.im/esp8266/Arduino or https://www.esp8266.com are the appropriate forums for general questions like this.

closed 2020-08-06 by earlephilhower