esp8266/Arduino #7508: Obtener mensaje en el campo de contraseña

juanketzer, 2020-08-06:
How can I get a message, from the password field, when a user wants to connect to my WifiAP?

I tried to find the part of the code of ESP8266WifiAP.h and Esp8266WifiAP.cpp where it is checked if the AP password entered by the user, from your device (computer, phone) is correct or not, to extract that variable, but I did not get results.

I try to give different privileges to the user who accesses my Wi-Fi network, depending on the password he enters.

[image]

@earlephilhower

juanketzer, 2020-08-06:
@igrr @mmiscool any ideas? please.

igrr, 2020-08-06:
Please excuse my limited knowledge of the Wi-Fi protocol, but I don't think this is possible. The password is used as an input to the key derivation function, which produces a pre-shared key. This key is then used in the 4-way handshake between the station and the AP. In other words, the plaintext password is never actually transferred to the AP during the handshake. And since the AP side expects a certain pre-shared key, you can't have two different passwords.

You may try to solve the problem at the application layer, by offering some captive portal where the user would be additionally authenticated.

earlephilhower, 2020-08-06:
The template was ignored, closing.

https://gitter.im/esp8266/Arduino or https://www.esp8266.com are the appropriate forums for general questions like this.

P.S. Hi @igrr!

closed 2020-08-06 by earlephilhower