esp8266/Arduino #7509: can`t upload a simpel sketch.

Skywalkerforce, 2020-08-06:
I was going to upload a new sketch,  as i change my WiFi, so i needed to update my esp8266, and u could not uplod it.
now i try to upload a simpel sketch from example "blink"

i can`t remember if i used Generic ESP8266 Module last time, or if it is some other card??

i get the following message

Arduino:1.8.13 (Windows Store 1.8.42.0) (Windows 10), Kort:"Generic ESP8266 Module, 80 MHz, Flash, Disabled, ck, 26 MHz, 40MHz, DOUT (compatible), 512K (no SPIFFS), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Sketch uses 258604 bytes (51%) of program storage space. Maximum is 499696 bytes.

Global variables use 26696 bytes (32%) of dynamic memory, leaving 55224 bytes for local variables. Maximum is 81920 bytes.

warning: espcomm_sync failed

error: espcomm_open failed

Ett fel inträffade under uppladdning av sketchen

error: espcomm_upload_mem failed

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

i have the correct port (com3) i have installed ch340g.

is it my card thats the fault?

earlephilhower, 2020-08-06:
The template was ignored, closing.

https://gitter.im/esp8266/Arduino or https://www.esp8266.com are the appropriate forums for general questions like this.

closed 2020-08-06 by earlephilhower

devyte, 2020-08-06:
Also, the espcomm errors are documented for the most part.