esp8266/Arduino #7522: ESP8266 resetting after few second of working

maktech, 2020-08-13:
hello fellas,
i am new to ESP 8266, and i am using Ch340 Wifi dev board for my project. i am using 3 library files for my project 
1) https://github.com/FirebaseExtended/firebase-arduino
2) https://github.com/bblanchon/ArduinoJson
3) https://github.com/tzapu/WiFiManager

after flashing the code, the dev board automatically resets after 40-50sec with an error 
 
06:55:24.083 ->  ets Jan  8 2013,rst cause:4, boot mode:(3,6)
06:55:24.083 -> 
06:55:24.083 -> wdt reset
06:55:24.083 -> load 0x4010f000, len 3584, room 16 
06:55:24.083 -> tail 0
06:55:24.083 -> chksum 0xb0
06:55:24.083 -> csum 0xb0
06:55:24.083 -> v2843a5ac
06:55:24.083 -> ~ld
06:55:24.151 -> *WM: 

sharing my code:

#include <ESP8266WiFi.h>
#include<FirebaseArduino.h>
#define SKIP_FP_CONTROL
#define FIREBASE_HOST "automation-de1d3.firebaseio.com"                     
#define FIREBASE_AUTH "fuUYK7W3fn11wUNrf1OT4TBlRaE5XXXXXXXXXX"            
#define WIFI_SSID "xxx"                                                                                            
#define WIFI_PASSWORD "yyy"    
                                  

#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>

#define Relay1 5  //D1
int val1=0;

#define Relay2 4  //D2
int val2=0;

#define Relay3 D5 //D5
int val3=0;

#define Relay4 12 //D6
int val4=0;

#define Relay5 13 //D7
int val5=0;

#define Relay6 15 //D8
int val6=0;

void setup() 
{
  Serial.begin(115200);                                                    // Select the same baud rate if you want to see the datas on Serial Monitor
  pinMode(Relay1,OUTPUT);
  pinMode(Relay2,OUTPUT);
  pinMode(Relay3,OUTPUT);
  pinMode(Relay4,OUTPUT);
  pinMode(Relay5,OUTPUT);
  pinMode(Relay6,OUTPUT);
  

  digitalWrite(Relay1,LOW);
  digitalWrite(Relay2,LOW);
  digitalWrite(Relay3,LOW);
  digitalWrite(Relay4,LOW);
  digitalWrite(Relay5,LOW);
  digitalWrite(Relay6,LOW);
  
  

 
 // Connect to WiFi access point.
  WiFiManager wifiManager;
  wifiManager.autoConnect("heart");
  Serial.println(); 
  

  Serial.println("WiFi connected");
  Serial.println("IP address: "); 
  Serial.println(WiFi.localIP()); 

  if (WiFi.status() != WL_CONNECTED) {
    delay(5000);
    Serial.print(".");
    ESP.restart();}

  
 
  
  Firebase.begin(FIREBASE_HOST,FIREBASE_AUTH);
  //Firebase.setInt("9540905077/bed1S1",0);                     
  //Firebase.setInt("9540905077/bed1S2",0); 
  //Firebase.setInt("9540905077/bed1S3",0); 
  //Firebase.setInt("9540905077/bed1S4",0);
  //Firebase.setInt("9540905077/bed1S5",0);
  //Firebase.setInt("9540905077/bed1S6",0);
  
}
void firebasereconnect()
{
  Serial.println("Trying to reconnect");
    Firebase.begin(FIREBASE_HOST,FIREBASE_AUTH);
     
    
  }

void loop() 
{  

    delay(0);
  
  /*if (Firebase.failed())
      {
      Serial.print("setting number failed:");
      ESP.restart();                                        // Restarts the device
     
      Serial.println(Firebase.error());
      firebasereconnect();
      return;
      }*/
      
  val1=Firebase.getString("9540905077/bed1S1").toInt();                   //Reading the value of the varialble Status from the firebase

  if(val1==1)                                                             // If, the Status is 1, turn on the Relay1
     {
      digitalWrite(Relay1,HIGH);
      Serial.println("light 1 ON");
    }
    else if(val1==0)                                                      // If, the Status is 0, turn Off the Relay1
    {                                      
      digitalWrite(Relay1,LOW);
      Serial.println("light 1 OFF");
    }

  val2=Firebase.getString("9540905077/bed1S2").toInt();                   //Reading the value of the varialble Status from the firebase
  
  if(val2==1)                                                             // If, the Status is 1, turn on the Relay2
     {
      digitalWrite(Relay2,HIGH);
      Serial.println("light 2 ON");
    }
    else if(val2==0)                                                      // If, the Status is 0, turn Off the Relay2
    {                                      
      digitalWrite(Relay2,LOW);
      Serial.println("light 2 OFF");
    }

   val3=Firebase.getString("9540905077/bed1S3").toInt();                  //Reading the value of the varialble Status from the firebase
  
  if(val3==1)                                                             // If, the Status is 1, turn on the Relay3
     {
      digitalWrite(Relay3,HIGH);
      Serial.println("light 3 ON");
    }
    else if(val3==0)                                                      // If, the Status is 0, turn Off the Relay3
    {                                      
      digitalWrite(Relay3,LOW);
      Serial.println("light 3 OFF");
    }

   val4=Firebase.getString("9540905077/bed1S4").toInt();                  //Reading the value of the varialble Status from the firebase
  
  if(val4==1)                                                             // If, the Status is 1, turn on the Relay4
     {
      digitalWrite(Relay4,HIGH);
      Serial.println("light 4 ON");
    }
    else if(val4==0)                                                      // If, the Status is 0, turn Off the Relay4
    {                                      
      digitalWrite(Relay4,LOW);
      Serial.println("light 4 OFF");
    } 

     val5=Firebase.getString("9540905077/bed1S5").toInt();                 //Reading the value of the varialble Status from the firebase
  
  if(val5==1)                                                              // If, the Status is 1, turn on the Relay5
     {
      digitalWrite(Relay5,HIGH);
      Serial.println("light 5 ON");
    }
    else if(val5==0)                                                       // If, the Status is 0, turn Off the Relay5
    {                                      
      digitalWrite(Relay5,LOW);
      Serial.println("light 5 OFF");
    } 

     val6=Firebase.getString("9540905077/bed1S6").toInt();                 //Reading the value of the varialble Status from the firebase
  
  if(val6==1)                                                              // If, the Status is 1, turn on the Relay6
     {
      digitalWrite(Relay6,HIGH);
      Serial.println("light 6 ON");
    }
    else if(val6==0)                                                       // If, the Status is 0, turn Off the Relay6
    {                                      
      digitalWrite(Relay6,LOW);
      Serial.println("light 6 OFF");
    }

}

Please help me on this issue

Tech-TX, 2020-08-13:
The folks at the ESP8266.com are the appropriate  place for this.  This is an area for issues with the code and libraries. 
Your question there should be something along the line of "why is this causing A WDT reset?"  There's  a bit of help at readthedocs that explains common WDT causes. If you check the common causes first most folks are more willing to help...

devyte, 2020-08-13:
Previous comment is correct. This is an issue tracker, not a general support forum.
Also, issue template was ignored. 
Closing due to not compliant.

closed 2020-08-13 by devyte