esp8266/Arduino #7523: 2.7.2 and up: http.getString(); returns empty String - 2.7.1 works

beejayf, 2020-08-13:
#7384 # Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[NodeMCU ESP-12E]
- Core Version:			[2.7.2 - 2.7.4]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[did not find info]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200)

### Problem Description

I have a very simple sketch that worked fine with ESP8266 board version 2.7.1 - but with 2.7.2 it just returns an empty string

This happened on a NodeMCU ESP-12E with the latest Adruino IDE (1.8.13 - Windows 10 Install (not the store version - if this is important) - since this is the only type I have right now, I cannot check it on different hardware. I tried two PCs - same result.

once I roll back to 2.7.1 it works - back to 2.7.2: empty string.

Am I doing something wrong or is this a bug in the latest releases?

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
 
const char* ssid = "mynet";
const char* password = "mypwd";
 
void setup () {
  Serial.begin(115200);
  WiFi.begin(ssid, password);
 
  while (WiFi.status() != WL_CONNECTED) {
 
    delay(1000);
    Serial.print("Connecting..");
  }
}
 
void loop() {
  Serial.println("Connected..");
  if (WiFi.status() == WL_CONNECTED) { //Check WiFi connection status
 
    HTTPClient http;  //Declare an object of class HTTPClient
 
    http.begin("http://192.168.10.101/temp");  //Specify request destination
    int httpCode = http.GET();                                                                  //Send the request
    Serial.print("htcode: "); 
    Serial.println(httpCode);
    if (httpCode > 0) { //Check the returning code
 
    String payload = http.getString();   //Get the request response payload
    Serial.print("###");
    Serial.print(payload);                     //Print the response payload
    Serial.println("###");
  }
 
  http.end();   //Close connection
 
}
 
delay(30000);    //Send a request every 30 seconds
 
}

### Debug Messages

```

```

d-a-v, 2020-08-13:
I tried and can't reproduce.
I tried with 2.7.1, 2.7.2 and 2.7.4.
The correct way is below, but with or without the change, it works.

```
[...]
    WiFiClient clent;
    HTTPClient http;  //Declare an object of class HTTPClient
 
    http.begin(client, "http://192.168.10.101/temp");  //Specify request destination
[...]
```

What does this command report ?
```
    curl -D - -o forgetme http://192.168.10.101/temp
```

devyte, 2020-08-15:
Closing for now due to can't reproduce.

closed 2020-08-15 by devyte

beejayf, 2020-08-15:
> 
> ```
>     curl -D - -o forgetme http://192.168.10.101/temp
> ```

```
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0HTTP/1.1 200 OK
Content-Type: text/html

100   245    0   245    0     0   1580      0 --:--:-- --:--:-- --:--:--  1590
```

beejayf, 2020-08-15:
Thanks - I'm assuming it is something about my setup - I'll keep on searching for the root cause

d-a-v, 2020-08-16:
The above curl output shows an incomplete http header coming back from your server.
At least `Content-Length` is missing and is mandatory.

```
$ curl -D - -o forgetme http://google.com
Location: http://www.google.com/
Content-Type: text/html; charset=UTF-8
Date: Sun, 16 Aug 2020 10:41:37 GMT
Expires: Tue, 15 Sep 2020 10:41:37 GMT
Cache-Control: public, max-age=2592000
Server: gws
Content-Length: 219
X-XSS-Protection: 0
X-Frame-Options: SAMEORIGIN
```

beejayf, 2020-10-29:
@d-a-v - Thanks! I had no clue that I need to advance the length - but now all my devices run their requests at lightning speed!