esp8266/Arduino #7528: No memory leak in ESP8266WiFi...

peychart, 2020-08-15:
#7384 ## Platform

- Hardware:				ESP8266
- Core Version:			current version 
- Development Env:		Platformio
- Operating System:		Ubuntu

### Settings in IDE

- Module:  			Wemos D1 mini r2
- Flash Mode:			[qio|dio|other]
- Flash Size: 		        4MB
- lwip Variant:                  [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	        40Mh]
- CPU Frequency:		80Mhz
- Upload Using:		OTA & SERIAL
- Upload Speed:              115200 (serial upload only)

### Problem Description
Mermory leak on WiFi use, particularly quickly in case of DNS problems, until the device reboot...

Hi friends and thank you for your work.

After several hours in google searching, I understood that there is no memory leak in this library (ESP8266WiFi).
However, the fact remains that the device ends up rebooting, particularly quickly when DNS problems occur (on cut of internet link for example).

My use : https://github.com/peychart/WiFiPowerStrip/blob/master/Next-version/WiFiManager.cpp, line 109
### [MCVE](https://stackoverflow.com/help/mcve) Sketch

My project : https://github.com/peychart/WiFiPowerStrip/tree/master/Next-version

 So, yes this project isn't very small... But, the punishment is the same with the lightweight C version of the prototype (https://github.com/peychart/WiFiPowerStrip) in operation for many months ... :-(

Code:
#include <Arduino.h>
#include <ESP8266HTTPUpdateServer.h>
#include <uart.h>

#include "untyped.h"                   //<-- de/serialization object for JSON communications and backups on SD card
#include "WiFiManager.h"           //<-- WiFi connection manager
#include "pins.h"                         //<-- pins object manager with serial pin extension manager
#include "switches.h"                  //<-- inputs management
#include "webPage.h"                 //<-- definition of a web interface
#ifdef DEFAULT_MQTT_BROKER
  #include "mqtt.h"                       //<-- mqtt input/output manager
#endif
#ifdef DEFAULT_NTPSOURCE
  #include "ntp.h"                         //<-- time-setting manager
#endif

#include "setting.h"                      //<--Can be adjusted according to the project...
#include "debug.h"                      //<-- telnet and serial debug traces

WiFiManager                         myWiFi;
...
void setup() {
   ...
    myWiFi.version                ( VERSION )
                .onConnect          ( onWiFiConnect )
                .onStaConnect     ( onStaConnect )
                .ifStaConnected   ( ifStaConnected )
                .ifConnected         ( ifWiFiConnected )
                .onStaDisconnect ( onStaDisconnect )
                .hostname            ( DEFAULTHOSTNAME )
                .setOutputPower  ( 17.5 )
               .restoreFromSD    ();
 ...
}

void loop() {
  ESPWebServer.handleClient(); delay(1L);                  //WEB server
  myWiFi.loop();                                                             //WiFi manager
  mySwitches.inputEvent(intr, rebound_completed);     //Switches management
  myPins.timers();                                                          //Timers control for outputs
  myPins.serialEvent();                                                  //Serial communication for the serial slave management
#ifdef DEFAULT_MQTT_BROKER
  myMqtt.loop();                                                            //MQTT manager
#endif
}

### Debug Messages
 Device reboot...
```
Debug messages go here
```

closed 2020-08-16 by earlephilhower

peychart, 2020-08-16:
Thanks for your help.

earlephilhower, 2020-08-16:
Fair enough @peychart , I probably should have explained the reason for closing.

This was closed because you've posted a sample that doesn't compile, using an external library and headers, with a very vague OOM error report that's not reproducible outside of your own setup.  So, there's really nothing we can do here.  Your best bet is https://www.esp8266.com or the Gitter chat at https://gitter.im/esp8266/Arduino .

If you can pull together a complete, failing MCVE we can give it a go, but w/o one (as said in the issue template) it's not going to be possible.

peychart, 2020-08-16:
809/5000
Yes, I understand ... but if I could make a simple, compilable example, which reproduces this error in ten lines, of course I would ... ;-)
The problem is that this malfunction only appears on important sources, which do not leave enough memory to allow this drift.
I worked for ten hours on this problem, checking my prore code hundreds of times ... Each time, I come back to the same Wifi problem, especially as I said, when the DNS is no longer network access.
the web server only sends const __FlashStringHelper * to the web servers, I comment the analysis of the responses ... so no memory leaks possible on that ... when I stop the communications, no more drift ...
Here is the state of my report ... :-(
I am running out of solution (although the workaround will do the trick and keep the memory pool free within 18k ... but with periodic disconnections / reconnections of the Wifi)
Regards

peychart, 2020-08-16:
Fantastic! ... I received an update (expressive ... I did not look well - I couldn't stay in front of the screen - but it seemed to go on for a while) in platformio a few hours ago and, I do not know if it is for just a coincidence but now, when I try to refresh the web page every second for more than several minutes ... I can no longer reach the low limit (16k8) of free memory which triggered the disconnection / reconnection of Wifi ( thus avoiding the reboot)! ...

Now the memory seems to stabilize durably towards the lowest 17k2 and remote very quickly to 18k5 when I stop torturing the interface ... LOL

I just have to find the right changelog files to make sure that I am not dreaming ... LOL
(and test it under DNS failures)...
:-)))