esp8266/Arduino #7529: SPIFFS dropped from the official distribution, why?

rin67630, 2020-08-16:
Apparently, SPIFFS was dropped from the official distribution

OK, LittleFS should be used instead, but it is NOT a replacement for me, since LittleFS has a minimum block size of 4K, which would mean a dramatic waste of the limited EEPROM space and invoke too much write operations for simple ongoing message logging with messages about 60 bytes length each.
The 256 bytes from SPIFFS are just tremendously more efficient and save unnecessary writes on an EEPROM that is very limited. That block size was already too big for a small EEPROM size, but to impose 4K is just unacceptable!

I can surely get the SPIFFS lib from GitHub and include it manually, but it will unnecessarily complicate the deployment of my project to other users.

Why did you drop a successful working library, that thousands people use?
Is might not be maintained any more, but it worked and has definitely advantages over littleFS.

Please restore the lib as is was, you may include a deprecation warning, but please, let it be used!

earlephilhower, 2020-08-16:
It's not dropped, it's deprecated.  If you have code that's not compiling at all now, let us know.  All you should be getting is lots of warnings on builds.

SPIFFS upstream is no longer maintained actively, there are many known issues and really bad behaviors seen, and it is very, very slow.  Feel free to use SPIFFS, but if you find weird errors or crashes, we won't be able to help you.

closed 2020-08-16 by earlephilhower

rin67630, 2020-08-16:
The library is not found any more in the Arduino environment
To distribute my software i now need to instruct how to additionally download the lib from Github. 
:-(

devyte, 2020-08-16:
The SPIFFS lib has not been dropped, the lib is where it has always been: in this core. It sounds like you have an installation issue.