esp8266/Arduino #7532: Symbol not found: _getentropy

infinity-computers-dot-net, 2020-08-17:
When attempting to compile any sketch against the ESP8266 core version 2.7.4 on a Mac, I get this error:
--
dyld: lazy symbol binding failed: Symbol not found: _getentropy
  Referenced from: /Users/infinity/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1/python3
  Expected in: /usr/lib/libSystem.B.dylib

dyld: Symbol not found: _getentropy
  Referenced from: /Users/infinity/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1/python3
  Expected in: /usr/lib/libSystem.B.dylib

signal: trace/BPT trap
/Applications/Arduino.app/Contents/Java/arduino-builder returned 255
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.
--
What the hell is going on?

Are you, by any chance, making assumptions about my system libraries without actually _checking?_

earlephilhower, 2020-08-17:
I think you need to direct your ire at XCode, which is used to build Python3.  We just use it's output!

What Mac OS version are you using?

No other Mac users have reported this issue, so it seems like something special in your config, likely related to the OS version, some libs you might have/not have, or maybe something path related.

You can remove the python3 executable installed in the tools dir and make a symlink to your system's native Python3 interpreter, in a pinch, to get going.

infinity-computers-dot-net, 2020-08-17:
I am running OSX 10.11.6.

My ire would have been directed at Apple, were it not for the fact that your core went from "working" to "broken" without any other changes to my system, simply by updating it to the latest version.

> $ cd ~/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1
> $ ./python3
> dyld: lazy symbol binding failed: Symbol not found: _getentropy
> ...etc...

Seems that error is thrown simply by _running the python executable_ you have bundled.

Strikes me that the problem here lies entirely with the version of python you have bundled with your core. If my version of python works, and yours does not, how is this "a problem with my os", "a problem with xcode" or "a problem with my path"? This is *your python!*

devyte, 2020-08-17:
> What the hell is going on?

Single warning: I will ask you to please contain your frustration and mind your phrasing.

The build on Mac is currently covered by CI. In addition, no other user has reported a build problem. 
I strongly encourage you to breathe, remember that we maintainers of this repo contribute our work for free as volunteers, as do contributors, and that you're making use of the results of that work at no cost. Then, while you keep that strongly in your mind, ask _politely_ for someone to assist you investigating why you're apparently the only user with this problem.
In addition, you ignored the issue template, which we require for several reasons. Please edit your original post and comply with the template instructions, or I will close this as not compliant. The only reason I haven't already closed it is @earlephilhower 's comments above.

closed 2020-08-17 by earlephilhower

julianrendell, 2020-09-25:
I can confirm that the latest versions of ESP8266 for Arduino aren't working for the 4 year old version of MacOS referenced.

With a little experimentation we've found 2.5.2 still works with MacOS 10.11.6. Our real answer is to upgrade to as new-a-MacOS as the machine in question can support.

infinity-computers-dot-net, 2020-09-25:
"Your OS is four years old, therefore it's your problem" is not a valid stance. Only three versions of Windows, for example, have been published in the last 20 years. Plus, older Macs won't run the newer OSs, for no good reason other than "Apple like money".

Your core used to work on 10.11.6 without issues. It's only recently that it's been broken.

The solution is *not* for me to make radical changes to my computer. The solution is for you to figure out what you did to break it, and then unbreak it.

devyte, 2020-09-25:
> It is your job to ensure that your software works with as many operating systems as possible.

Nope! Nobody here is getting paid. It is certainly not _our job_ to maintain _your_ build environment. I will repeat: this is a community driven development.

> The solution is for you to figure out what you did to break it, and then unbreak it.

Also nope! None of the current maintainers use Mac. We support Mac in a best-effort basis, similar to how PIO is covered, e. g. we have building on Mac covered by CI, but that's it. Anything beyond that must come from the community, including more CI jobs to improve coverage. 
If _you_ have a build issue on _your_ Mac, then I suggest teaming up with anyone else who has the same issue, figuring out a fix, and contributing it here so that all may benefit. We're always glad to look at and discuss PRs.