esp8266/Arduino #7537: Symbol not found: _getentropy

infinity-computers-dot-net, 2020-08-17:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] I have filled **in** all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[2.7.4]
- Development Env:		[Arduino IDE]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[any]
- Flash Mode:			[any]
- Flash Size: 		[any]
- lwip Variant:   [any]
- Reset Method:		[any]
- Flash Frequency:	[any]
- CPU Frequency:		[any]
- Upload Using:		[any]
- Upload Speed:   [any] (serial upload only)

### Problem Description

See #7532

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

infinity-computers-dot-net, 2020-09-25:
What feedback are you waiting for, exactly?

infinity-computers-dot-net, 2020-09-25:
So, let me see if I have this right.

1. Due to recent changes, this project is now *broken* on Macs more than 4 years old.

2. None of the active devs on this project take ownership of this problem, and aren't going to.

3. If your product doesn't work on my machine, it's *my* job to fix it, despite knowing nothing about how it works, and despite the fact that it wasn't me who broke it.

That about cover it?

devyte, 2020-09-26:
> What feedback are you waiting for, exactly?

Some other member of the community to offer you help. It's been 23 days and no one has offered to help you.

devyte, 2020-09-26:
> 1. Due to recent changes, this project is now _broken_ on Macs more than 4 years old.

That is possible, yes. 

> 2. None of the active devs on this project take ownership of this problem, and aren't going to.

That is correct, the issue does not apply to us, and we're not going to buy a Mac. 

> 3. If your product doesn't work on my machine, it's _my_ job to fix it, despite knowing nothing about how it works, and despite the fact that it wasn't me who broke it.

You still don't seem to understand that it's not _our_ product. I don't work for Espressif, nor do I sell these modules, nor do I make money in _any_ way related with this repo. We maintainers are _volunteers_, this is an open source project (for the most part), and the contributions come from users like you. If you have no idea how something here works, you have all the tools freely available and at your disposal to investigate it and find your own solution. You can find peers to work with you on that. You can (politely) ask others in the community questions and get clarifications or hints about how to proceed. But you do not get to make demands of others nor expect others to fix things that are relevant only to you. If what you want is support for yourself, then find someone willing to meet with you privately, and I'm sure you'll reach some financial arrangement.

infinity-computers-dot-net, 2020-09-26:
Of course it's "your" product. It's certainly not mine. Your name is on the door. You have the authority to lock threads (and have already done so). I don't have any of these things. Ergo, it is not my product, but it *is* yours.

What I want *isn't* "personal tech support", as you well know. The problem here isn't "the product works everywhere else but not on my computer". The situation is, instead, "the product worked five minutes ago, and then YOU went and made changes that broke it on Mac".

So, I opened a BUG REPORT on your BUG TRACKER. Which is what the damn thing is FOR!!

In response, you tell me that I should fix it myself. So what, then, is the point in having a bug tracker? What is the point in telling the maintainers of a project about bugs and regressions, if they're just going to tell me to bugger off and fix it myself?

devyte, 2020-09-26:
> what, then, is the point in having a bug tracker? 

To allow tracking of the issue, and give others in the community a chance to provide additional details and work with you towards a solution. 

> What is the point in telling the maintainers of a project about bugs and regressions, if they're just going to tell me to bugger off and fix it myself?

It seems I have to explain yet again: we are maintainers of this community driven repo. We are not maintainers of a commercial repo. We are not obliged to provide a service for you or anyone else. As maintainers, we coordinate the efforts put forth by the community, because we choose to volunteer our personal time and effort towards that end. In addition, we are also users, just like you, and as users we contribute fixes that are relevant to us, as you may contribute fixes that are relevant to you, if you wish to do so. 
You do not report bugs and regressions to us maintainers, you report bugs and regressions to the community, to allow coordination with other users who are also interested in those bugs and regressions.

> it's relevant to all Mac owners with an older version of the OS. 

> just going to tell me to bugger off and fix it myself

This issue remains open precisely to allow other Mac owners with an older version of the OS to offer additional info and/or to work with you. It has been 23 days without any progress or any additional information at all. That hints strongly at this issue being relevant to a very few, and yes, maybe just you. 

Enough. Standing policy is to keep valid and compliant issues that are waiting for feedback open for 20 days. If no feedback has been received in that period of time, the issue may be closed by any maintainer. Unless another maintainer objects, I offer to keep this issue open waiting for feedback from other users for an additional week from today. If instead of focusing on a solution you insist on pursuing this pointless discussion, I will withdraw that offer. 

Your choices for a solution here are:
1. Fix it yourself 
2. Find someone willing to work with you towards a fix
3. Do nothing and hope someone else fixes it for you, or at least provides meaningful info, within the remaining week

earlephilhower, 2020-09-26:
I'm closing this @devyte.  You've made yourself clear and there's nothing to do here.

MacOS 10.11.6 stopped receiving updates 2 years ago from Apple, as best I can tell, so it's not supported and not something I'd want to connect to the internet with.  10.15 has been out for about a year, and 11.0 is around the corner.  Older Macs are running just fine with the current repo, as long as they can/have been updated to a supported OS version.

The workaround for anyone else stuck at such an old OS is to stick w/the older releases.  They're available through the Arduino installer.

closed 2020-09-26 by earlephilhower