esp8266/Arduino #7548: ESP8266 TCP transfer is slow

durlabhjain, 2020-08-24:
I'm trying to transfer about 400KB of data from ESP8266 (NodeMCU) to a server. What we get:

- NodeMCU to Azure Server - 7 to 9 seconds
- NodeMCU to Cloud Server 1 - 25 seconds
- NodeMCU to Cloud Server 2 - 37 seconds
- NodeMCU to Local Server - 4 seconds

The bandwidth on all the Cloud Servers is at-least 1GBPS while ping time does differ from my local network. What I noticed that the TCP packet size is 1460 bytes so I'm assuming that the difference is due to the ping time on each packet being sent. 

On all these servers, using a FTP client or simple HTTP upload via web-browser, I can upload the file within 1-2 seconds. I was wondering, it there any further optimization possible to improve upload? I do believe, I've setNoDelay correctly set.

The attached code should work as is with just changing the WiFi settings. I've the current server open for internet for now for file uploads.
[ClientUploadPsuedo.ino.txt](https://github.com/esp8266/Arduino/files/5118395/ClientUploadPsuedo.ino.txt)

The relevant portion of the code is:

```
void setup()
{
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);

  startTime = millis();
  lastDebugMessageTime = startTime;

  DEBUG_PRINT("starting wifi connection....");
  
  WiFi.begin(ssid, password);
  delay(INITIAL_WIFI_CONNECT_TIMEOUT);
}

bool connectWiFi() {

  DEBUG_PRINT("Connecting to WiFi");
  for(int trial = 0; trial < MAX_WIFI_RETRY; trial++) {
    if(trial > 0) {
      delay(WIFI_RETRY_DELAY);
      DEBUG_PRINT("retrying...");
    }
    if (WiFi.status() == WL_CONNECTED) {
        DEBUG_PRINT("WiFi connected");
      
        // Print the IP address
        //DEBUG_PRINT(WiFi.localIP());
        return true;
    }
  }
  DEBUG_PRINT("Error connecting to wifi. Aborting...");
  return false;
}

uint32_t connectToServer()
{
    uint8_t maxRetries = 2;
    uint8_t retry_counter = 0;
    bool canContinue = false;
    for (uint8_t attempt = 0; attempt < maxRetries; attempt++)
    {
    if(client.connect(SERVER, SERVER_PORT))
        {
            client.setNoDelay(true);
            canContinue = true;
            break;
        }
    }
    if (!canContinue)
    {
        return false;
    }
    return true;
}

//ImageStream imageStream = ImageStream(200 * KB, 200 * KB, 

void loop()
{
  uint8_t image_header[40];

  uint32_t image1size = 200 * KB;
  uint32_t image2size = 200 * KB;

  if(connectWiFi())
  {
    DEBUG_PRINT("creating image stream");

    ImageStream imageStream = ImageStream(image1size, image2size, image_header, 40);
    
    uint32_t image1_address = 0;
    uint32_t image2_address = 0;
    imageStream.setAddresses(image1_address, image2_address);

    DEBUG_PRINT("starting upload");
    DEBUG_PRINT(imageStream.size());
    if (connectToServer())
    {
      client.println("POST /Handler.ashx HTTP/1.1");
      client.println("Host: linux.stream4.tech");
      client.println("Content-Type: application/octet-stream");
      client.println("Content-Length: " + String(imageStream.size()));
      client.println();
      client.write(imageStream);
      client.stop();
      DEBUG("Http Response Code");
    }
    else
    {
      DEBUG_PRINT("Error connecting");
    }
  }
  timeTaken();
  delay(5000);
  timeTaken();
}

void DEBUG(const int number) {
  Serial.print(number);
  Serial.print("\t");
  Serial.println(timeTaken(), DEC);
}

void DEBUG(const String& message) {
  Serial.print(message);
  Serial.print("\t");
  Serial.println(timeTaken(), DEC);
}

long timeTaken() {
  long time = millis() - lastDebugMessageTime;
  lastDebugMessageTime = millis();
  return time;
}
```

Code is also available at https://gist.github.com/durlabhjain/b9310ed5c5b4d213ce4c8f333d8ea3c9#file-clientuploadpsuedo-ino

I'm totally confused why would there be such a huge different in upload speed. Due to speed issues, currently ESP8266 doesn't seem like a good WiFi Client option which doesn't make sense to me.

durlabhjain, 2020-08-24:
Thanks for such a quick reply and I apologize for not following the format.

I'm using ESP v2.7.4. Following is information on my setup. I had lwIP v2 lower memory. High bandwidth has improved the speeds - to double. Is there any documentation on lwIP settings? I apologize if it is is already documented but I couldn't find with search.

Platform:
- Hardware: [NodeMCU 1.0 (ESP-12 Module)]
- Core Version: [2.7.4]
- Development Env: Arduino 1.8.10
- Operating System: Windows

Settings in IDE:
- Module: [NodeMCU 1.0 (ESP-12 Module)]
- Flash Size: [4MB]
- lwip Variant: [v2 Lower Memory]
- CPU Frequency: [80MHz]
- Upload Serial: [115200]

Are there any other such settings that I should be aware of that I may have missed.

d-a-v, 2020-08-24:
> Are there any other such settings that I should be aware of that I may have missed.

You might try to run @ 160MHz.
Currently TCP options are described in #7519.

closed 2020-08-24 by durlabhjain