esp8266/Arduino #7549: esp8285 generic board missing the sdk options

5chufti, 2020-08-25:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:			[ESP8285 device]
- Core Version:			[latest git]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8285 Module]
- Flash Mode:			[-]
- Flash Size: 			[1MB]
- lwip Variant:   		[v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:		[-]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   		[other] (serial upload only)

### Problem Description
esp8285 generic board missing sdk options

### [MCVE](https://stackoverflow.com/help/mcve) Sketch
not relevant

d-a-v, 2020-08-25:
This was originally on purpose. Only generic esp8266 has this menu entry.
All other boards use the default sdk entry are a specific set of esp8266 generic menu.
esp8285 is esp8266 with DOUT and 1MB.

Given that esp8285 can be considered as a generic board too, this is a fair request.
Would you be able to propose a pull request for that ?

5chufti, 2020-08-25:
for my taste there are allready too many variants (and options) - slowing down Arduino IDE (start) massively, only working with the two generic modules in boards.txt. Maybe I should cut down to only one???

@d-a-v pull request #7550 done but fails for check on Boards.txt.
Started locally boards.txt is built ok .... so I don't know.

5chufti, 2020-08-26:
ok, pr merged so issue closed

closed 2020-08-26 by 5chufti

mcspr, 2020-09-01:
Not to continue discussion in the closed issue...

I wanted to also point out the #7340 (...no gh labels so far, seems to be missed...)
Unless, there are more reasons to keep 8285 and 8266 generic variants separate