esp8266/Arduino #7555: LED-Stripe music-controlled by ESP8266

Rekke, 2020-08-27:
I have a 5 meter WS2812b LED strip with 240 LEDs that I connected to an ESP8266 Nodemcu.
I would like to control it from the PC via WLAN in time with the music.
As a program I would like to use jinx! (www.ledstyles.de) (or could you suggest a better program).
**Which firmware / settings do I have to upload on the ESP8266** and how do I have set in jinx! as output device?

[image]

closed 2020-08-27 by devyte

Rekke, 2020-08-27:
why close my issue?

devyte, 2020-08-28:
Because you didn't even read the first line of the issue template, which clearly says that this is not the right place to ask "how do I" type questions.

Rekke, 2020-08-28:
Ok sorry. But maybe the answer for my problem is, that i have to upload a firmware and to install a LED-library. But i dont now which. It is not a special 3rd party problem. Maybe you have a hint.

Pablo2048, 2020-08-28:
The hint for you was in the issue template - use the forum @ esp8266.com .