esp8266/Arduino #7571: Exception 3: LoadStoreError: Processor internal physical address or data error during load or store PC: 0x4000bf64 EXCVADDR: 0x4023d9e5

9ghost9, 2020-09-02:
----------------------------- Delete below -----------------------------

----------------------------- Delete above -----------------------------

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

Detailed problem description goes here.

Just an FYI that may help you with bugs  etc .

This sketch works great with ver 2.4.2
It compiles with ver 2.7.4  but will never load the server page . 
Reloading the web page resets the ESP and throws the error and stack dump .

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>

#include "pwm_slider_button.h" // file where html page is written

#define STASSID "" // name of router
#define STAPSK  "" // pwd for router

#define PWM_pin 2  // PWM pin
#define button_pin 12 // button pin for LED etc.

const char* ssid = STASSID;
const char* password = STAPSK;

ESP8266WebServer server(80); // port 80 (standard)

// prints out main page with pwm slider and button etc...

void handle_root() {
  String root_code = html_page; // html code from .h file
  server.send(200, "text/html", root_code); // upload code to show webpage
}

// handling button press on webpage

void handle_button() {
  String button_state = server.arg("button_state"); // on page ./setButton we are reading the variable 'button_state'

  String curr_state = "OFF"; // for sending back to the page
  if (button_state == "1")
  {
    digitalWrite(button_pin, HIGH); //LED ON
    curr_state = "ON";
  }
  else
  {
    digitalWrite(button_pin, LOW); //LED OFF
    curr_state = "OFF";
  }

  server.send(200, "text/plane", curr_state); //Send state back to page
}

// handling PWM:

void handle_pwm() {
  String pwm_val = server.arg("PWMval"); // reading from slider on html page
  Serial.print("slider val: ");
  Serial.println(pwm_val);
  analogWrite(PWM_pin, pwm_val.toInt()); // changing the value on the NodeMCU board
  server.send(200, "text/plane", "0"); // handling the webpage update
}
void setup() {
 Serial.begin(115200);
  //WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);     //Connect to your WiFi router using ssid and pwd given above
  Serial.println("");

  // choose output for both PWM and button pins
  pinMode(PWM_pin,OUTPUT);
  pinMode(button_pin,OUTPUT); 

  // Connection wait
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  // print some parameters pertaining to server IP etc.
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());  // IP address given to NodeMCU
 
  server.on("/", handle_root);      // primary page where html from .h file will be shown
  server.on("/setPWM", handle_pwm); // handles the PWM values
  server.on("/setButton", handle_button); // handles button values

  server.begin();                  // begin server
  Serial.println("HTTP server started");
}

void loop() {
 server.handleClient();          //Handle tapping of buttons and sliders
}
##############################################
pwm_slider_button.h.
const char html_page [] PROGMEM = R"=====(
<!DOCTYPE HTML lang="en-US">
<html>
<head>
<meta name="viewport" content="width=device-width, initial-scale=1">
<style>
.slidecontainer {
  width: 100%;
}

.slider {
  -webkit-appearance: none;
  width: 100%;
  height: 15px;
  border-radius: 5px;
  background: #d3d3d3;
  outline: none;
  opacity: 0.7;
  -webkit-transition: .0s;
  transition: opacity .2s;
}

.slider:hover {
  opacity: 1;
}

.slider::-webkit-slider-thumb {
  -webkit-appearance: none;
  appearance: none;
  width: 25px;
  height: 25px;
  border-radius: 50%;
  background: #326C88;
  cursor: pointer;
}

.slider::-moz-range-thumb {
  width: 25px;
  height: 25px;
  border-radius: 50%;
  background: #326C88;
  cursor: pointer;
}

/* Rounded switch */

.switch {
  position: relative;
  display: inline-block;
  width: 60px;
  height: 34px;
}

.switch input { 
  opacity: 0;
  width: 0;
  height: 0;
}

.slider1 {
  position: absolute;
  cursor: pointer;
  top: 0;
  left: 0;
  right: 0;
  bottom: 0;
  background-color: #ccc;
  -webkit-transition: .4s;
  transition: .4s;
}

.slider1:before {
  position: absolute;
  content: "";
  height: 26px;
  width: 26px;
  left: 4px;
  bottom: 4px;
  background-color: white;
  -webkit-transition: .2s;
  transition: .2s;
}

input:checked + .slider1 {
  background-color: #326C88;
}

input:focus + .slider1 {
  box-shadow: 0 0 1px #326C88;
}

input:checked + .slider1:before {
  -webkit-transform: translateX(26px);
  -ms-transform: translateX(26px);
  transform: translateX(26px);
}

.slider1.round {
  border-radius: 34px;
}

.slider1.round:before {
  border-radius: 50%;
}

</style>
</head>
<body>

<h2>NodeMCU Control</h2>

<div class="slidecontainer">
  <p>GPIO 2 PWM Value: <span id="slider_p"></span></p>
  <input type="range" min="0" max="1023" value="0" class="slider" id="myRange">

<p>GPIO 12 LED Switch: <span id="led_state">OFF</span></p> 
<label class="switch">
  <input type="checkbox" onchange="state_change(this)">
  <span class="slider1 round"></span>
</label>
</div>

<script src="http://canvasjs.com/assets/script/canvasjs.min.js"></script>

<script>
var slider = document.getElementById("myRange");
var output = document.getElementById("slider_p");
output.innerHTML = slider.value;

slider.onchange = function() {
  output.innerHTML = this.value;
  pwm_change(this.value);
}
</script>

<script>
function pwm_change(val) {
  var xhttp = new XMLHttpRequest();
  xhttp.open("GET", "setPWM?PWMval="+val, true);
  xhttp.send();
}
</script>

<script>
function state_change(element) {
  var xhttp = new XMLHttpRequest();
  
  if (element.checked){
    xhttp.open("GET", "setButton?button_state=1", true);
    document.getElementById("led_state").innerHTML = "ON";
  } else if (!element.checked){
    xhttp.open("GET", "setButton?button_state=0", true);
    document.getElementById("led_state").innerHTML = "OFF";
  }
  xhttp.send();
}
</script>

</body>

</html>
)=====";

```

### Debug Messages

```
Debug messages go here
```

Tech-TX, 2020-09-02:
You'll need to install the ESP8266 Exception Decoder to make sense of the error address,

https://github.com/me-no-dev/EspExceptionDecoder

9ghost9, 2020-09-02:
Good day . 
The error code I  posted is from running the exception decoder .
There is far to many highlighted lines for me to understand the cause specifically .
 I believe that there may always be issues as such with release updates and so on .
It just struck me as odd that this particular should be resolved as there is nothing in the code that is non conforming that I can see . 
Thanks for your reply .

devyte, 2020-09-02:
@9ghost9 what @Tech-TX means is that the title:
> Exception 3: LoadStoreError: Processor internal physical address or data error during load or store PC: 0x4000bf64 EXCVADDR: 0x4023d9e5

is absolutely useless information for us, because it is specific to your particular binary, which results from your particular build. We can't really replicate the reported address in your dump. 
The stack dump you saw has detailed info about what to do. Assuming that you have the ESPExceptionDecoder installed, you copy the stack dump as indicated, paste it into the decoder, and you'll get a translation of what it all means: the function call stack at the point of crash, and what the top function was.

earlephilhower, 2020-09-03:
Please check the documentation on installing and using the EspExceptionDecoder Arduino plugin.  With what you've posted, nobody can offer any help.  You can try https://gitter.im/esp8266/Arduino for assistance in real-time on decoding exceptions.

You should also enable full debug and rerun and capture any warnings it spews out.

Closing while you get a usable debug dump and exceptino decode.

closed 2020-09-03 by earlephilhower