esp8266/Arduino #7574: Failed to connect to a broker on MQTT/TLS(failed to wait for handshak)

dorbae, 2020-09-03:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					ESP-8266
- Core Version:			2.7.4 (Package installed from Arduino IDE board manager)
- Development Env:		Arduino IDE 1.8.13
- Operating System:		MacOS Catalina 10.15.6

### Settings in IDE

- Module:  			Wemos D1 mini r2
- Flash Mode:			dio?
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description
I have a problem to connect a broker server(RabbitMQ) on MQTT/TLS. It is difficult for me to find the cause in my program because WifiClientSecure.getLastSSLError returns Reason: **Unknown error code(-2)**.
Actually, my code is a very simple code to publish a message on MQTT/TLS. It works well if I change the code as the logic to connect the server on MQTT/PLAIN without TLS. And also, the certification files are valid because I ran on a client program written by Java and using the same certification files. It works well.
I hope that I can solve this problem. Thank you :)

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
// WiFi Library 
#include <ESP8266WiFi.h>
// MQTT Client Libary
#include <PubSubClient.h>
#include <time.h>

#define DEFAULT_LOOP_INTERVAL 5000L

const char *ssid = "ssid";
const char *passwd = "password";
const char *deviceId = "1234";
const char *host = "my.mqtt.server";
const char *topic = "test/esp8266"; 
long loopInterval = 5000;

// xxxx-cert.pem)
const char cert_str[] PROGMEM = R"EOF(
// privacy
)EOF";

// xxxx-private.pem)
const char key_str[] PROGMEM = R"EOF(
// privacy
)EOF";

// rootCaCert.pem
const char ca_str[] PROGMEM = R"EOF(
// privacy
)EOF";

// ca_Cahin.pem
const char ca_chain_str[] PROGMEM = R"EOF(
// privacy
)EOF";
[... 133 lines cut ...]
0x4020550b: BearSSL::WiFiClientSecure::connect(char const*, unsigned short) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 231
0x40205a60: PubSubClient::connect(char const*, char const*, char const*, char const*, unsigned char, bool, char const*, bool) at /Users/user/Documents/Arduino/libraries/PubSubClient/src/PubSubClient.cpp line 196
0x402081d1: uart_write(uart_t*, char const*, size_t) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/uart.cpp line 509
0x40206208: HardwareSerial::write(unsigned char const*, unsigned int) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/HardwareSerial.h line 164
0x40205c50: PubSubClient::connect(char const*) at /Users/user/Documents/Arduino/libraries/PubSubClient/src/PubSubClient.cpp line 167
0x4020679c: Print::print(char const*) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/Print.cpp line 130
0x40201078: reconnect() at /Users/user/Documents/Arduino/tutorial_mqtt_client_pub_pem_test/tutorial_mqtt_client_pub_pem_test.ino line 215
0x4020781a: __delay(unsigned long) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring.cpp line 57
0x4020c730: sprintf at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/sprintf.c line 646
0x40206208: HardwareSerial::write(unsigned char const*, unsigned int) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/HardwareSerial.h line 164
0x40206214: HardwareSerial::write(unsigned char const*, unsigned int) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/HardwareSerial.h line 165
0x40206208: HardwareSerial::write(unsigned char const*, unsigned int) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/HardwareSerial.h line 164
0x401008fd: check_poison_neighbors(uint16_t) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/umm_malloc/umm_local.c line 71
0x401009b9: umm_malloc_core(size_t) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/umm_malloc/umm_malloc.cpp line 458
0x402081d1: uart_write(uart_t*, char const*, size_t) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/uart.cpp line 509
0x40206208: HardwareSerial::write(unsigned char const*, unsigned int) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/HardwareSerial.h line 164
0x40206214: HardwareSerial::write(unsigned char const*, unsigned int) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/HardwareSerial.h line 165
0x40206208: HardwareSerial::write(unsigned char const*, unsigned int) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/HardwareSerial.h line 164
0x4020413e: BearSSL::WiFiClientSecure::connected() at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 260
0x402094d0: PubSubClient::connected() at /Users/user/Documents/Arduino/libraries/PubSubClient/src/PubSubClient.cpp line 689
0x402012e1: loop() at /Users/user/Documents/Arduino/tutorial_mqtt_client_pub_pem_test/tutorial_mqtt_client_pub_pem_test.ino line 322
0x40206818: Print::println(char const*) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/Print.cpp line 198
0x4020126e: setup() at /Users/user/Documents/Arduino/tutorial_mqtt_client_pub_pem_test/tutorial_mqtt_client_pub_pem_test.ino line 307
0x40207324: loop_wrapper() at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 197
```

earlephilhower, 2020-09-03:
You need to paste the whole thing into the decoder so that it can print the faulting PC instruction and not just the stack.  That would show that functino it was in when it died.

dorbae, 2020-09-07:
Sorry, I'm late. I attached the whole my debugging message. I pasted it into the Exception decoder but, the result is similar to the old one. I think this error occurred when trying to execute PubSubClient.connect() function. I hope this one will be helpful. Thank you.

* Debug
```plain
14:34:45.728 -> {l d⸮⸮< ⸮d⸮|⸮l⸮c|⸮⸮⸮⸮s⸮c⸮c⸮⸮nn⸮log⸮⸮⸮bp⸮⸮l{d;lp⸮o⸮ 
14:34:46.019 -> SDK:2.2.2-dev(38a443e)/Core:2.7.3-3-g2843a5ac=20703003/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-30-g92add50/BearSSL:5c771be
14:34:46.019 -> 
14:34:46.019 -> 
14:34:46.019 -> [INFO ] Connecting to ddungbaeHome
14:34:46.597 -> .....scandone
14:34:49.904 -> state: 2 -> 2 (b0)
14:34:49.904 -> ..state: 2 -> 0 (2)
14:34:50.926 -> reconnect
14:34:50.926 -> ......scandone
14:34:53.768 -> state: 0 -> 2 (b0)
14:34:53.768 -> state: 2 -> 3 (0)
14:34:53.768 -> state: 3 -> 5 (10)
14:34:53.768 -> add 0
14:34:53.768 -> aid 6
14:34:53.768 -> cnt 
14:34:53.868 -> 
14:34:53.868 -> connected with ddungbaeHome, channel 6
14:34:53.868 -> dhcp client start...
14:34:53.906 -> .ip:192.168.0.119,mask:255.255.255.0,gw:192.168.0.1
14:34:54.426 -> .
14:34:54.426 -> [INFO ] Connected to Wi-Fi AP.
14:34:54.426 -> [DEBUG] IP address: 192.168.0.119
14:34:54.426 -> [DEBUG] Set TLS configuration.
14:34:54.426 -> [INFO ] Waiting for NTP time sync..................pm open,type:2 0
14:35:03.941 -> ...........................................
14:35:24.950 -> [DEBUG] Current time: Mon Sep  7 05:35:24 2020
14:35:24.950 -> [INFO ] Connecting to the server...BSSL:_connectSSL: start connection
14:35:30.232 -> BSSL:_wait_for_handshake: failed
14:35:30.232 -> BSSL:Couldn't connect. Error = 'Unknown error code.'
14:35:30.232 -> [ERROR] Failed to connect to IoT Core(rc=-2 ).
14:35:30.232 -> Reason: Unknown error code.
14:35:30.232 -> Try it again in 5 seconds
14:35:35.256 -> [INFO ] Connecting to the server...BSSL:_connectSSL: start connection
14:35:40.374 -> BSSL:_wait_for_handshake: failed
14:35:40.374 -> BSSL:Couldn't connect. Error = 'Unknown error code.'
14:35:40.374 -> [ERROR] Failed to connect to IoT Core(rc=-2 ).
14:35:40.374 -> Reason: Unknown error code.
14:35:40.374 -> Try it again in 5 seconds
14:35:45.374 -> [INFO ] Connecting to the server...BSSL:_connectSSL: start connection
14:35:49.618 -> 
14:35:49.618 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
14:35:49.618 -> 
14:35:49.618 -> Soft WDT reset
14:35:49.618 -> 
14:35:49.618 -> >>>stack>>>
14:35:49.618 -> 
14:35:49.618 -> ctx: bearssl
14:35:49.618 -> sp: 3fff21e0 end: 3fff3110 offset: 01a0
14:35:49.618 -> 3fff2380:  044d9000 00000045 0000000c 3fff4374  
14:35:49.654 -> 3fff2390:  0000003c 00001b2f 00000044 00000004  
14:35:49.654 -> 3fff23a0:  3fff26c2 3fff25aa 3fff274e 00000000  
14:35:49.654 -> 3fff23b0:  00000045 00000001 0000008a 3fff274c  
14:35:49.654 -> 3fff23c0:  0000555f 3fff2638 000050ca 00000001  
14:35:49.654 -> 3fff23d0:  3fff25a8 00000001 00000000 00000000  
14:35:49.654 -> 3fff23e0:  3fff274c 000816a3 00014916 3fff2636  
14:35:49.654 -> 3fff23f0:  3fff2636 3fff25aa 00000000 3fff2636  
14:35:49.654 -> 3fff2400:  3fff26c2 3fff27da 3fff2750 4022e4c5  
14:35:49.690 -> 3fff2410:  3fff25aa 0000555f 0000008c 00000000  
14:35:49.690 -> 3fff2420:  3fff2636 00000003 00000002 00000008  
14:35:49.690 -> 3fff2430:  00000002 3fff274e 00000001 0000008c  
14:35:49.690 -> 3fff2440:  6443dbe2 00000444 3fff27da 3fff26c2  
14:35:49.690 -> 3fff2450:  3fff0582 00000046 ffffffff 3fff4374  
14:35:49.690 -> 3fff2460:  f446b87b 004fc32d f6e5563e 3fff2636  
14:35:49.690 -> 3fff2470:  3fff251e 3fff25aa 00000001 40220208  
14:35:49.690 -> 3fff2480:  00000350 ff7db300 fff296aa 000000b7  
[... 320 lines cut ...]
0x401008fd: check_poison_neighbors(uint16_t) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/umm_malloc/umm_local.c line 71
0x40100c20: umm_malloc(size_t) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/umm_malloc/umm_malloc.cpp line 552
0x401007e8: get_poisoned(void*, size_t) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/umm_malloc/umm_poison.c line 117
0x4021b2e4: ip4_output_if_opt at core/ipv4/ip4.c line 1577
0x40100260: malloc(size_t) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/heap.cpp line 232
0x4021b30a: ip4_output_if at core/ipv4/ip4.c line 1550
0x4021beb3: ip_chksum_pseudo at core/inet_chksum.c line 395
0x401001f6: millis() at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring.cpp line 188
0x402173ee: tcp_output at core/tcp_out.c line 1361
0x402131ed: sys_timeout_abs at core/timeouts.c line 189
0x4020720c: __esp_yield() at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 119
0x40207806: __delay(unsigned long) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring.cpp line 54
0x40203910: WiFiClient::connect(IPAddress, unsigned short) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WiFi/src/include/ClientContext.h line 144
0x4020550b: BearSSL::WiFiClientSecure::connect(char const*, unsigned short) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 231
[... cut ...]
0x402012e1: loop() at /Users/user/Documents/Arduino/tutorial_mqtt_client_pub_pem_test/tutorial_mqtt_client_pub_pem_test.ino line 318
0x40206818: Print::println(char const*) at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/Print.cpp line 198
0x4020126e: setup() at /Users/user/Documents/Arduino/tutorial_mqtt_client_pub_pem_test/tutorial_mqtt_client_pub_pem_test.ino line 303
0x40207324: loop_wrapper() at /Users/user/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 197
```

earlephilhower, 2020-09-07:
Ah, not a crash but a WDT.

Compile all SSL stuff at 160MHz and retry.  If the certificate (both your private one and the trust anchors) is installed, that is probably the issue.  If you're not setting a trust anchor or `setInsecure()` then you'll still have issues and need to refer to the BearSSL_Validaiton example.

80MHz is too slow to connect to many servers and could result in WDT.  Even at 160MHz, it can take *5 seconds* to do an EC handshake, so make sure your server doesn't drop out on long connect attempts...

dorbae, 2020-09-08:
Thank you for a kind answer. I'm gonna try it and write a comment on the reason why I was failed to connect the TLS server. And also a few days after, I'll close this issue. Thank you 👍

earlephilhower, 2020-09-28:
Closing as no update and it seems that the issue was solved w/the higher speed.

closed 2020-09-28 by earlephilhower