esp8266/Arduino #7582: GPF(p) macro or pinMode causes crash in isr

roberttidey, 2020-09-05:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [x ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12
- Core Version:			7/23/2020
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			espino
- Flash Mode:			qio
- Flash Size: 		4MB
- lwip Variant:  v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	[40Mhz]
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   921600

### Problem Description
Use of GPF(p) macro or pinMode (which uses this macro within an interrupt routine triggers a crash after a few seconds.

I believe this is caused by the commit

https://github.com/esp8266/Arduino/commit/d14419efa216088347e9c9c29864b2c061b95b4f#diff-7dfedad694dac48ffda319d4f65abaa6

which moved the look up table from RAM to EPROM to save 16 bytes.

Sketches which used the construct in an isr worked before OK.

Use of pinMode within an isr is fairly unusual but I have some applications where pins are used bidirectionally and need to be reversed in an isr. Currently I am working around that by using my own macro to reverse direction without using pinMode or GPF(p)

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

int digitState = 0;

void ICACHE_RAM_ATTR testIsr() {
	if(digitState == 0) {
		pinMode(5, INPUT_PULLUP);
		digitState = 1;
	} else {
		pinMode(5, OUTPUT);
		digitState = 0;
	}
	timer1_write(200);
	timer1_enable(TIM_DIV256, TIM_EDGE, TIM_SINGLE);
}

void setup() {
	Serial.begin(115200);
	noInterrupts();
	timer1_isr_init();
	timer1_attachInterrupt(testIsr);
	timer1_enable(TIM_DIV256, TIM_EDGE, TIM_SINGLE);
	timer1_write(1000);
	interrupts();
}
	
void loop() {
	delay(10);
}
```

### Debug Messages

```
Debug messages go here
```

earlephilhower, 2020-09-10:
Thank you for the MCVE, but `pinMode` was never safe in IRQs.  It may have worked some/most times, but if you check the source it's not in IRAM (which is a requirement for ISRs).

So, depending on what was going on w/the flash you could end up trying to jump to a flash address while the flash was being written/erased, causing a machine hang.

Closing as working-as-expected.

closed 2020-09-10 by earlephilhower

roberttidey, 2020-09-10:
Thanks. pinMode uses macros which compile in-line with the isr and therefore within the cache attribute. It is the address translation table reference change that made it unsafe.

I respect the decision if the desire to save 16 bytes of RAM is more important. I know that one has to be careful with external code in ISRs but had thought that low level basic i/o functions should have been a basic requirement to work.

earlephilhower, 2020-09-10:
> Thanks. pinMode uses macros which compile in-line with the isr and therefore within the cache attribute. It is the address translation table reference change that made it unsafe.

`pinMode` isn't in IRAM, so where the constant array is located is immaterial.
https://github.com/esp8266/Arduino/blob/a460cb793574dd4b055ef4c409ebc57d6394888f/cores/esp8266/core_esp8266_wiring_digital.cpp#L35

Without `ICACHE_RAM_ATTR` on the function it also ends up in flash.  It's not there because `pinMode` is relatively large (several hundred bytes IIRC) and IRAM is a fixed size with a large portion taken up by the Espressif blobs, leaving little for users or the core.

roberttidey, 2020-09-10:
Yes, you are right about the full pinMode as that has to handle all the other cases. I should have clarified that in my apps that do switching between  input and output I just use the macros that set the bits that are different for input and output. I should have used these in the example rather than the simplified one using the full pinMode. The GPF(p) is the macro which broke things due to the relocation of the address look up table and I now use my own version of that which avoids the problem.

I accept that this is a fairly esoteric use case and will continue with my work around.