esp8266/Arduino #7583: ESP32, serial and latencies

Koxx3, 2020-09-06:
** Platform **

    Hardware: [ESP-32]
    Core Version: [1.0.4]
    Development Env: [Arduino IDE]
    Operating System: [Windows]

** Settings in IDE **

    Module: [ESP32 Dev module]
    Flash Mode: [qio]
    Flash Size: [4MB]
    CPU Frequency: [240Mhz]
    Upload Using: [SERIAL]
    Upload Speed: [921600] (serial upload only)

** Problem Description **

Hello,

I need to copy serial data as soon each caracter arrives.
I have much troubles with latencies during reads and writes.

Here is my sketch... pretty simple. I've just added IO output to monitor read/write operations with a logic analyzer.

  ```
  HardwareSerial hwSerCntrlToLcd(1);

    #define SERIAL_CNTRL_TO_LCD_RXPIN 34
    #define SERIAL_CNTRL_TO_LCD_TXPIN 13

    #define PIN_READ_FLAG 33
    #define PIN_WRITE_FLAG 32

    #define BAUD_RATE 1200

    ////// Setup
    void setup() {

      hwSerCntrlToLcd.begin(BAUD_RATE, SERIAL_8N1, SERIAL_CNTRL_TO_LCD_RXPIN, SERIAL_CNTRL_TO_LCD_TXPIN);

      pinMode(PIN_READ_FLAG, OUTPUT);
      pinMode(PIN_WRITE_FLAG, OUTPUT);

    }

    void loop() {
      readHardSerial();

    }

    int readHardSerial() {

      byte var;

      while (hwSerCntrlToLcd.available() > 0) {

        digitalWrite(PIN_READ_FLAG,HIGH);
        var = hwSerCntrlToLcd.read();
        digitalWrite(PIN_READ_FLAG,LOW);

        digitalWrite(PIN_WRITE_FLAG,HIGH);
        hwSerCntrlToLcd.write(var);
        digitalWrite(PIN_WRITE_FLAG,LOW);

      }

    }
```

[image]

First issue : the available() returns only after the full data packet is recevied (see D0 line for source data, and D4 for available() starting to read data), and not when each caracter arrives. Can this be changed ?

Second issue : the write output is very delayed. The latency is so important that the visible data packed sent (D1) is the previous packet (check the second byte in the packet : 24 read / 23 write). The write() operations can be observed on D5. Why is that ?

Thanks

jandrassy, 2020-09-06:
https://github.com/espressif/arduino-esp32

closed 2020-09-06 by devyte

Koxx3, 2020-09-06:
closed ?
i had to modify the uart file from the framework to get it working. this is not a normal behavior...

Pablo2048, 2020-09-06:
Did you read the name of this repository and check it with the name of your ESP? ;-) ...

Koxx3, 2020-09-06:
oh... crap... sorry... posted on the wrong one 🤣