esp8266/Arduino #7587: wemos D1 reset

Ali13776, 2020-09-08:
- Hardware:					[Wemos D1]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows|Ubuntu]

### Problem Description

I use wemos D1, servo motor, nrf 24l01+
I have problem wemosD1 don't work.
this is the problem: ets Jan 8 2013,rst cause:4, boot mode:(3,6)
wdt reset
load 0x4010f000
len 1384
room 16 tail 8 
chksum 0x2d
csum 0x2d v951aeffa ~ld
Hi
S:
How can i solve this???
thank you.
my code is: 
#include <SPI.h>
#include <NRFLite.h>
#include <Servo.h>
NRFLite _radio;
uint8_t _data;

int servo_pin = 5;
Servo fire_servo; 

void setup()
{
  fire_servo.attach(servo_pin);
  pinMode(5, OUTPUT);
  Serial.begin(115200);
 fire_servo.write(0);
 fire_servo.write(20);
 fire_servo.write(0);
  Serial.println("Hi");
  Serial.print("S:");
  Serial.println(_radio.init(2, 2, 8)); // Set this radio's Id = 1, along with its CE and CSN pins
}

void loop()
{
  while (_radio.hasData())
  {
    _radio.readData(&_data);
    Serial.println(_data);
    if (_data == 5) 
    {
      fire_servo.write(90);
    }
  }
}

Tech-TX, 2020-09-09:
The folks at the ESP8266.com are the appropriate place for this. This is an area for issues with the code and libraries,  not general "help me,  my code is broken" problems,  as is clearly stated in the issue template you ignored. 

Your question at ESP8266.com should be something along the line of "why is this causing A WDT reset?" There's a bit of help at readthedocs that explains common WDT causes. If you check the common causes first most folks are more willing to help...

https://arduino-esp8266.readthedocs.io/en/latest/faq/a02-my-esp-crashes.html

devyte, 2020-09-09:
Previous comment is correct. In addition, you have a 3rd party lib, so you need to rule that out first.
Closing due to not compliant.

closed 2020-09-09 by devyte