esp8266/Arduino #7589: Update to GCC 10.1 breaks multitasking lib CoopTask

dok-net, 2020-09-09:
The attached MCVE works up to release 2.7.4, and also works with ESP8266 Arduino Core, master HEAD, if the update to GCC 10.1 is manually reverted to the previous compiler toolchain. I am working with Arduino IDE (actually VisualMicro) on Windows 10 x86_64.

Clean compiling under master HEAD, then running results in, stack decoded:
```
   :Error:2 -> InstructionFetchError: Attempt to execute instructions from an illegal address, or read error when fetching instruction
   :?:::0x40101705:ets_timer_arm_new
   0x40201600 __esp_yield
   0x40201c00 delay_end
   0x4020138b esp_yield
   0x40201c92 __delay
   0x40100190 ets_post
   0x402014a7 loop
   0x4020171c loop_wrapper()
   0x40100c61 cont_wrapper 
```
[Task.ino.txt](https://github.com/esp8266/Arduino/files/5197094/Task.ino.txt)

dok-net, 2020-09-12:
@earlephilhower I've updated the Task.ino to contain only the minimal code, and also the output and stack decoder message.

dok-net, 2020-09-12:
The value of `CoopTaskBase::TaskStackTop = 0x3ffef23c` in GCC 10.1. With the compiler from 2.7.4, it's `TaskStackTop = 0x3ffef1d4`.

earlephilhower, 2020-09-12:
Your code seems to be trying to jump to an address in RAM which is going to make the processor very unhappy (EPC=0x3f...).

That said, longjmp/setjmp seem fine with focused tests:
````
#include <stdio.h>      /* printf, scanf */
#include <stdlib.h>     /* exit */
#include <setjmp.h>     /* jmp_buf, setjmp, longjmp */

jmp_buf env;

void setup() {
  int val;
  Serial.begin(115200);
  Serial.printf("\n\nHello\n");
  val = setjmp (env);
  if (val) {
    Serial.printf("Error %d happened\n",val);
    while (1) delay(100);
    return;
  }

  Serial.printf("\nDo some stuff\n");

  return ;
}

void loop() {
  static int cnt=0;
  if (cnt++ > 10)
    longjmp (env,101);   /* signaling an error */
}
````
gives
````
Hello

Do some stuff
Error 101 happened
````

So, I'm disinclined to think it's setjmp/longjmp related.  The setjmp is run during setup(), and the longjmp to it is run 10x after loop() has been called (i.e. so add'l OS context switches/etc.).

I would recommend running this under GDB and single-step to get to the bottom of things.

Random ideas:

Could it be trying to jump to a `std::function` (which is a struct in RAM with a pointer to flash) somewhere?

Looking at the crash in my setup, I get a slightly different address w/your code:
````
Exception (2):
epc1=0x3ffee3f8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffee3f8 depc=0x00000000
````

Looking at the OBJDUMP -T output I see that 0x3ffee3f8 == a global variable.  So, the code is trying to jump to a RAM address defined as the loop queue:
````
....
3ffee3e6 l     O .bss   0000000e _ZL19ets_intr_lock_stack
3ffee3f4 l     O .bss   00000004 _ZL23s_cycles_at_yield_start
3ffee3f8 l     O .bss   00000008 _ZL12s_loop_queue
3ffee400 g     O .bss   0000001c resetInfo
3ffee41c g     O .bss   00000004 __stack_chk_guard
....
````

It is also strangely close to the end of the stack.  Accessing any stack-local vars which no longer exist?

dok-net, 2020-09-12:
@earlephilhower Please check my comment about task stack top and the address of a stack object in the loop. Is alloca broken?
Also, one time setjmp/longjmp obviously works in my MCVE, the issue arises on the next time longjmp'ing into the task.

dok-net, 2020-09-12:
hang on, copy&paste confusing in a haste...

dok-net, 2020-09-12:
Whatever GCC 10.1 does with the auto object, it's not looking good to me.
I've zero experience running GDB on ESP8266.

earlephilhower, 2020-09-12:
Longjmp will reset the stack and all regs to their values, so it's not possible to multiply longjmp to the same spot AIUI.

In my small sample code, doing a `return` from inside the longjmp destination would end up popping the stack at whatever offset it was at when the setjmp() was called initially.  That's absolutely illegal because the stack is no longer valid at that address, it was reclaimed on the first return after setup().

earlephilhower, 2020-09-12:
`auto` is syntactic sugar.  It might be choosing a different type, but that's about it...nothing magic once the type is inferred from the RHS.

dok-net, 2020-09-12:
I don't think there's any mistake in my use of setjmp, longjmp. Just read it, I knew/know not to longjmp multiple times to the same setjmp-target. It gets reset every time. At least that's how it's intended, I you find otherwise, mea culpa.

dok-net, 2020-09-12:
@earlephilhower argh, "auto" was my shorthand for automatic object, not latest generation C++ syntax...

earlephilhower, 2020-09-12:
> I don't think there's any mistake in my use of setjmp, longjmp. Just read it, I knew/know not to longjmp multiple times to the same setjmp-target. It gets reset every time. At least that's how it's intended, I you find otherwise, mea culpa.

Actually, it kind of smells like that to me.  If a `return` was called after a `longjmp` it will pop the stack at some address that it was at before...and, it's quite possible the loop_queue variable was placed on the stack in the core code to do a fcn call or something.  So, the `return` would `jmp(pop(old-sp)) ` and you'd try to jump to the global variable used.

earlephilhower, 2020-09-12:
As a simple check, can you verify if in your crash the address being jumped to is also `loop_queue`?  `objdump -t | sort -k1` should make it easy to find

dok-net, 2020-09-12:
just a very early finding, but it seems that the optimizer is more aggressive...

dok-net, 2020-09-12:
OK... `std::atomic_thread_fence(std::memory_order_acq_rel);` after `alloca` does the trick.
Now... it is OK that `alloca()` gets optimized around with in the first place.?

earlephilhower, 2020-09-12:
That's up to the GCC guys.  I would say yes because alloca semantics are completely non-standard and the compiler can do whatever it pleases....no POSIX or compatibility requirements.

dok-net, 2020-09-12:
It's obviously a breaking change for the ESP8266, no matter what. I'm OK with it now that I know and fix this. If nobody else has / is going to have any trouble with this, fine :-)

earlephilhower, 2020-09-12:
I think you're the only person using alloca and set/longjmp in this way.  I'll close this for now since we understand the issue, it's GCC and related to a non-standard call, and there's a workaround in your code.  Good joint debug!

closed 2020-09-12 by earlephilhower