esp8266/Arduino #7593: Exception 3: LoadStoreError: Processor internal physical address or data error during load or store

TanayBhalani, 2020-09-12:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[ESP8266EX]
- Core Version:			[2.5.2=20502000]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[dout]
- Flash Size: 		        [4MB]
- lwip Variant:                  [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	        [80Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:              [115200] (serial upload only)

### Problem Description

The chip goes to LoadStoreError Exception and restarts, few times, not random though

Currently cannot share code, let me know if its the only way to solve it
Purpose is to read temperature from Dallas ds18b20 sensor and upload to server,
and also fetch a minimum temperature set by user (which changes many times),
and make some small calculation on base of this data and upload to server.
All of this happens with API

Exception is not coming always, but still not random

Before fetching data i check for availability of connection, i feel it is around this block
```
  #define usid 123456
  #define connection_led 16
  #define get_set_temperature  "/api/get_set_temperature.php?usid=" + String(usid) 
  const char *endpointBaseUrl = "url.com";
  const uint16_t httpPort = 80;

  Serial.print("connecting to ");
  Serial.println(endpointBaseUrl);

  if (!client.connect(endpointBaseUrl, httpPort))
  {
    Serial.println("connection failed");
    //  Set connection_led to ON because in error state
    digitalWrite(connection_led, LOW);
    return;
  }

  Serial.print("Requesting URL: ");
  Serial.println(get_set_temperature);

  if (client.connected())
  {
    client.print(String("GET ") + get_set_temperature + " HTTP/1.1\r\n" +
                 "Host: " + endpointBaseUrl + "\r\n" +
                 "Connection: close\r\n\r\n");
  }
  delay(50);

[... 188 lines cut ...]

Decoding stack results
0x40231c50: ip4_output_if_opt at core/ipv4/ip4.c line 820
0x402165ec: _umm_free(void*) at C:\Users\tanay\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1304
0x40216902: get_unpoisoned(void*) at C:\Users\tanay\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 950
0x40216d08: free(void*) at C:\Users\tanay\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1751
0x402326f8: mem_free at core/mem.c line 237
0x40229952: memp_free at core/memp.c line 447
0x40229a54: tcp_free at core/tcp.c line 217
0x4022a017: tcp_abandon at core/tcp.c line 622
0x402165ec: _umm_free(void*) at C:\Users\tanay\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1304
0x40216d14: free(void*) at C:\Users\tanay\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1764
0x4022a040: tcp_abort at core/tcp.c line 641
0x4022ccf9: tcp_input at core/tcp_in.c line 497
0x40214659: malloc(size_t) at C:\Users\tanay\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\heap.cpp line 95
0x40216af8: umm_malloc(size_t) at C:\Users\tanay\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1685
0x40214659: malloc(size_t) at C:\Users\tanay\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\heap.cpp line 95
0x40230000: dhcp_handle_ack at core/ipv4/dhcp.c line 641
0x402319ed: ip4_input at core/ipv4/ip4.c line 710
0x402165ec: _umm_free(void*) at C:\Users\tanay\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1304
0x40229811: ethernet_input_LWIP2 at netif/ethernet.c line 188
0x40229647: esp2glue_ethernet_input at glue-lwip/lwip-git.c line 441
0x40248a52: ethernet_input at glue-esp/lwip-esp.c line 363
0x40248a63: ethernet_input at glue-esp/lwip-esp.c line 371
```

TanayBhalani, 2020-09-12:
I update to latest 2.7.4, still facing error, have updated the exception occurred using 2.7.4 in the issue description

earlephilhower, 2020-09-12:
Without a complete MCVE there's little we can do.  Your traces look like your code is either corrupting memory or running out of RAM.   I suggest you enable all debug options and check for OOM errors.

Please use https://gitter.im/esp8266/Arduino for further general assistance.

closed 2020-09-12 by earlephilhower

TanayBhalani, 2020-09-15:
@earlephilhower I found that i was using enum to store/extract response but was mapping it to an int type variable.
Though my enum was mapping 1 and 2, changing the type for mapping variable solved the issue

Something like this
```cpp
typedef enum Status
{
  ON = 1,
  OFF = 2
};

int active_status = ON;
```
I replaced it with
```cpp
typedef enum Status
{
  ON = 1,
  OFF = 2
};

Status active_status = ON;
```

i wonder even using c++17 on vscode, intellisense could not find this issue