esp8266/Arduino #7597: Unstable AP_Mode, terminated after connection & Exception 29 Error

Devilscave, 2020-09-13:
#7384 # Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below. (i hope)

#### Platform

- Hardware:			        [ESP-12E]
- Core Version:			[latest git hash or date]  ??????
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Nodemcu]
- Flash Mode:			[fixed by IDE]
- Flash Size: 		        [4MB]
- lwip Variant:                  [v2_lower_Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	        [40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:              [921600]

### Problem Description

I'm using a sketch that was originally written for WiFi_STA, in WiFi_AP mode. This is the "webinterface" example from the "ws2812fx" library. Now the problem is that it works but after a short time the connection breaks and then takes 2-3 minutes until the ESP does a reset by itself with the message "Exception 29".
I have tried this with the "ESP Exception Decoder" tool to find out why the ESP is doing this and where exactly the error is. Unfortunately without success, but many references go to "core_esp8266_main.cpp" I have attached my error output, can someone tell me a little more about what could be the cause? The sketch also includes an "index.html.cpp" and "main.js.cpp" file. You can simply load this via the example and then insert my sketch into the original.
I also asked by "ws2812fx", but got no answer except for "AP_Mode" which seems to be very unstable. (OHH really?)

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <WS2812FX.h>
 
extern const char index_html[];
extern const char main_js[];
 
#ifndef APSSID
#define APSSID "ws2812fx"                                                     // WLan Name des ESP
#define APPSK  "ledmagic"                                                     // WLan Password    ### NOT NEEDED ###
#endif
 
// QUICKFIX...See https://github.com/esp8266/Arduino/issues/263
#define min(a,b) ((a)<(b)?(a):(b))
#define max(a,b) ((a)>(b)?(a):(b))
 
#define LED_PIN 5                                // LED PIN
#define LED_COUNT 100                      // LED Menge
 
#define WIFI_TIMEOUT 30000              // checks WiFi every ...ms. Reset after this time, if WiFi cannot reconnect.
#define HTTP_PORT 80
 
#define DEFAULT_COLOR 0xFF5900
#define DEFAULT_BRIGHTNESS 128
#define DEFAULT_SPEED 1000
#define DEFAULT_MODE FX_MODE_STATIC
 
const char *ssid = APSSID;
const char *password = APPSK;
 
unsigned long auto_last_change = 0;
unsigned long last_wifi_check_time = 0;
[... 177 lines cut ...]
0x402198e0: etharp_output_to_arp_index at core/ipv4/etharp.c line 770
0x4010075b: umm_free_core(void*) at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 351
0x40100a14: malloc(size_t) at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 552
0x4021be20: mem_malloc at core/mem.c line 210
0x40216759: tcp_create_segment at core/tcp_out.c line 190
0x401001f8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_main.cpp line 177
0x40211ab1: glue2esp_linkoutput at glue-esp/lwip-esp.c line 301
0x401001f8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_main.cpp line 177
0x401001f8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_main.cpp line 177
0x401001f8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_main.cpp line 177
0x401001f8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_main.cpp line 177
0x401001f8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_main.cpp line 177
0x401001f8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_main.cpp line 177
0x40100277: millis() at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_wiring.cpp line 185
0x40100274: millis() at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_wiring.cpp line 185
0x402092ec: loop_wrapper() at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_main.cpp line 192
0x402092ec: loop_wrapper() at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_main.cpp line 192
0x401002a4: millis() at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_wiring.cpp line 188
0x40205902: WiFiServer::available(unsigned char*) at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\libraries\ESP8266WiFi\src\WiFiServer.cpp line 140
0x40205b1e: WS2812FX::service() at C:\Users\Devilscave\Documents\Arduino\libraries\WS2812FX\src\WS2812FX.cpp line 72
0x4020aea8: std::_Function_handler ::_M_invoke(std::_Any_data const&) at c:\users\devilscave\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2/functional line 2069
0x401001f8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_main.cpp line 177
0x40100219: esp_schedule() at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_main.cpp line 125
0x40209309: loop_wrapper() at C:\Users\Devilscave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_main.cpp line 199
```

Tech-TX, 2020-09-13:
Looks to me like the stack blew up, but I don't know the built-in WiFi libraries well enough to know whether your example is valid.  

Your MCVE shouldn't include third-party libraries, as that only muddies the waters.

earlephilhower, 2020-09-13:
The MCVE you attached isn't possible for others to use, @Devilscave , as @Tech-TX  said.  We need MCVEs without external dependencies so the team can actually run them.

Your problem is most likely an Out of Memory (OOM) error because your code is .  Enable all the debug options in the IDE and you'll get more info, including a note where the OOM happened.  The EXCADDR of 0x18 means some structure was attempted to be allocated, got a NULL, and then the app ignore the NULL and tried to access a field in the structure at offset NULL+0x18.

Also you should at least upgrade to 2.7.4 from 2.7.2 because there were several important fixes.

closed 2020-09-13 by earlephilhower