esp8266/Arduino #7598: Exception (29) on standart Example Node MCU

kofe88, 2020-09-14:
#### Platform

- Hardware:					ESP8266
- Core Version:			2.7.4
- Development Env:		Arduino IDE 1.8.13
- Operating System:		Windows

### Settings in IDE
[image]

### Problem Description

Hello there! Tried on 6 modules, there is a memory leak, then exception 29 and reboot. I tried Node MCU (CP210x and CH340) and Wemos D1 Mini, the result is the same everywhere. I load a standard unmodified example, somewhere in half an hour all the memory leaks (judging by println(ESP. getFreeHeap (), DEC)). Ask for help.

 Sketch

`/*
  ESP8266 Blink by Simon Peter
  Blink the blue LED on the ESP-01 module
  This example code is in the public domain

  The blue LED on the ESP-01 module is connected to GPIO1
  (which is also the TXD pin; so we cannot use Serial.print() at the same time)

  Note that this sketch uses LED_BUILTIN to find the pin with the internal LED
*/

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);     // Initialize the LED_BUILTIN pin as an output
  Serial.begin(115200);
}

// the loop function runs over and over again forever
void loop() {

  digitalWrite(LED_BUILTIN, LOW);   // Turn the LED on (Note that LOW is the voltage level
  // but actually the LED is on; this is because
  // it is active low on the ESP-01)
  delay(1000);                      // Wait for a second
  digitalWrite(LED_BUILTIN, HIGH);  // Turn the LED off by making the voltage HIGH
  delay(2000);                      // Wait for two seconds (to demonstrate the active low LED)
  Serial.println(ESP.getFreeHeap(),DEC);
}`

### Debug Messages

```
13:18:15.173 -> 51712
13:18:18.175 -> 51712
13:18:21.193 -> 51712
13:18:24.199 -> 51712
13:18:27.179 -> 51712
13:18:30.167 -> 51712
13:18:33.178 -> 51712
13:18:36.178 -> 51712
13:18:39.184 -> 51712
13:18:42.195 -> 51712
13:18:45.171 -> 51712
13:18:48.188 -> 51712
13:18:51.190 -> 51712
13:18:54.169 -> 51712
13:18:57.177 -> 51712
13:19:00.192 -> 51712
13:19:03.201 -> 51712
13:19:06.183 -> 51712
13:19:09.195 -> 51712
13:19:12.174 -> 51712
13:19:15.186 -> 51712
13:19:18.182 -> 51688
13:19:21.191 -> 51688
[... 849 lines cut ...]
DATA   : 1248  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 712   ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 25128 )         - zeroed variables      (global, static) in RAM/HEAP 
Скетч использует 262572 байт (25%) памяти устройства. Всего доступно 1044464 байт.
Глобальные переменные используют 27088 байт (33%) динамической памяти, оставляя 54832 байт для локальных переменных. Максимум: 81920 байт.
esptool.py v2.8
Serial port COM4
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: a4:cf:12:d5:af:3c
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 266720 bytes to 196624...
Wrote 266720 bytes (196624 compressed) at 0x00000000 in 17.4 seconds (effective 122.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

```

d-a-v, 2020-09-14:
I tried and can't reproduce.

My esp automatrickally connects to the last network it was connected to by default.
I played with nmap in order to shake the internal wifi library.
I also tried with `WiFi.mode(WIFI_OFF);` and `WiFi.mode(WIFI_STA); WiFi.begin("myssid","wrongpassword");` and also `WiFi.mode(WIFI_STA); WiFi.begin("garbage","wrongpassword");` with same result (stable heap, wifi not connected).

Can you enable debug on Serial and restart so we can also check what firmware you are running.
Here are my logs, time is not accurate when I lock/unlock the terminal to make my copy paste but all lines are there, 3 seconds per line.

I suggest the option "Erase all flash" (once per board is sufficient), debug option enabled.

```
15:39:19.706698638 SDK:2.2.2-dev(38a443e)/Core:2.7.4=20704000/lwIP:IPv6+STABLE-2_1_2_RELEASE/glue:1.2-30-g92add50/BearSSL:5c771be
15:39:19.813383303 scandone
15:39:19.819211668 state: 0 -> 2 (b0)
15:39:19.834581889 state: 2 -> 3 (0)
15:39:19.843977047 state: 3 -> 5 (10)
15:39:19.854552229 add 0
15:39:19.863400138 aid 2
15:39:19.870822165 cnt
15:39:19.877253455 connected with myssid, channel 11
15:39:19.885594822 dhcp client start...
15:39:19.926010796 ip:10.0.1.154,mask:255.255.255.0,gw:10.0.1.254
15:39:20.772032445 ip:10.0.1.154,mask:255.255.255.0,gw:10.0.1.254
15:39:20.780528201 ip:10.0.1.154,mask:255.255.255.0,gw:10.0.1.254
15:39:21.664735786 ip:10.0.1.154,mask:255.255.255.0,gw:10.0.1.254
15:39:22.663876452 ip:10.0.1.154,mask:255.255.255.0,gw:10.0.1.254
15:39:22.670734776 ip:10.0.1.154,mask:255.255.255.0,gw:10.0.1.254
15:39:22.691475144 50056
15:39:25.690566228 50056
15:39:28.692618098 50056
15:39:29.834470409 pm open,type:2 0
15:39:31.691606917 49296
15:39:34.692470679 49152
15:39:37.693200846 48144
15:39:40.693713940 49008
15:39:43.695580545 49296
15:39:46.695360643 49296
15:39:49.695139310 49296
15:39:57.205535229 48864
15:39:57.211973391 49296
15:39:58.695925958 49152
15:40:01.698384108 49152
15:40:04.699381507 49152
15:40:07.698640071 49296
15:40:10.696783527 49296
15:40:13.697588239 49296
15:40:16.698409065 49296
15:40:19.697968690 49296
15:40:22.700179695 49296
15:40:25.699348269 49296
15:40:28.699861666 49296
15:40:31.700279585 49296
15:40:34.701886810 49152
15:40:37.702762377 49296
15:40:40.703056251 49296
15:40:40.703056251 49296
15:40:43.702787879 49296
15:40:46.702149599 49296
15:40:49.702378344 49296
15:40:52.704307524 49296
15:40:55.705049970 49296
15:40:58.705416619 49296
15:41:01.709990994 49296
15:41:04.705623607 49296
15:41:07.706756831 49296
15:41:10.707811279 49296
15:41:13.704398013 48720
15:41:16.711494143 49296
15:41:19.709592550 49152
[... 128 lines cut ...]
15:48:05.371903260 49296
15:48:05.379555515 49296
15:48:05.386824118 49296
15:48:05.395359475 49296
15:48:05.402804179 49296
15:48:05.410831350 49296
15:48:05.417186889 49296
15:48:07.759228154 49296
15:48:10.760671499 49296
15:48:13.760323800 49296
15:48:16.760382047 49296
15:48:19.763239278 49296
15:48:22.763115283 49296
15:48:25.763028171 49296
15:48:28.765010714 49296
15:48:31.764429588 49296
15:48:34.764180737 49296
15:48:37.767810396 49296
15:48:40.766477579 49296
15:48:43.763805592 49296
15:48:46.762693911 49296
15:48:49.765237590 49296
15:48:52.764951174 49296
15:48:55.766439980 49296
```

Tech-TX, 2020-09-15:
Unless there's a good reason, I run my code with **WiFi.persistent(false);** so that it's not hammering the Flash each time it reconnects.  As a result, I don't have a stored WiFi configuration in the module I'm testing now, and there aren't any open routers around here it can connect to on it's own.  WiFi is completely disconnected.

When I run his test code, I get exactly the same free heap (53528), even after running for 30 minutes.  I'd suspect he has a marginal WiFi connection, and it's making and breaking frequently using the stored configuration.  That _shouldn't_ cause a heap leak, but I don't know what happens if the connection breaks while it's being established; maybe there's a small leak there that's being exaggerated over time and multiple reconnects.

I had a project a year ago that ran continuously for months.  It would idle for 25 minutes, then power up a sensor, take readings for 5 minutes, power down the sensor, turn on WiFi and squirt the results to ThingSpeak, then turn the modem off and idle for another 25 minutes.  That ran for several months with no memory leak, so the core files and WiFi libraries are working correctly if your setup and code is valid.  I _did_ have memory leaks during development, but I fixed my newbie mistakes in the code and it ran OK.

kofe88, 2020-09-15:
> I tried and can't reproduce.
>
> My esp automatrickally connects to the last network it was connected to by default.
> I played with nmap in order to shake the internal wifi library.
> I also tried with [...]

Hello there! Your recommendation helped with the Blink example, but the same nonsense happens with the WebServer example. Memory is gradually leaking, the recommendation above did not help. Turned on debugging, I apply output from the terminal.

## Sketch
```
/*
   Copyright (c) 2015, Majenko Technologies
   All rights reserved.

   Redistribution and use in source and binary forms, with or without modification,
   are permitted provided that the following conditions are met:

 * * Redistributions of source code must retain the above copyright notice, this
     list of conditions and the following disclaimer.

 * * Redistributions in binary form must reproduce the above copyright notice, this
     list of conditions and the following disclaimer in the documentation and/or
     other materials provided with the distribution.

 * * Neither the name of Majenko Technologies nor the names of its
     contributors may be used to endorse or promote products derived from
     this software without specific prior written permission.

   THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS" AND
   ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE IMPLIED
   WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE
   DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE LIABLE FOR
   ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES
   (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES;
   LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON
   ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT
   (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE OF THIS
   SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
*/

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

#ifndef STASSID
#define STASSID "MikroTik-2ghz"
#define STAPSK  "2093301i"
#endif

const char *ssid = STASSID;
const char *password = STAPSK;

ESP8266WebServer server(80);

const int led = 13;

void handleRoot() {
  digitalWrite(led, 1);
  char temp[400];
  int sec = millis() / 1000;
  int min = sec / 60;
  int hr = min / 60;

  snprintf(temp, 400,

           "<html>\
  <head>\
    <meta http-equiv='refresh' content='5'/>\
    <title>ESP8266 Demo</title>\
[... 1437 lines cut ...]
09:15:12.594 -> :urd 1, 46, 2
09:15:12.594 -> :urd 1, 46, 3
09:15:12.594 -> :urd 2, 46, 4
09:15:12.594 -> :urd 2, 46, 6
09:15:12.594 -> :urd 2, 46, 8
09:15:12.594 -> :urd 2, 46, 10
09:15:12.594 -> [MDNSResponder] _parseMessage: Done (Succeeded after 25 ms, ate 0 bytes, remaining 18488)
09:15:12.594 -> 
09:15:17.572 -> :urch 74, 46
09:15:17.572 -> [MDNSResponder] _callProcess (1283901, triggered by: 192.168.0.11)
09:15:17.572 -> [MDNSResponder] _parseMessage (Time: 1283902 ms, heap: 17240 bytes, from 192.168.0.11(5353), to 224.0.0.251(5353))
09:15:17.572 -> :urd 2, 46, 0
09:15:17.572 -> :urd 1, 46, 2
09:15:17.572 -> :urd 1, 46, 3
09:15:17.572 -> :urd 2, 46, 4
09:15:17.572 -> :urd 2, 46, 6
09:15:17.572 -> :urd 2, 46, 8
09:15:17.572 -> :urd 2, 46, 10
09:15:17.572 -> [MDNSResponder] _parseMessage: Done (Succeeded after 25 ms, ate 0 bytes, remaining 17240)
09:15:17.606 -> 
09:15:18.428 -> 17240

```
## Settings in Arduino IDE
[image]

d-a-v, 2020-09-16:
You ***must not*** do that:

```
void loop(void) {
  server.handleClient();
  MDNS.update();
  delay(10000);
  Serial.println(ESP.getFreeHeap(),DEC);
}
```

The `loop()` function must not be blocked by any waiting call when background process is required (like "somelib.**handle**something()", "otherlib.**update**()").
Instead, you can do that:

```
#include <PolledTimeout.h>                   // that should be part of Arduino.h in esp8266/Arduino
using esp8266::polledTimeout::periodicMs;    // but it is not yet.

periodicMs itIsTimeToShowSomeInsight(10000); // is true once every 10s

void loop(void) {
  server.handleClient();
  MDNS.update();
  if (itIsTimeToShowSomeInsight) {
    Serial.println(ESP.getFreeHeap(),DEC);
  }
}
```

devyte, 2020-11-22:
Closing due to lack of response in over 2 months.

closed 2020-11-22 by devyte