esp8266/Arduino #7601: Exception (3) on standard example WiFiAccessPoint

Exocet22, 2020-09-16:
### Platform

Hardware: ESP8266 (original AI-Thinker ESP12-F)
Core Version: 2.7.4
Development Env: Arduino IDE 1.8.13
Operating System: Mac OS (Catalina)

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			qio
- Flash Size: 			4MB/1MB
- lwip Variant:  		 v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		160MHz
- Upload Using:		SERIAL
- Upload Speed:   		460800

### Problem Description

Hi everybody.

Exception (3) occurs on the standard example every time when Safari makes sudden HNAP calls(Home Network Administration Protocol) on the server. These HTTP calls made by Safari are a little bit hectic and sometimes we need to wait several minutes before receiving them. Chrome doesn't seem to send these kind a HTTP calls.

Thanks in advance for the help.

### Sketch

See standard example : ESP8266WiFi/examples/WiFiAccessPoint/WiFiAccessPoint.ino

### Debug Messages

17:26:33.740 -> SDK:2.2.1(cfd48f3)/Core:2.7.3-3-g2843a5ac=20703003/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-30-g92add50/BearSSL:5c771be
17:26:33.740 -> 
17:26:33.740 -> Configuring access point...bcn 0
17:26:33.852 -> del if1
17:26:33.852 -> usl
17:26:33.852 -> add if1
17:26:33.852 -> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
17:26:33.852 -> bcn 100
17:26:33.852 -> AP IP address: 192.168.4.1
17:26:33.852 -> HTTP server started
17:27:59.594 -> add 1
17:27:59.628 -> aid 1
17:27:59.628 -> station: 3c:22:fb:3f:d0:0f join, AID = 1
17:28:41.619 -> New client
17:28:41.619 -> request: GET /this/is/a/test/ HTTP/1.1
17:28:41.619 -> method: GET url: /this/is/a/test/ search: 
17:28:41.619 -> headerName: Host
17:28:41.619 -> headerValue: 192.168.4.1
17:28:41.619 -> headerName: Upgrade-Insecure-Requests
17:28:41.619 -> headerValue: 1
17:28:41.619 -> headerName: Accept
17:28:41.619 -> headerValue: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
17:28:41.619 -> headerName: User-Agent
17:28:41.619 -> headerValue: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_6) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/13.1.2 Safari/605.1.15
17:28:41.654 -> headerName: Accept-Language
17:28:41.654 -> headerValue: fr-fr
17:28:41.654 -> headerName: Accept-Encoding
17:28:41.654 -> headerValue: gzip, deflate
17:28:41.654 -> headerName: Connection
17:28:41.654 -> headerValue: keep-alive
17:28:41.654 -> args: 
17:28:41.654 -> args count: 0
17:28:41.654 -> args: 
17:28:41.654 -> args count: 0
17:28:41.654 -> Request: /this/is/a/test/
17:28:41.654 -> Arguments: 
17:28:41.654 -> final list of key/value pairs:
[... 428 lines cut ...]
0x40100170: ets_post(uint8, ETSSignal, ETSParam) at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x4020b0b8: _printf_i at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c line 246
0x4020f465: glue2esp_linkoutput at glue-esp/lwip-esp.c line 301
0x401008cd: check_poison_neighbors(uint16_t) at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/umm_malloc/umm_local.c line 71
0x4020f6f2: new_linkoutput at glue-lwip/lwip-git.c line 260
0x4020fb10: ethernet_output at netif/ethernet.c line 312
0x4010087e: check_poison_neighbors(uint16_t) at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/umm_malloc/umm_local.c line 59
0x40100170: ets_post(uint8, ETSSignal, ETSParam) at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x40100170: ets_post(uint8, ETSSignal, ETSParam) at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x40100170: ets_post(uint8, ETSSignal, ETSParam) at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x40100170: ets_post(uint8, ETSSignal, ETSParam) at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x40100170: ets_post(uint8, ETSSignal, ETSParam) at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x402047e2: WiFiClient::connected() at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WiFi/src/WiFiClient.cpp line 338
0x40208d69: WiFiClient::operator bool() at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WiFi/src/WiFiClient.cpp line 349
0x4020479c: WiFiClient::connected() at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WiFi/src/WiFiClient.cpp line 333
0x40206e9c: loop_wrapper() at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 189
0x40100bc0: umm_poison_free_fl(void*, char const*, int) at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/umm_malloc/umm_local.c line 149
0x401002b4: free(void*) at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/heap.cpp line 259
0x4020548e: WiFiServer::available(unsigned char*) at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WiFi/src/WiFiServer.cpp line 140
0x40204a87: ClientContext::unref() at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WiFi/src/include/ClientContext.h line 128
0x40208d69: WiFiClient::operator bool() at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WiFi/src/WiFiClient.cpp line 349
0x4020373d: esp8266webserver::ESP8266WebServerTemplate ::handleClient() at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WebServer/src/ESP8266WebServer-impl.h line 312
0x40100170: ets_post(uint8, ETSSignal, ETSParam) at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x40100191: esp_schedule() at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 125
0x40206ec5: loop_wrapper() at /Users/admin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 199

mhightower83, 2020-09-16:
Try your test running with Flash set to DIO instead of  QIO.

I have several of these boards and they are only stable in DIO mode, in contrast, the [Ai Thinker's specification for the ESP-12-F has a photo showing QIO](https://docs.ai-thinker.com/_media/esp8266/docs/esp-12f_product_specification_en.pdf#page=4). Check the back of your module's PCB Antenna area for silkscreen markings, mine is marked with `ESP-12-F    DIO L4`. Ai Thinker's photo shows `ESP-12-F    QIO L4`. As far as I can see, Ai Thinker specification's text does not indicate which mode to use.

Exocet22, 2020-09-16:
Thanks for your reply.

Unfortunately, I did the test and all configurations failed : DOUT, DIO, QOUT and QIO.
My ESP-12F is marked "ESP-12-F L4". No QIO or DIO.

I'll continue to investigate with WireShark. It also fails in mode WIFI_AP_STA.

d-a-v, 2020-09-16:
Then HTTP captured protocol indeed will hopefully help.

We also have now an internal packet capture facility, check the provided  `Netdump` library (git version, post 2.7.4).

Jason2866, 2020-09-17:
From which manufactor is the flash chip?
I have modules with PUYA chips. Some work, some not.
The issues i have with the bad ones are weird. 
It took me a long time until i recognized everything is okay it is "just" the flash chip

Exocet22, 2020-09-17:
This is ESP-12-F bought directly to AI-Thinker.
I'll try to upload slower and see what happens.

I'm currently adding the packet capture to the code to feed the discussion.

Exocet22, 2020-09-17:
Uploading at lower speed did not affect the result.
See logs attached.

[Logs_HNAP.txt](https://github.com/esp8266/Arduino/files/5241667/Logs_HNAP.txt)

Thanks.

Jason2866, 2020-09-18:
Still not answered which flash chip is used in the module. From whom build or bought is not a info from interest.

devyte, 2020-09-18:
cc @ChocolateFrogsNuts @Tech-TX

mhightower83, 2020-09-18:
Since you are buying bare AI-Thinker modules (not a DEV board), what do you use for a power source?

On new designs, I like to rule out possible power supply issues, before going crazy with a problem. Power issues can create a lot of different and unexpected symptoms that are difficult to associate with power. Part of the issue with the ESP8266 is it can go from sipping 60ma to gulping 300ma in an instant. That is a 400% load increase. A switch-mode power supply usually doesn't have that kind of _Transient Load_ response. All of the IoT designs that I have seen, pass the switch-mode power supply output through an LDO regulator to feed the ESP8266.

Exocet22, 2020-09-19:
I use a classic LDO regulator to power the module. The source can draw easily up to 1000 mA. I've already checked if a voltage drop can cause this issue but did not find one.

Monday, I'll bypass the LDO and use my laboratory power supply to see if it helps. In the meantime, I'll continue my investigations and keep you informed.

Thanks for the help.

Jason2866, 2020-09-20:
My bet is the XMC flash chip as trouble maker.

Exocet22, 2020-09-21:
Still the same issue with power bypassed by a laboratory power supply. Tested on 10 modules (same batch coming from AI-Thinker). No glitches, all good on power supply.

Now, I'll remove the flash chip on some modules and compare. Any suggestion for a good flash ?
Thanks.

Jason2866, 2020-09-21:
Never had issues with winbond flash chips

Exocet22, 2020-09-21:
OK, let's go with that. I'll keep you posted after the change.

In the meantime, I noticed that the problem happens every time on a DHCP incoming call after the HNAP calls.
It happens also on all the 10 modules. Whatever the flash mode and frequency.

Jason2866, 2020-09-21:
We will know more after you did swap the flash.

Exocet22, 2020-09-26:
I changed the flash chip but it we still have the same issue. after looking at a lot of crash logs, it always crashes at the same line in esp-dhcpserver.c (line 626).

A newer version of Safari was also released but it did not change this behavior.

Fortunately, no crashes with Chrome on MacOS because it does not implement HNAP.

I try now to make HNAP calls from another "more controlled" source than Safari to try some debugging.

d-a-v, 2020-09-26:
I understand that running standard example : `ESP8266WiFi/examples/WiFiAccessPoint/WiFiAccessPoint.ino` and accessing the esp using current macOS's Safari is enough to try and reproduce this issue.
Am I right ?

Exocet22, 2020-09-26:
Correct.

Every time with this configuration. Sometimes it takes several minutes to get the HNAP calls, but they will arrive and crash the module.

d-a-v, 2020-09-26:
To restrict traffic to AP and have packet capture, I added in the example
```
#include <Netdump.h>
NetCapture::Netdump nd;
...
  // in setup():
  WiFi.mode(WIFI_AP);
  nd.printDump(Serial, NetCapture::Packet::PacketDetail::FULL, [](NetCapture::Packet p) { return true; });
```

Safari on iPad macOS 12.4.8 did not trigger this bug after several minutes.
I will try with an apple laptop.

Exocet22, 2020-09-26:
I try and keep you informed.

No issue found on iOS (iPhone & iPad). Probably a different implementation of Safari.
I have a Mac Book Pro (2020) / Catalina (Version 10.156) / Safari (Version 14.0  - 15610.1.28.1.9, 15610).

Exocet22, 2020-09-26:
Same crash...
See logs attached.

[logs.txt](https://github.com/esp8266/Arduino/files/5287087/logs.txt)

d-a-v, 2020-09-27:
I'm sorry to not be able to reproduce, using a macmini, updated to latest macOS and safari.

That's 580 seconds after esp boot:

```
request: GET / HTTP/1.1
method: GET url: / search:  keepAlive=: 1
headerName: Host
headerValue: 192.168.4.1
headerName: Upgrade-Insecure-Requests
headerValue: 1
headerName: Accept
headerValue: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
headerName: User-Agent
headerValue: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15
headerName: Accept-Language
headerValue: fr-fr
headerName: Accept-Encoding
headerValue: gzip, deflate
headerName: Connection
headerValue: keep-alive
```

Would there be another way to reproduce the HNAP request than by using safari ?

d-a-v, 2020-09-27:
dhcp server is brought to main repository in PR #6680.
Would you be able to test with this PR (that I just updated) ? You need first to use [the git version of this core](https://arduino-esp8266.readthedocs.io/en/latest/installing.html#using-git-version).
Or you can install the [alpha release v0.0.2](https://d-a-v.github.io/esp8266/Arduino/index.html) with the arduino IDE, that already includes this PR.

Exocet22, 2020-09-28:
Sometimes, it takes about 15 minutes to get the HNAP calls. I guess there is a cache somewhere.
The best solution I found to lower this time is to change the AP SSID name.

Unfortunately, I did not find tools that can emulates the HNAP calls.

I'll try this today and will give you an update.

Exocet22, 2020-09-28:
Usually, when I change the SSID name on the module and cancel the WiFi configuration on my Mac, it occurs in less than a minute...

Exocet22, 2020-09-30:
Same issue with the alpha release v0.0.2.

See logs attached.
[logs.txt](https://github.com/esp8266/Arduino/files/5304140/logs.txt)

I'll uninstall Avast Antivirus now because it looks like a security analysis of the access point.

Exocet22, 2020-09-30:
Avast disabled. Sounds promising because no HNAP calls during these last 15 minutes...
I'll continue my tests to validate this.

Exocet22, 2020-10-02:
Confirmed : HNAP calls are an Avast security analysis of the access point through Safari.

closed 2020-10-15 by Exocet22

Exocet22, 2020-10-15:
Sorry !

Exocet22, 2020-10-28:
I continued the tests : the problem does not occur when the module is in STA mode. It occurs only in AP mode.

d-a-v, 2020-10-28:
dhcp server is indeed enabled only in AP mode

Exocet22, 2020-10-28:
Correct ! It's the reason why it does not crash !

d-a-v, 2021-02-28:
Title is renamed, and milestone pushed.
Help is requested reproduce HNAP calls.