esp8266/Arduino #7602: Unable to find size and start point in file eboot.elf ?

dominikheinz, 2020-09-17:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Additional Information
This issue is related to the version used by EspHome under the hood.
I like to cross reference this issue, as it seems to be an Arduino problem, rather than an EspHome one.
Please see: https://github.com/esphome/issues/issues/1476

#### Platform

- Hardware: ESP8266
- Core Version: Latest of EspHome
- Development Env: EspHome
- Operating System: HassOS (Linux)

### Settings in IDE
- Module: Generic ESP8266 Module
- Flash Mode: Unknown/Default
- Flash Size: Unknown/Default of Generic ESP8266
- lwip Variant: Unknown
- Reset Method: Unknown
- Flash Frequency: Unknown
- CPU Frequency: Unknown
- Upload Using: Irrelevant, process fails during linking (Doesn't reach upload stage)
- Upload Speed: Irrelevant, process fails during linking (Doesn't reach upload stage)

### Problem Description
Generating a binary for Generic ESP8266 in EspHome causes a crash during linking (Unable to find size and start point in file eboot.elf)
See this screenshot for more information as well as the issue I linked above:

[image]

### [MCVE](https://stackoverflow.com/help/mcve) Sketch
This is the sketch used in EspHome. The sketch is valid, linking of the binary fails, as shown in the screenshot.

```yaml
esphome:
  name: smartplug1
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: censored
  password: censored
  
  manual_ip:
    static_ip: 192.168.1.195
    gateway: 192.168.1.1
    subnet: 255.255.255.0

  ap:
    ssid: censored
    password: censored

api:

logger:

ota:

web_server:
  port: 80

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Sonoff S20 Button"
  - platform: status
    name: "Sonoff S20 Status"

switch:
  - platform: gpio
    name: "Sonoff S20 Relay"
    pin: GPIO12

output:
  # Register the green LED as a dimmable output ....
  - platform: esp8266_pwm
    id: s20_green_led
    pin:
      number: GPIO13
      inverted: True

light:
  # ... and then make a light out of it.
  - platform: monochromatic
    name: "Sonoff S20 Green LED"
    output: s20_green_led
```

**Steps to reproduce:**
1. Install HassOS
2. Install EspHome plugin
3. Try to compile the yaml script linked above
4. Observe the crash

### Debug Messages
Not available

earlephilhower, 2020-09-17:
You have a shared library installation as shown in your log, see the "Error loading shared libraries..." message.  That means you're not actually completing any compiles so you're not going to have a valid ELF to convert to BIN in.  I suggest working with the PlatformIO repo/gitter or the HassOS (I assume it's some minimal custom Linux so might not have many needed .SOs to do lots of things like compiling, etc.).

closed 2020-09-17 by earlephilhower

panakos, 2022-12-21:
I have the same issue but without shared library installation error

panakos, 2022-12-24:
Citrix Workspace was causing the issue. I updated Citrix and  problem gone!