esp8266/Arduino #7606: Bypass 8 client limit esp8266

notsolowki, 2020-09-19:
Is there a way to allow the esp8266 soft_ap to allow up to 10 clients connected at one time?

notsolowki, 2020-09-19:
Everything im reading is showing 8 maximum. i am hoping for a slightly different answer.  Is the esp32 a different stry? can it handle more than 8?

Tech-TX, 2020-09-22:
Google suggests the limit is in the 2.2.1 SDK,  so that'll be a 'no'.
https://arduino.stackexchange.com/questions/72845/increasing-esp32-ap-max-connections/73143#73143
That stackexchange is for ESP32, but I don't expect it's any different for the '8266.

closed 2020-09-22 by devyte

jandrassy, 2020-09-23:
https://arduino.stackexchange.com/questions/55693/how-many-clients-can-connect-to-an-esp32/55780#55780

notsolowki, 2020-09-23:
Actually esp now looks pretty interesting for this