esp8266/Arduino #7613: HTTPClient mystery crash after http.end(), why?

aldoaldoaldo, 2020-09-26:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [x ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12S|
- Core Version:			[2.7.4]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4 Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

This is within a biggest project were I am running the standard sync webserver.
To exclude cross-errors, anyway, I wrote this simple test procedure that is called at a certain time from loop()
The ESP is 192.168.1.11 and it's connected to my wifi router, while 192.168.1.12 is another ESP connected to the same wifi.
The GET call is a success (I see it OK on the receiver), but as you see the "logg" after http.end is not called and instead there is a restart (this implies a crash...).
Heap space is good and >40K you see is logged; logg() is a my years old log procedure.
I see no reasons for crash ???!???!?!???!
NOTICE:: I have a similar procedure that downloads file from Web and it works.
Well... it used to work also the mother of the procedure below (months ago while I used 2.4.2 or so) - so this may be caused by the web response of the other ESP (maybe this is changed from 2.4.2 to 2.7.4) but in any case a http client proc is expected to NOT crash depending on server reply.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

void testsend()
{
      HTTPClient http;
      String surl="http://192.168.1.12/x?&id=1046&n=TS46&w=1&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0";
      logg(surl+"<< TEST");
      http.begin(surl);
      logg("after begin");
      int httpcode= http.GET();
      logg("after GET");
      String s=http.getString();
      logg(s+"<< s - after getString");
      http.end();
      logg(String(httpcode)+"; (after end) s="+s);
}

loop()
{
  deledc(50);  //somewhat like delay but calls dns/web service
  ....
  switch (azione)
  {
    case AZ_TESTSEND:
      {
        deledc(2000);
        testsend();
        azione=AZ_DO_NOTHING;
        break;
      }
    //AZ_DO_NOTHING;
    default:
  }

}

```

### Debug Messages

```
this is the log:
[2020-09-26 22:22.02] G 18az >42608
http://192.168.1.12/x?&id=1046&n=TS46&w=1&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0<< TEST
[2020-09-26 22:22.02] G 18az >42672
after begin
[2020-09-26 22:22.02] G 18az >42360
after GET
[2020-09-26 22:22.02] G 18az >42200
RXOK<< s - after getString
[1970-00-00 00:00.00] G 99az >44824
(setup)WiFi.scanNetworks started
[1970-00-00 00:00.00] G 99az >44840
##############################

  START START START START
##############################
[2020-09-26 22:22.07] G -1az >43736
(wifiEndScanAction)

```

aldoaldoaldo, 2020-09-26:
HTTP/1.1 200 OK
Content-Type: text/html; charset=iso-8859-1
Content-Length: 4
Connection: close

RXOK

(from Firefox debugger, this is what is returned from
http://192.168.1.12/x?&id=1046&n=TS46&w=1&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0

earlephilhower, 2020-09-26:
@aldoaldoaldo you need to post a complete, compile-able and runnable MCVE.  Your snippet isn't useful as it's not complete.  You also did not post any decoded (or undecoded) ESP crash dumps.  W/o these two bits, we can't do anything here.

aldoaldoaldo, 2020-09-27:
I guessed so, but my total code is too big AND that is a very basic procedure that MUST work (I am doing just a GET). Also (sorry) I updated my first post before seeing your reply, but basically see below, now I additionally know that I can ask any URL on Web, but when I ask a URL that is on the other ESP on http://192.168.1.12/ it crashes.

[2020-09-27 01:49.00] G -1az >42128
2020-09-27 01:49.00<< Set time >>2020-09-27 01:49.00
[2020-09-27 01:49.02] G 19az >42736
http://termoeasy.com/1.txt<< TEST
[2020-09-27 01:49.02] G 19az >42792
after begin
[2020-09-27 01:49.02] G 19az >42456
after GET
[2020-09-27 01:49.02] G 19az >42256
(this is a test)
<< s - after getString
[2020-09-27 01:49.02] G 19az >42328
200; (after end) s=(this is a test)

[2020-09-27 01:49.09] G 19az >42528
http://termoeasy.com/1.txt<< TEST
[2020-09-27 01:49.09] G 19az >42584
after begin
[2020-09-27 01:49.09] G 19az >42272
after GET
[2020-09-27 01:49.09] G 19az >42072
(this is a test)
<< s - after getString
[2020-09-27 01:49.09] G 19az >42144
200; (after end) s=(this is a test)

[2020-09-27 01:49.18] G 20az >42376
http://192.168.1.12/<< TEST
[2020-09-27 01:49.18] G 20az >42400
after begin
[2020-09-27 01:49.19] G 20az >42112
after GET
[2020-09-27 01:49.19] G 20az >22864
<html><head><meta name='Content-Type' content='text/html; charset=iso-8859-1'>
[...omissis...]
<br>&copy; 2020 VisualVision - All rights reserved</font></div></body></html>
<< s - after getString
[1970-00-00 00:00.00] G 99az >44792
(setup)WiFi.scanNetworks started
[1970-00-00 00:00.00] G 99az >44808
##############################

  START START START START
##############################
[2020-09-27 01:49.24] G -1az >43704
(wifiEndScanAction)

Now I am writing a new simple sketch on a new ESP........

aldoaldoaldo, 2020-09-27:
``
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

void setup() {
  // put your setup code here, to run once:
  pinMode(2,OUTPUT);
  digitalWrite(2,HIGH); //led off

  WiFi.begin("Vodafone-123","wifi4488aldo");

  int i=350,j=0;
  while (i>0) {
    j=wifi_station_get_connect_status();
    switch(j) {
      case STATION_CONNECT_FAIL:
        i=0;
        break;
      case STATION_WRONG_PASSWORD:
        i=-1;
        break;
      case STATION_NO_AP_FOUND:
        i=-2;
        break;
      case STATION_GOT_IP:
        i=-3;
        break;
    }
    delay(100);
    if (i>0) i--;
  }
  Serial.begin(115200);
  Serial.println();
  Serial.println("Setup; Wifi status="+String(i));
}

bool done=false,led=false;

void testsend()
{
      HTTPClient http;
      //
      //this URL (on the Web) works
      ////String surl="http://termoeasy.com/1.txt";    

      //now other URLs supplied by the web server of another ESP, same wifi, on 192.168.1.12
      //It correctly works and supplies pages with Firefox

      //if this works
      //with w=1 will be received (and logged) as "channel 1" of the ESP at 192.168.1.12
      //(*** at present that ESP receives, so the GET works; but http.end() crashes )
      String surl="http://192.168.1.12/x?&id=1046&n=TS46&w=1&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0";

      //or that url will just show http://192.168.1.12/ homepage
      //String surl="http://192.168.1.12/";
      //(*** at present GET works but http.end() create a crash also with this URL)
      //
      http.begin(surl);
      Serial.println(surl);
      int httpcode= http.GET();
      String s=http.getString();
      Serial.println(s);
      Serial.println("BEFORE http END");
      http.end();
      Serial.println("AFTER http END");
      //
      //if it arrives here, led should blink
      led=true;
      //
      //with w=2 will be received (and logged) as "channel 2" of the ESP at 192.168.1.12
      surl="http://192.168.1.12/x?&id=1046&n=TS46&w=2&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0";
      http.begin(surl);
      Serial.println(surl);
      httpcode= http.GET();
      s=http.getString();
      http.end();
      Serial.println(s);
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(100);
  if (!done)
  if (millis()>2000) 
  {
    testsend();
    done=true;
  }
  bool a=!(digitalRead(2));
  if (led) digitalWrite(2,a);
}

aldoaldoaldo, 2020-09-27:
Setup; Wifi status=-3
http://192.168.1.12/x?&id=1046&n=TS46&w=1&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0
RXOK
BEFORE http END

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (9):
epc1=0x40206dfc epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000023d depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd10 end: 3fffffc0 offset: 0190
3ffffea0:  40203b64 3ffee3f0 3ffe870a 40203e45  
3ffffeb0:  3fffdad0 0000000f 3ffffee0 402029b0  
3ffffec0:  3fffdad0 00000000 3ffffee0 40203754  
3ffffed0:  3fffdad0 00000000 3ffee3f0 4020116f  
3ffffee0:  00000000 00000000 3ffefbec 3ffefcd4  
3ffffef0:  000c000f 00000020 3f010050 3f001388  
3fffff00:  3ffefcec 0040004f 000d0020 70747468  
3fffff10:  00000c00 8400018c 3ffefe4c 0000001f  
3fffff20:  00fef971 3ffef95c 0011001f 00fee46c  
3fffff30:  00000000 4bc6a7f0 80b851eb 00000000  
3fffff40:  00000000 000000c8 00000004 00000000  
3fffff50:  00000000 0020000a 40100200 00001a99  
3fffff60:  80000000 00000000 3ffefd94 3ffefb00  
3fffff70:  0053005f 80fee4c8 4b4f5852 40205000  
3fffff80:  84205701 3ffefb84 0053005f 00fee46c  
3fffff90:  3fffdad0 00000000 3ffee3b9 4020122f  
3fffffa0:  feefeffe 00000000 3ffee42c 40205214  
3fffffb0:  feefeffe feefeffe 3ffe84e8 40100c55  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

aldoaldoaldo, 2020-09-27:
by the way it does the GET (it is received by the other ESP, that responds "RXOK", and the response is correctly received)

earlephilhower, 2020-09-27:
You need to decode the stack dump with the EspExceptionDecoder for it to be use.  There is an Arduino plugin used to do so.  It's described in the readme.

Tech-TX, 2020-09-27:
https://github.com/me-no-dev/EspExceptionDecoder

Copy and paste EVERYTHING between the two 'CUT HERE' points and drop it in the Exception Decoder.  This has to be done on your CURRENT COMPILE.  If you recompile, then that stack trace above is even more meaningless than the dump you posted.  The Exception Decoder gets it's code references from your latest *.elf file, and that changes each time you make a tweak and recompile.

aldoaldoaldo, 2020-09-27:
To be sure I recompiled and got a new dump (that seems identical to previous)

Exception (9):
epc1=0x40206dfc epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000023d depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd10 end: 3fffffc0 offset: 0190
3ffffea0:  40203b64 3ffee3f0 3ffe870a 40203e45  
3ffffeb0:  3fffdad0 0000000f 3ffffee0 402029b0  
3ffffec0:  3fffdad0 00000000 3ffffee0 40203754  
3ffffed0:  3fffdad0 00000000 3ffee3f0 4020116f  
3ffffee0:  00000000 00000000 3ffefbec 3ffefcd4  
3ffffef0:  000c000f 00000020 3f010050 3f001388  
3fffff00:  3ffefcec 0040004f 000d0020 70747468  
3fffff10:  00000c00 8400018c 3ffefe4c 0000001f  
3fffff20:  00fef971 3ffef95c 0011001f 00fee46c  
3fffff30:  00000000 4bc6a7f0 80f3b645 00000000  
3fffff40:  00000000 000000c8 00000004 00000000  
3fffff50:  00000000 001b000a 40100200 0000168c  
3fffff60:  80000000 00000000 3ffefd94 3ffefb00  
3fffff70:  0053005f 80fee4c8 4b4f5852 40205000  
3fffff80:  84205701 3ffefb84 0053005f 00fee46c  
3fffff90:  3fffdad0 00000000 3ffee3b9 4020122f  
3fffffa0:  feefeffe 00000000 3ffee42c 40205214  
3fffffb0:  feefeffe feefeffe 3ffe84e8 40100c55  
<<<stack<<<

but there's no way to copy and paste the decoded results

aldoaldoaldo, 2020-09-27:
a ok no, now the text can be selected. Here we are:

Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
PC: 0x40206dfc: HTTPClient::connected() at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 475
EXCVADDR: 0x0000023d

Decoding stack results
0x40203b64: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/HardwareSerial.h line 164
0x40203e45: Print::write(char const*) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/Print.h line 62
0x402029b0: HTTPClient::disconnect(bool) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 434
0x40203754: HTTPClient::end() at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 425
0x4020116f: testsend() at C:\Users\User\Documents\Arduino\test_httpclient/test_httpclient.ino line 64
0x40100200: millis() at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_wiring.cpp line 188
0x40205000: base64::encode(unsigned char const*, unsigned int, bool) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\base64.cpp line 67
0x4020122f: loop() at C:\Users\User\Documents\Arduino\test_httpclient/test_httpclient.ino line 86
0x40205214: loop_wrapper() at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 197

aldoaldoaldo, 2020-09-27:
(ANOTHER STRANGE BEHAVIOR: DELAY IN MULTIPLE SUBSEQUENT FILE DOWNLOAD / BY STREAM)
While I am here, in 2.7 (& 1.4 H.B.) I've seen also a non explicable slow down when I download multiple very small files on web. This with the usual stream procedure (below) that I copied from a example.
This huge delay was not present when I used 2.4.2 (and additionally I was not compiling in 1.4 High Bandwidth but in default mode).

``
    http.begin(url);   
    int httpCode = http.GET();
    if (httpCode > 0) {
      {
        File ff = SPIFFS.open(destination, "w");
        if (ff) {
          int len = http.getSize();
          uint8_t buff[512] = { 0 };
          WiFiClient * stream = http.getStreamPtr();
          while(http.connected() && (len > 0 || len == -1)) {
             size_t size = stream->available();
             if (size) {
                int c = stream->readBytes(buff, ((size > sizeof(buff)) ? sizeof(buff) : size));
                ff.write(buff, c);
             }
          }
          ff.close();
        }        
      }
      http.end();
``

No crashes here, but around 11-12 seconds to download a small text file (!!).
I have a proc that downloads a list.txt then download the listed items with a while that calls the above proc.
Here the log:

update files
[2020-09-27 13:51.26] G 96az >42160
200 - download OK, URL:DEST http://wfeasy.com/txtemp/list.txt:/list
[2020-09-27 13:51.37] G 96az >41800
200 - download OK, URL:DEST http://wfeasy.com/txtemp/m/td.htm:/td.htm
[2020-09-27 13:51.48] G 96az >42008
404 - download OK, URL:DEST http://wfeasy.com/txtemp/m/ps.htm:/ps.htm
[2020-09-27 13:52.00] G 96az >42064
200 - download OK, URL:DEST http://wfeasy.com/txtemp/m/pl.htm:/pl.htm

earlephilhower, 2020-09-27:
We can't run your MCVE because it needs your local server app, but given the decoded crash signature I am pretty sure that you're running out of memory.

Please enable full debug (Tools->Debug Port->Serial, Tools->Debug Level->(all, the longest one)) and rerun.

You may be GETting an object that's simply too big for RAM.

aldoaldoaldo, 2020-09-29:
No, it's 100 bytes and RAM is almost all available.
Please read all that I posted. 
The GET reponse (obtained with Firefox and copied above) is around 100 bytes including headers.
From the log, in the big original software where I've first seen the bug the heap was 40K so plenty of RAM; in the small example posted above it is obviously a lot more, maybe 60K.
How can it hang for 100 bytes?
Also the GET response is correctly received and logged without any RAM problem.
It HANGS only WHEN CLOSING, while executing http.end() - or if I remove http.end() it hangs at the end of the proc, I guess when the compiler destroys/frees the object "http" as I do not explicitly destroy it.
The bug happens only when calling that server on same wifi / router, by IP (it seems also by DNS if I run a DNS on the other ESP), so it may be connected to timings or... who knows.
To run the MCVE and to test it you just need another ESP with a web server turned on within the same network and that serves any page. While I copy below the first page where I found the bug, I can say it happens also for 2 other pages so it seems definitely unrelated on the page. Maybe it will happen also with any other webserver on the same local net.

HTTP/1.1 200 OK
Content-Type: text/html; charset=iso-8859-1
Content-Length: 4
Connection: close

RXOK

(this is the page served as reply (with standard sync webserver) - it just confirms data have been received from the other ESP.)

aldoaldoaldo, 2020-09-29:
The handler on the ESP server is (basically)

void xdata()
{
  String testo="RXOK";
  server.send(200, "text/html; charset=iso-8859-1", testo); 
}

d-a-v, 2020-09-29:
Trying to narrow the issue:

Can you add `Serial.flush();` after all your `Serial.print...` ?

Can you declare a `WiFiClient httpClient;` just above `HTTPClient http;` and use it in all your `begin`:
```
      http.begin(httpClient, surl);
```

aldoaldoaldo, 2020-09-29:
Here without declaring the WifiClient

SDK:2.2.2-dev(38a443e)/Core:2.7.3-3-g2843a5ac=20703003/lwIP:1.4.0rc2/BearSSL:5c771be

Setup; Wifi status=-3
[HTTP-Client][begin] url: http://192.168.1.12/x?&id=1046&n=TS46&w=1&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0
[HTTP-Client][begin] host: 192.168.1.12 port: 80 url: /x?&id=1046&n=TS46&w=1&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0
http://192.168.1.12/x?&id=1046&n=TS46&w=1&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0
[HTTP-Client][sendRequest] type: 'GET' redirCount: 0
[HTTP-Client] connected to 192.168.1.12:80
[HTTP-Client] sending request header
-----
GET /x?&id=1046&n=TS46&w=1&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0 HTTP/1.1
Host: 192.168.1.12
User-Agent: ESP8266HTTPClient
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
Connection: keep-alive
Content-Length: 0

-----
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 200 OK
'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: text/html; charset=iso-8859-1
'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 4
'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close
'
[HTTP-Client][handleHeaderResponse] RX: '
'
[HTTP-Client][handleHeaderResponse] code: 200
[HTTP-Client][handleHeaderResponse] size: 4
[HTTP-Client][writeToStreamDataBlock] end of chunk or data (transferred: 4).
[HTTP-Client][end] tcp stop
RXOK
BEFORE http END
Fatal exception 28(LoadProhibitedCause):
epc1=0x40207578, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000034, depc=0x00000000

----

Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
PC: 0x40207578: HTTPClient::connected() at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 475
EXCVADDR: 0x00000034

Decoding stack results
0x40203fbc: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/HardwareSerial.h line 164
0x40205caa: __delay(unsigned long) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_wiring.cpp line 54
0x40100234: delayMicroseconds(unsigned int) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_wiring.cpp line 210
0x40203905: HTTPClient::disconnect(bool) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 434
0x402039c8: HTTPClient::end() at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 425
0x40201181: testsend() at C:\Users\User\Documents\Arduino\test_httpclient/test_httpclient.ino line 74
0x40100200: millis() at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_wiring.cpp line 188
0x4020124f: loop() at C:\Users\User\Documents\Arduino\test_httpclient/test_httpclient.ino line 100
0x402057c8: loop_wrapper() at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 197

aldoaldoaldo, 2020-09-29:
Instead, if I declare WiFiClient httpClient; and change the begin() , it appears working.

earlephilhower, 2020-09-29:
Again, please enable full debug (Tools->Debug Port->Serial, Tools->Debug Level->(all, the longest one)) and rerun.

The EXCVADDR = <about 0> likely means that there's a NULL somewhere, most likely from a malloc.  If not from an allocation, debug may also catch other error cases.

earlephilhower, 2020-09-29:
> Instead, if I declare WiFiClient httpClient; and change the begin() , it appears working.

That would imply that the HTTPClient compatibility (deprecated) wrapper is assigning NULL to the client pointer for some reason.  Narrows down the potential codepaths a lot.

aldoaldoaldo, 2020-09-29:
You are correct, I tried only with HTTPclient debug enabled (I had fear of too much debug lines).
With that and http.begin(surl) keeps crashing and I posted the stuff above.
With that and http.begin(httpClient,surl) it works.

Now I tried with the most long debug available; and again http.begin() and no httpClient.
But.... with this..... it works ??!?? (maybe it is something about timing and with the full debug the serial printing time solves?).
I am puzzled.

SDK:2.2.2-dev(38a443e)/Core:2.7.3-3-g2843a5ac=20703003/lwIP:1.4.0rc2/BearSSL:5c771be

Setup AA; Wifi status=-3
[HTTP-Client][begin] url: http://192.168.1.12/x?&id=1046&n=TS46&w=1&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0
[HTTP-Client][begin] host: 192.168.1.12 port: 80 url: /x?&id=1046&n=TS46&w=1&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0
http://192.168.1.12/x?&id=1046&n=TS46&w=1&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0
[HTTP-Client][sendRequest] type: 'GET' redirCount: 0
[hostByName] Host: 192.168.1.12 is a IP!
:ref 1
[HTTP-Client] connected to 192.168.1.12:80
[HTTP-Client] sending request header
-----
GET /x?&id=1046&n=TS46&w=1&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0 HTTP/1.1
Host: 192.168.1.12
User-Agent: ESP8266HTTPClient
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
Connection: keep-alive
Content-Length: 0

-----
:wr 226 0
:wrc 226 226 0
:ack 226
:rn 102
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 200 OK
'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: text/html; charset=iso-8859-1
'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 4
'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close
'
:c0 1, 102
[HTTP-Client][handleHeaderResponse] RX: '
'
[HTTP-Client][handleHeaderResponse] code: 200
[HTTP-Client][handleHeaderResponse] size: 4
:rn 4
:c0 1, 4
[HTTP-Client][writeToStreamDataBlock] end of chunk or data (transferred: 4).
[HTTP-Client][end] tcp stop
:close
:ur 1
:dsrcv 0
:del
RXOK
BEFORE http END
[HTTP-Client][end] tcp is closed
AFTER http END
[HTTP-Client][begin] url: http://192.168.1.12/x?&id=1046&n=TS46&w=2&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0
[HTTP-Client][begin] host: 192.168.1.12 port: 80 url: /x?&id=1046&n=TS46&w=2&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0
http://192.168.1.12/x?&id=1046&n=TS46&w=2&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0
[HTTP-Client][sendRequest] type: 'GET' redirCount: 0
[hostByName] Host: 192.168.1.12 is a IP!
:ref 1
[HTTP-Client] connected to 192.168.1.12:80
[HTTP-Client] sending request header
-----
GET /x?&id=1046&n=TS46&w=2&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0 HTTP/1.1
Host: 192.168.1.12
User-Agent: ESP8266HTTPClient
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
Connection: keep-alive
Content-Length: 0

-----
:wr 226 0
:wrc 226 226 0
:ack 226
:rn 102
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 200 OK
'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: text/html; charset=iso-8859-1
'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 4
'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close
'
:c0 1, 102
[HTTP-Client][handleHeaderResponse] RX: '
'
[HTTP-Client][handleHeaderResponse] code: 200
[HTTP-Client][handleHeaderResponse] size: 4
:rn 4
:c0 1, 4
[HTTP-Client][writeToStreamDataBlock] end of chunk or data (transferred: 4).
[HTTP-Client][end] tcp stop
:close
:ur 1
:dsrcv 0
:del
[HTTP-Client][end] tcp is closed
RXOK

aldoaldoaldo, 2020-09-29:
I resume for clarity:
- with http.begin(surl) and either without debug, or with HTTPclient debug enabled, it crashes; With super-full debug it does not crash
- with declared WifiClient and http.begin(httpClient,surl) it works (and I tried declaring either within the testsend() or also at program start)

To test it by yourself you need another ESP with a standard sync http server connected to the same Wifi network, and to return a simple page like shown, I'd bet that you don't need my precise ESP server.
All of this does not happen when a "real Web" URL is required.

By the way if this is like you guess, I understand that (meantime) if I explicity declare 
WiFiClient httpClient;
I should be safe, am I correct?

d-a-v, 2020-09-29:
>  The .flush() appears not relevant.

Meaning what's displayed just before crashing remains as is ?

> I understand that (meantime) if I explicity declare
`WiFiClient httpClient;`
I should be safe, am I correct?

Yes.

aldoaldoaldo, 2020-09-29:
I recompiled without Serial.flush() and without WiFiClient, with HTTPclient debug only.
This now works.

'''
SDK:2.2.2-dev(38a443e)/Core:2.7.3-3-g2843a5ac=20703003/lwIP:1.4.0rc2/BearSSL:5c771be

Setup ver without flush without httpClient; Wifi status=-3
[HTTP-Client][begin] url: http://192.168.1.12/x?&id=1046&n=TS46&w=1&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0
[HTTP-Client][begin] host: 192.168.1.12 port: 80 url: /x?&id=1046&n=TS46&w=1&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0
http://192.168.1.12/x?&id=1046&n=TS46&w=1&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0
[HTTP-Client][sendRequest] type: 'GET' redirCount: 0
[HTTP-Client] connected to 192.168.1.12:80
[HTTP-Client] sending request header
-----
GET /x?&id=1046&n=TS46&w=1&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0 HTTP/1.1
Host: 192.168.1.12
User-Agent: ESP8266HTTPClient
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
Connection: keep-alive
Content-Length: 0

-----
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 200 OK
'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: text/html; charset=iso-8859-1
'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 4
'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close
'
[HTTP-Client][handleHeaderResponse] RX: '
'
[HTTP-Client][handleHeaderResponse] code: 200
[HTTP-Client][handleHeaderResponse] size: 4
[HTTP-Client][writeToStreamDataBlock] end of chunk or data (transferred: 4).
[HTTP-Client][end] tcp stop
RXOK
BEFORE http END
[HTTP-Client][end] tcp is closed
AFTER http END
[HTTP-Client][begin] url: http://192.168.1.12/x?&id=1046&n=TS46&w=2&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0
[HTTP-Client][begin] host: 192.168.1.12 port: 80 url: /x?&id=1046&n=TS46&w=2&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0
http://192.168.1.12/x?&id=1046&n=TS46&w=2&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0
[HTTP-Client][sendRequest] type: 'GET' redirCount: 0
[HTTP-Client] connected to 192.168.1.12:80
[HTTP-Client] sending request header
-----
GET /x?&id=1046&n=TS46&w=2&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0 HTTP/1.1
Host: 192.168.1.12
User-Agent: ESP8266HTTPClient
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
Connection: keep-alive
Content-Length: 0

-----
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 200 OK
'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: text/html; charset=iso-8859-1
'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 4
'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close
'
[HTTP-Client][handleHeaderResponse] RX: '
'
[HTTP-Client][handleHeaderResponse] code: 200
[HTTP-Client][handleHeaderResponse] size: 4
[HTTP-Client][writeToStreamDataBlock] end of chunk or data (transferred: 4).
[HTTP-Client][end] tcp stop
[HTTP-Client][end] tcp is closed
RXOK
'''

aldoaldoaldo, 2020-09-29:
This instead crashes with HTTPClient debug on Serial on:

'''
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

//WiFiClient httpClient;

void setup() {
  // put your setup code here, to run once:
  pinMode(2,OUTPUT);
  digitalWrite(2,HIGH); //led off

  WiFi.begin("Vodafone-123","wifiXXXXaldo");

  int i=350,j=0;
  while (i>0) {
    j=wifi_station_get_connect_status();
    switch(j) {
      case STATION_CONNECT_FAIL:
        i=0;
        break;
      case STATION_WRONG_PASSWORD:
        i=-1;
        break;
      case STATION_NO_AP_FOUND:
        i=-2;
        break;
      case STATION_GOT_IP:
        i=-3;
        break;
    }
    delay(100);
    if (i>0) i--;
  }
  Serial.begin(115200);
  Serial.println();
  Serial.println("Setup; Wifi status="+String(i));
  Serial.flush();
}

bool done=false,led=false;

void testsend()
{
      //WiFiClient httpClient;
      HTTPClient http;
      //
      //this URL (on the Web) works
      ////String surl="http://termoeasy.com/1.txt";    

      //now other URLs supplied by the web server of another ESP, same wifi, on 192.168.1.12
      //It correctly works and supplies pages with Firefox

      //if this works
      //with w=1 will be received (and logged) as "channel 1" of the ESP at 192.168.1.12
      //(*** at present that ESP receives, so the GET works; but http.end() crashes )
      String surl="http://192.168.1.12/x?&id=1046&n=TS46&w=1&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0";

      //or that url will just show http://192.168.1.12/ homepage
      //String surl="http://192.168.1.12/";
      //(*** at present GET works but http.end() create a crash also with this URL)
      //
      //http.begin(httpClient, surl);
      http.begin(surl);
      Serial.println(surl);
      Serial.flush();
      int httpcode= http.GET();
      String s=http.getString();
      Serial.println(s);
      Serial.println("BEFORE http END");
      Serial.flush();
      http.end();

      //at present never goes here....
      
      Serial.println("AFTER http END");
      Serial.flush();
      //
      //if it arrives here, led should blink
      led=true;
      //
      //with w=2 will be received (and logged) as "channel 2" of the ESP at 192.168.1.12
      surl="http://192.168.1.12/x?&id=1046&n=TS46&w=2&b=26&dm=15&pw=pippo&r=CE6400&t=-12700&h=0";
      //http.begin(httpClient, surl);
      http.begin(surl);
      Serial.println(surl);
      Serial.flush();
      httpcode= http.GET();
      s=http.getString();
      http.end();
      Serial.println(s);
      Serial.flush();
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(100);
  if (!done)
  if (millis()>2000) 
  {
    testsend();
    done=true;
  }
  bool a=!(digitalRead(2));
  if (led) digitalWrite(2,a);
}
'''

aldoaldoaldo, 2020-09-29:
and the decode
```
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
PC: 0x40207578: HTTPClient::connected() at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 475
EXCVADDR: 0x00000034

Decoding stack results
0x40203fbc: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/HardwareSerial.h line 164
0x40205caa: __delay(unsigned long) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_wiring.cpp line 54
0x40100234: delayMicroseconds(unsigned int) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_wiring.cpp line 210
0x40203905: HTTPClient::disconnect(bool) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 434
0x402039c8: HTTPClient::end() at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 425
0x40201181: testsend() at C:\Users\User\Documents\Arduino\test_httpclient/test_httpclient.ino line 74
0x40100200: millis() at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_wiring.cpp line 188
0x4020124f: loop() at C:\Users\User\Documents\Arduino\test_httpclient/test_httpclient.ino line 100
0x402057c8: loop_wrapper() at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 197
```

aldoaldoaldo, 2020-10-01:
I tested 
WiFiClient httpClient;
in the big application (the one where I discovered the error; I put it within the procedure where I use it). All right now with this modif.

But I noticed now that, at global level, I already have a
WiFiClient espClient;
used for another protocol.
Is this fine or should I have to use only one WiFiClient??

earlephilhower, 2020-10-02:
That should be fine, no need to only have a single WiFiClient (other than a little memory used).  For WiFiClientSecure, though, it's a good idea to only have 1 copy because it requires 20KB+ of buffer space for SSL buffers.

Closing this as the busted, deprecated HTTPClient begin method w/o a client are now removed.  If you have a crash using the supported one (passing in your own WiFiClient) let us know and we can reopen.

closed 2020-10-02 by earlephilhower

PA4WD, 2021-01-28:
```
HTTP/1.1 200 OK  \r\n
Content-type: text/plain\r\n  
Connection: close\r\n  
\r\n  
OK
```
Crashes with this trace
```
Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
PC: 0x4021ea58: HTTPClient::connected() at C:\Users\ms\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 475
EXCVADDR: 0x00000459

Decoding stack results
0x4021b4ac: String::operator=(String const&) at C:\Users\ms\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\WString.cpp line 262
0x40212274: HTTPClient::disconnect(bool) at C:\Users\ms\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 434
0x402130f8: HTTPClient::end() at C:\Users\ms\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 425
0x402061b5: prtgServerUpdate() at C:\Arduino\H1602_WSEN/H1602_WSEN_prtg.ino line 43
0x4020c757: loop() at C:\Arduino\H1602_WSEN/H1602_WSEN.ino line 372
0x4021c6a4: loop_wrapper() at C:\Users\ms\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 197
```
Removing the "Connection: close" header from the server returned message gives no crash.
Adding a WiFiClient works for both

d-a-v, 2021-01-28:
@PA4WD can you open a new issue, fill the issue template with the exact version of the core you are using and a sketch (MCVE) to reproduce it ?
Can you also try with the [git master version](https://arduino-esp8266.readthedocs.io/en/latest/installing.html#using-git-version) (or the unofficial [snapshot release](https://d-a-v.github.io/esp8266/Arduino)) ?

stomachcontents, 2021-11-30:
I've had the same issue, even with one WifiClient.   Turned out I needed to wipe the flash completely.  I assume my WIFI settings on the flash were bad. I had tried several boards because I couldn't identify my NodeMCU version easily. I assume something bogus was written to the Wifi settings caused the http.end() to bomb indirectly.  After a full wipe everything works perfectly.

barneyman, 2022-05-05:
I've just started to experience this 'exception 9 at httpClient.end()' 

it's caused by `keep-alive` keeping a pointer to the wificlient instance when the `dtor` for `wificlient` is called before the `dtor` for `httpclient`, causing the latter to hit hyperspace (and possibly explains why the stack can't be decoded).

Two fixes

Either protect the `dtor` order with scope
``` 
WiFiClient wfc
{ // guard scope for wfc by killing hClient first
   HTTPClient hClient;
   ....
   hClient.end()
}
```

or turn the 'feature' that causes it off `hClient.setReuse(false);`

Hope this helps someone else