esp8266/Arduino #7614: Installation package ESP8266 on Arduino IDE 1.8.13 for Linux(ARM) misses FS.h and TZ.h.

rin67630, 2020-09-26:
I have installed on a Raspberry Pi (current Raspbian Buster) the ESP8266 board manager from the library of the Arduino IDE 1.8.13 using the URL: http://arduino.esp8266.com/stable/package_esp8266com_index.json.

I seems to work  so far, but upon compling, the compiler complained to miss the libraries FS.h and TZ.h.

Both libraries are part of the core and should have been installed together with the board manager.
Are some parts of the core missing in the ARM-version?

earlephilhower, 2020-09-26:
Please post the exact messages.  Those should be part of the core ZIP itself and not the toolchain (ARM specific) zip.

earlephilhower, 2020-09-26:
One thing that came up in offline discussion is that there might be a filesystem case-sensitivity issue.  If you're running your home/install dir on a VFAT FS, for example, you may have issues where another file `fs.h` is being picked up instead of `FS.h`.    You would also probably have tons of other issues, too, unrelated to Arduino...

rin67630, 2020-09-28:
Here is the compiler Error:

`a1_Libs___Vars:9:53: fatal error: TZ.h: No such file or directory
Multiple libraries were found for "WiFiUdp.h"
 Used: /home/pi/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi
 Not used: /home/pi/arduino-1.8.13/libraries/WiFi
 #include <TZ.h>            // default from Espressif
                                                     ^
compilation terminated.
exit status 1
TZ.h: No such file or directory
`

earlephilhower, 2020-09-28:
@rin67630 you're running a very old version of the core, or somehow an older folder got leftover during an upgrade.  If that's the case, all bets are off.

Start by clearing out whatever is there:
* Please remove the 8266 repo from the Arduino boards manager, exit the IDE
* `rm -rf /home/pi/.arduino15/packages/esp8266/hardware/esp8266/2.5.2` to delete any cruft leftover
* Restart Arduino and install 2.7.4 from the boards manager

Retry a build then.  You may also need to remove `/home/pi/arduino-1.8.13/libraries/WiFi` as that is incompatible with the 8266.

If that fails the same way, do a
`ls -la /home/pi/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/` and verify that the file `TZ.h` is there and readable.

rin67630, 2020-09-29:
You are my hero!
THANK YOU.

closed 2020-09-29 by rin67630