esp8266/Arduino #7616: Setting a callback for SNTP time sync event [feature request]

JakubRakus, 2020-09-30:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12]
- Core Version:			[2.7.4]
- Development Env:		[Platformio]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Function `settimeofday_cb()` sets a callback executed every time when `settimeofday()` is invoked. SNTP synchronization event change system time using `settimeofday()` - which is ok, but if I run `settimeofday()` manually also callback function is invoked. There is no mechanism to show what was the source of callback - network synchronization or manual set.

### Expected Behavior

I can see at least three possible solutions:
1. Add a parameter to callback function declaration - maybe some pointer to `bool` type variable which will be set to true if SNTP was a source of sync.
2. Add another `settimeofday`-like function which do the same, but do not fire any callback.
3. Expose ESP SDK function `sntp_set_time_sync_notification_cb()` from `lwip/apps/sntp/sntp.c` which sets a callback ONLY for SNTP synchronization.

Personally I prefer the 3rd one (only because I tested it in "plain" not "Arduino-based" ESP projects and works like a charm).

d-a-v, 2020-09-30:
1. Possible but that would break the current API
2. Possible but that would be a pity because `settimeofday` is part of the posix standard.
3. I was not aware and found no documentation of `sntp_set_time_sync_notification_cb()` in nonos-sdk. Let's try anyway to not further depend on ESP API.

There is a 4:
- Do not call the callback when user is setting time with `settimeofday()`
(this requires a change in lwip2)
This also breaks the API but in a less invasive way: only when user explicitely sets time *and* relies on this CB.

JakubRakus, 2020-10-01:
@d-a-v
>     2. Possible but that would be a pity because `settimeofday` is part of the posix standard.
Don't think it's "pity". It's just a new Arduino specific additional funcionality. Plus it don't breaks current API.
>     3. I was not aware and found no documentation of `sntp_set_time_sync_notification_cb()` in nonos-sdk. Let's try anyway to not further depend on ESP API.
I checked again and you are right, `sntp_set_time_sync_notification_cb()` is only in RTOS-SDK.
>     * Do not call the callback when user is setting time with `settimeofday()`
>       (this requires a change in lwip2)
>       This also breaks the API but in a less invasive way: only when user explicitely sets time _and_ relies on this CB.
May be a solution, but don't think is "less invasive". If there are existing apps (I bet they are) that somehow relies on `settimeofday` callback, devs wouldn't be happy about that - need to remember that they have to invoke existing callback on their own after `settimeofday` explicit usage.

Anyway, any working solution will be appreciated. It should address problems like syncing with external RTC, setting time by the end user via some button/display interface etc. which were, and still are, discussed here on Github issues, Arduino forums and other places.

PR #7637 (2020-10-23): settimeofday_cb: distinguish from user or sntp
This PR overrides `settimeofday_cb()` with a new callback signature:
- currently `void myCB ()` is and remains accepted
- `void myBetterCB (bool from_sntp)` is added: `from_sntp` is true when time is set by SNTP, otherwise it is false.

Consequently:
- lwip2 is updated with a trivial change
- [NTP_TZ_DST](https://github.com/esp8266/Arduino/blob/a0648ee62f60a24e67bc79176c0a8a24307ae440/libraries/esp8266/examples/NTP-TZ-DST/NTP-TZ-DST.ino) example is also updated to show this new callback

Fixes #7616 @JakubRakus
closed 2020-10-23 by d-a-v