esp8266/Arduino #7620: WiFi Multi library endless connect loops and missing scan timeout handling

Erriez, 2020-09-30:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[ESP8266]
- Core Version:			[master]
- Development Env:		[N.A.]
- Operating System:		[N.A.]

### Settings in IDE

N.A.

### Problem Description

1. A missing scan completed results in endless connect loops

2. The `WiFiMulti` example aborts connecting to other networks on failed WiFi connection.

### Sketch

```cpp
#include <ESP8266WiFiMulti.h>

ESP8266WiFiMulti wifiMulti;

void setup() 
{
    Serial.begin(115200);
  
    WiFi.mode(WIFI_STA);
    wifiMulti.addAP("MySSID1", "wrong_password");
    wifiMulti.addAP("MySSID2", "correct_password");
  
    Serial.println("Connecting Wifi...");
    if (wifiMulti.run() == WL_CONNECTED) {
        Serial.println("");
        Serial.println("WiFi connected");
        Serial.println("IP address: ");
        Serial.println(WiFi.localIP());
    }
}

void loop() 
{
    if (wifiMulti.run() != WL_CONNECTED) {
        Serial.println("WiFi not connected!");
        delay(1000);
    }
}
```

### Debug Messages

```
[WIFI] Connecting BSSID: XX:XX:XX:XX:XX:XX SSID: MySSID1 Channel: 11 (-72)
[WIFI] Connecting Failed (6).
WiFi not connected!
[WIFI] delete old wifi config...
[WIFI] start scan
WiFi not connected!
WiFi not connected!
WiFi not connected!
[WIFI] scan done
[WIFI] 8 networks found
 --->  0: [1][XX:XX:XX:XX:XX:XX] MySSID2 (-80) *
       1: [1][XX:XX:XX:XX:XX:XX] XXXXXXXXXXX (-90) *
       2: [1][XX:XX:XX:XX:XX:XX] XXXXXXXXX (-89) *
       3: [1][XX:XX:XX:XX:XX:XX] XXXXXXXXXXX (-90) *
 --->  4: [11][XX:XX:XX:XX:XX:XX] MySSID1 (-70) *
       5: [3][XX:XX:XX:XX:XX:XX] XXXXXXXXXXX (-85) *
       6: [11][XX:XX:XX:XX:XX:XX] XXXXXX (-70) *
       7: [11][XX:XX:XX:XX:XX:XX] XXXXXXXXX (-94) *

[WIFI] Connecting BSSID: XX:XX:XX:XX:XX:XX SSID: MySSID1 Channel: 11 (-70)
[WIFI] Connecting Failed (6).
WiFi not connected!
[WIFI] delete old wifi config...
[WIFI] start scan
```

I've redesigned the complete library and created a pull request:

https://github.com/esp8266/Arduino/pull/7619

Fixed critical issues WiFiMulti library:

- WiFi scan timeout implemented to prevent endless connect loop
- Fallback implemented on WiFi connect failure to prevent endless loop
- Fast WiFi connection at startup
- Improved debug prints
- Doxygen added
- Code maturing
- Example update

The API is identical.

MonishMathew, 2020-10-01:
> ### Basic Infos
> * [x]  This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
> * [x]  I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
> * [x]  I have tested that the issue is present in current master branch (aka latest git).
> * [x]  I have searched the issue tracker for a similar issue.
> * [ ]  If there is a stack dump, I have decoded it.
> * [x]  I have filled out all fields below.
> 
> #### Platform
> * Hardware:				[ESP8266]
> * Core Version:			[master]
> * Development Env:		[N.A.]
> * Operating System:		[N.A.]
> 
> ### Settings in IDE
> N.A.
> 
> ### Problem Description
> 1. A missing WiFi scan completed results in endless connect loops
> 2. The `WiFiMulti` example does not connect to other registered networks on WiFi connection failure such as incorrect password.
> 
> ### Sketch
> ```c++
> #include <ESP8266WiFiMulti.h>
> 
> ESP8266WiFiMulti wifiMulti;
> 
> void setup() 
> {
>     Serial.begin(115200);
>   
>     WiFi.mode(WIFI_STA);
>     wifiMulti.addAP("MySSID1", "wrong_password");
>     wifiMulti.addAP("MySSID2", "correct_password");
>   
>     Serial.println("Connecting Wifi...");
>     if (wifiMulti.run() == WL_CONNECTED) {
>         Serial.println("");
>         Serial.println("WiFi connected");
>         Serial.println("IP address: ");
>         Serial.println(WiFi.localIP());
>     }
> }
> 
> void loop() 
> {
>     if (wifiMulti.run() != WL_CONNECTED) {
>         Serial.println("WiFi not connected!");
>         delay(1000);
>     }
> }
> ```
> 
> ### Debug Messages
> ```
> [WIFI] Connecting BSSID: XX:XX:XX:XX:XX:XX SSID: MySSID1 Channel: 11 (-72)
> [WIFI] Connecting Failed (6).
> WiFi not connected!
> [WIFI] delete old wifi config...
> [WIFI] start scan
> WiFi not connected!
> WiFi not connected!
> WiFi not connected!
> [WIFI] scan done
> [WIFI] 8 networks found
>  --->  0: [1][XX:XX:XX:XX:XX:XX] MySSID2 (-80) *
>        1: [1][XX:XX:XX:XX:XX:XX] XXXXXXXXXXX (-90) *
>        2: [1][XX:XX:XX:XX:XX:XX] XXXXXXXXX (-89) *
>        3: [1][XX:XX:XX:XX:XX:XX] XXXXXXXXXXX (-90) *
>  --->  4: [11][XX:XX:XX:XX:XX:XX] MySSID1 (-70) *
>        5: [3][XX:XX:XX:XX:XX:XX] XXXXXXXXXXX (-85) *
>        6: [11][XX:XX:XX:XX:XX:XX] XXXXXX (-70) *
>        7: [11][XX:XX:XX:XX:XX:XX] XXXXXXXXX (-94) *
> 
> 
> [WIFI] Connecting BSSID: XX:XX:XX:XX:XX:XX SSID: MySSID1 Channel: 11 (-70)
> [WIFI] Connecting Failed (6).
> WiFi not connected!
> [WIFI] delete old wifi config...
> [WIFI] start scan
> ```
> 
> I've redesigned the complete library and created a pull request:
> 
> #7619
> 
> Fixed critical issues WiFiMulti library:
> 
> * WiFi scan timeout implemented to prevent endless connect loop
> * Fallback implemented on WiFi connect failure to prevent endless loop
> * Fast WiFi connection at startup
> * Improved debug prints
> * Doxygen added
> * Code maturing
> * Example update
> 
> The API is identical.

Try to use these libraries,
#include <ESP8266WiFi.h>
#include <ConfigManager.h>
#include <PubSubClient.h>

MonishMathew, 2020-10-01:
Try to add these libraries,
#include <ESP8266WiFi.h>
#include <ConfigManager.h>
#include <PubSubClient.h>

Erriez, 2020-10-03:
Resolved in pull request #7619.

closed 2020-10-03 by Erriez