esp8266/Arduino #7622: ESP8266WebServer Arg Max Length 32

ripper121, 2020-10-02:
When I try to send an argument to my ESP8266WebServer, it only handles 32 Bytes of it and everything longer will be cut off.
Is there definition which I could change to get longer argument length?

I Send out:
http://192.168.178.73/save?myArg=01234567890123456789012345678901234567890123456789
and i only get:
01234567890123456789012345678901

` for (uint8_t i = 0; i < server.args(); i++) {
    if (server.argName(i) == "myArg") {
      myArg= server.arg(i);
    }
  }`

lukasostendorf, 2020-10-05:
I can not reproduce this behavior. I used the [AdvancedWebServer.ino](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/AdvancedWebServer/AdvancedWebServer.ino) example and your test URL.

In handleNotFound() this executes:
```
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET) ? "GET" : "POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";

  for (uint8_t i = 0; i < server.args(); i++) {
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
```

This returns:
```
File Not Found

URI: /save
Method: GET
Arguments: 1
 myArg: 01234567890123456789012345678901234567890123456789
```

Are you sure that you are not limiting the string length in your `myArg` variable or in later processing? Are you using the current library version 2.7.4?

ripper121, 2020-10-05:
Yes I was on 2.7.4.
After reinstallation via Boards Manager the Bug is gone :)

closed 2020-10-05 by ripper121