esp8266/Arduino #7632: NTP network flood

skorokithakis, 2020-10-03:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					WeMos D1 mini
- Core Version:			Unsure
- Development Env:		Platformio
- Operating System:		Ubuntu

### Problem Description

I have this code that doesn't use NTP:

https://gitlab.com/stavros/internet-cube/-/blob/master/src/main.ino

However, I see tens of NTP packets per second onto my network's NTP server. The NTP server is advertised by DHCP, so I assume that's how the ESP discovers it. I noticed the NTP flood when using Wireshark for an unrelated issue.

pipi61, 2020-10-03:
Poweroff this ESP, and check again the NTP req. packets?

skorokithakis, 2020-10-03:
What do you mean? There are no packets if I power off the ESP, which packets are you referring to?

pipi61, 2020-10-04:
Yes, check if power off esp, with wireshark.
Check another Esp example, then check NTP packet exist? If not, then analyze used libraryes

skorokithakis, 2020-10-04:
How would the ESP send packets if it's not powered on? I am very confused.

pipi61, 2020-10-04:
If ESP not powered, and NTP packet found, then other device requ NTP...
This first check...

skorokithakis, 2020-10-04:
I can see the source IP in the packet dump, it's definitely coming from the ESP.

pipi61, 2020-10-04:
ok. Next check, another simple example check. If not see NTP packet, then check the used libraries (include files, and source) in your main.ino

skorokithakis, 2020-10-04:
I have another project where I see the NTP flood from as well. I also have various others where I don't, but those are older. I will try to remove libraries one by one, but WiFiManager is the only one that could be interfering...

pipi61, 2020-10-04:
Yes remove step by step, and check :)

d-a-v, 2020-10-06:
@skorokithakis Any news ?

Can you filter incoming/outgoing packets by mac address (not IP) and show dhcp (udp ports 67, 68) and ntp (udp, port 123) so we try and understand what happens.

Also in the source code you give in OP, `WiFiManager` instance is declared as local in `setup()`. I'm not sure that's the best way to use it. Can you try to move it as global instance ?

FYI, current master has an integrated "wireshark" facility in the Netdump library.

skorokithakis, 2020-10-06:
Moving WiFiManager to a global seems to have fixed the issue, thank you!

closed 2020-10-06 by skorokithakis