esp8266/Arduino #7642: wifi_fpm_get_sleep_type() checks before wifi_fpm_do_wakeup()

Erriez, 2020-10-07:
The [datasheet](https://www.espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf) page 69 describes for API call `wifi_fpm_do_wakeup()`:

_This API can only be called when MODEM_SLEEP_T force sleep function is enabled_

At three places in `ESP8266WiFiGeneric.cpp`, the code checks against `NONE_SLEEP_T` with `LIGHT_SLEEP_T` included:

https://github.com/esp8266/Arduino/blob/bc2b13ae1691f94f9844fa1b0a740cda12107dae/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp#L421

I expected:
```c++
if (wifi_fpm_get_sleep_type() == MODEM_SLEEP_T) {
    wifi_fpm_do_wakeup();
    wifi_fpm_close();
}
```

devyte, 2020-10-08:
CC @Tech-TX

Tech-TX, 2020-10-10:
I'll have to look at it more closely, but on the face of it you are correct.  If the modem is force-slept, then it's utterly off: *sleep_type = {MODEM_SLEEP_T}*, and you have to hit it with *wifi_fpm_do_wakeup()* to turn it back on again.  If it's in light sleep *sleep_type = {LIGHT_SLEEP_T}* then the modem is ON but cycling off whenever it's not transmitting, and the CPU also halts during 'idle' / delay().  It shouldn't hurt to bring it out of light modem sleep with *wifi_fpm_do_wakeup()*, but I'll have to test it with the change to see what it does currently and with the modification.  They may have done it that way in the code to give the user a way to abort light modem sleep without doing SDK calls.  The other way to end automatic light sleep would be to do *wifi_fpm_set_sleep_type(NONE_SLEEP_T)*.

Tech-TX, 2020-10-10:
Without peppering test points throughout the core files, it appears that *wifi_fpm_do_wakeup()* is ignored if it's not in forced modem sleep.  I put it in light sleep, looked at the dynamic currents, then executed wifi_fpm_do_wakeup() and saw no difference: it was still in Automatic Light Sleep (modem off when not transmitting, CPU off during occasional 'idle' periods).

Looks like a poor translation in the API Reference to me.  It should have probably read:
"_This API only wakes the modem when MODEM_SLEEP_T force sleep function is enabled_"

[image]

closed 2021-03-23 by earlephilhower