esp8266/Arduino #7647: Is OTA detecting bad FLASH now?

pidloop, 2020-10-09:
Just a quick question (I hope). Reading the recent [release notes](https://github.com/esp8266/Arduino/releases) and some of the [development traffic](https://github.com/esp8266/Arduino/pull/6566) I understand that there is now a CRC value at the end of bin files and this is checked by the OTA FLASH loader. What I'm not sure about is what happens if this is found to be incorrect. So my question is:

- With recent versions, such as 2.7.4, what does ESPhttpUpdate.update() do if it finds an incorrect CRC? In particular, does it refuse to commit to the new image and retain the original?

The reason for my question is I have about 5,000 IoT devices in the field running 2.4.2 (yes, I know it's old, that's why I am exploring 2.7.4). If any of these units have bad FLASH they blissfully never knew. But if I deploy 2.7.4 will they no longer be able to update if they have bad FLASH? I'm not suggesting this a bad thing, I realize executing a program from bad FLASH is unreliable, but I need to fully understand the impact of making this upgrade before making the commitment.

Many thanks for your time.

Tech-TX, 2020-10-10:
The way I read it, Earle's new CRC check is separate from the OTA upload.  If you read that thread, dok-net did one upload to a different-sized flash than what he'd specified in the compile, and the upload was accepted.  It failed the later ESP.checkFlashCRC() because the flash size was wrong.  When doc-net corrected his board type (flash size) the upload passed AND the ESP.checkFlashCRC() passed.

ESP.checkFlashCRC() is a separate (new) check that you can run to see if your flash has been corrupted.

Earle, if I'm wrong, just [https://www.youtube.com/watch?v=Z8VD4JXUozM](boot me in the head). :D

earlephilhower, 2020-10-10:
Thanks, @Tech-TX !  As you said, the CRC check is optional and not part of OTA.  There is a CRC check on the OTA commands, IIRC, but it's always been there and it's only stored in RTCMEM anyway.

closed 2020-10-10 by earlephilhower

pidloop, 2020-10-10:
Copy that, many thanks gentlemen.