esp8266/Arduino #7651: Exception(2) thrown after OTA update

kamran00f, 2020-10-11:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-01]
- Core Version:			[2020-10-11]
- Development Env:		[Platformio]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[DOUT]
- Flash Size: 		[1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Exception thrown after OTA update. Decoded exception message:
```
Exception 2: InstructionFetchError: Processor internal physical address or data error during instruction fetch
PC: 0x3ffee790
EXCVADDR: 0x3ffee790
```

The exception is thrown after Progress 100% and before printing "End" in <ArduinoOTA.onEnd()> callback.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <errno.h>
#include <ArduinoOTA.h>

int ConnectToWifi(const char *ssid, const char *password, uint32_t timeout_ms)
{
    Serial.printf("Connecting to %s ", ssid);
    WiFi.begin(ssid, password);
    unsigned long ref_millis = millis();
    while(WiFi.status() != WL_CONNECTED)
    {
        Serial.print(".");
        delay(500);

        if(millis() - ref_millis > timeout_ms)
            return -ETIMEDOUT;
    }

    Serial.printf("\r\nConnected! Local IP: ");
    Serial.println(WiFi.localIP());
    return 0;
}

void StartOtaService(char *serverName, char *password, uint16_t port)
{
    ArduinoOTA.setHostname(serverName);
    ArduinoOTA.setPassword(password);
[... 155 lines cut ...]
3fffff80:  00000000 00000000 00000001 4010022c                                  
3fffff90:  3fffdad0 00000000 3ffeed84 3ffeedc4                                  
3fffffa0:  3fffdad0 00000000 3ffeed84 40206500                                  
3fffffb0:  feefeffe feefeffe 3ffe84f4 40100e91                                  
<<<stack<<<                                                                     
                                                                                
--------------- CUT HERE FOR EXCEPTION DECODER ---------------                  
                                                                                
 ets Jan  8 2013,rst cause:1, boot mode:(3,6)                                   
                                                                                
load 0x4010f000, len 3664, room 16                                              
tail 0                                                                          
chksum 0xcc                                                                     
csum 0xcc                                                                       
v00000000                                                                       
~ld                                                                             
      Scanning...                                                   
        Stehmann-guest                                                          
        WIFI                                                                    
        Hulsuls_internet                                                        
        CelenoInitialAP290F4B                                                   
                                                                                
Connecting to WIFI .......                                               
Connected! Local IP: 192.168.1.140     
```

earlephilhower, 2020-10-11:
Please decode the stack dump as requested in the template, using your own executable.  We can't use the raw stack dump or PC/EXECVADDR.

kamran00f, 2020-10-11:
Here is the decoded stack dump:
```
Exception Cause: 2  [InstructionFetchError: Processor internal physical address or data error during instruction fetch]

0x40204d28: MD5Builder::calculate() at ??:?
0x40205964: UpdaterClass::end(bool) at ??:?
0x40203284: WiFiClient::write(unsigned char const*, unsigned int) at ??:?
0x40204dbd: Print::write(char const*) at ??:?
0x40205165: unsigned int Print::printNumber<unsigned long>(unsigned long, unsigned char) at ??:?
0x40201084: std::_Function_handler<void (unsigned int, unsigned int), StartOtaService(char*, char*, unsigned short)::{lambda(unsigned int, unsigned int)#3}>::_M_invoke(std::_Any_data const&, unsigned int, unsigned int) at common.cpp:?
0x40206ae9: __delay at ??:?
0x40204651: ArduinoOTAClass::_runUpdate() at ??:?
0x4020d6e0: esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter::findCachedDomainOffset(void const*, bool) const at ??:?
0x4020922c: esp8266::MDNSImplementation::MDNSResponder::_updateProbeStatus() at ??:?
0x4020d770: esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter::findCachedDomainOffset(void const*, bool) const at ??:?
0x4020d770: esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter::findCachedDomainOffset(void const*, bool) const at ??:?
0x40239c5b: wifi_station_get_connect_status at ??:?
0x40204684: ArduinoOTAClass::handle() at ??:?
0x4020141c: HandleOtaService() at ??:?
0x4020152c: loop at ??:?
0x402082ba: esp8266::MDNSImplementation::MDNSResponder::enableArduino(unsigned short, bool) at ??:?
0x4020d770: esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter::findCachedDomainOffset(void const*, bool) const at ??:?
0x402041fe: ArduinoOTAClass::begin(bool) at ??:?
0x4020d770: esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter::findCachedDomainOffset(void const*, bool) const at ??:?
0x402036f4: std::_Function_base::_Base_manager<std::_Bind<std::_Mem_fn<void (ArduinoOTAClass::*)()> (ArduinoOTAClass*)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at ??:?
0x4020d158: std::_Function_handler<void (), std::_Bind<std::_Mem_fn<void (ArduinoOTAClass::*)()> (ArduinoOTAClass*)> >::_M_invoke(std::_Any_data const&) at ??:?
0x4020d770: esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter::findCachedDomainOffset(void const*, bool) const at ??:?
0x402011ca: std::_Function_base::_Base_manager<StartOtaService(char*, char*, unsigned short)::{lambda(ota_error_t)#4}>::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager<StartOtaService(char*, char*, unsigned short)::{lambda(ota_error_t)#4}> const&, std::_Manager_operation) at common.cpp:?
0x40201402: StartOtaService(char*, char*, unsigned short) at ??:?
0x4010022c: ets_post at ??:?
0x40206500: loop_wrapper() at core_esp8266_main.cpp:?
0x40100e91: cont_wrapper at ??:?
```

kamran00f, 2020-10-11:
Ok, I could narrow down where this is happening. It happens inside crc32(...) inside eboot_command_calculate_crc32(...) in eboot_command_write(...) in UpdaterClass::end(bool evenIfRemaining) inside if (_command == U_FLASH) {...}.

I checked the parameters that are passed to crc32 functions and they seem fine. This is the code:
```
static uint32_t eboot_command_calculate_crc32(const struct eboot_command* cmd)
{
    Serial.println("in eboot_command_calculate_crc32");
    Serial.print("cmd=");
    Serial.println((uint32_t)cmd);
    Serial.print("offset=");
    uint32_t offset = offsetof(struct eboot_command, crc32);
    Serial.println(offset);
    Serial.print("crc32 address=");
    Serial.println((uint32_t)crc32);
    
    uint32_t crc = crc32((const uint8_t*) cmd, offsetof(struct eboot_command, crc32), 0xffffffff);

    Serial.println("after crc32");
    return crc;
}
```
output:
```
in eboot_command_calculate_crc32
cmd=1073741216
offset=124
crc32 address=1073670160
```
But for some reason I didn't get inside crc32 function. So I made a copy of the crc32 function as shown in the following code:
```
static uint32_t local_crc32 (const void* data, size_t length, uint32_t crc)
{
    const uint8_t* ldata = (const uint8_t*)data;
    while (length--)
    {
        uint8_t c = pgm_read_byte(ldata++);
        for (uint32_t i = 0x80; i > 0; i >>= 1)
        {
            bool bit = crc & 0x80000000;
            if (c & i)
                bit = !bit;
            crc <<= 1;
            if (bit)
                crc ^= 0x04c11db7;
        }
    }
    return crc;
}

static uint32_t eboot_command_calculate_crc32(const struct eboot_command* cmd)
{
    Serial.println("in eboot_command_calculate_crc32");
    Serial.print("cmd=");
    Serial.println((uint32_t)cmd);
    Serial.print("offset=");
    uint32_t offset = offsetof(struct eboot_command, crc32);
    Serial.println(offset);
    Serial.print("crc32 address=");
    Serial.println((uint32_t)crc32);
    
    uint32_t crc = local_crc32((const uint8_t*) cmd, offsetof(struct eboot_command, crc32), 0xffffffff);

    Serial.println("after crc32");
    return crc;
}
```
And after this change, the OTA worked perfectly fine! So something is going wrong while calling crc32 function, but I could not figure out what.

earlephilhower, 2020-10-12:
Your MCVE is not failing for me.  I had to fix several compile errors, but after doing so I was able to upload 1st via serial port and then via Arduino OTA from the IDE using the "Auth" password.  I never got PlatformIO working w/my own setup so can't try it out, but it should use espota.py, same as the IDE to do the upload.

Are you sure it fails with just the MCVE, and doesn't need your full sketch uploaded to crash?

The stack dump doesn't really shed much light, but the faulting PC = **inside RAM ** which is a bad thing.

This could indicate stack overflow (use Tools->Stack Protection->Enabled to turn on the runtime check...it's not perfect, but it can catch some things).  It could also be RAM corruption somewhere along the way and a VTABLE/ptr. that was going into flash got overwritten with a 0x3fff(ram) address.

kamran00f, 2020-10-12:
I tested the MCVE and OTA works again perfectly fine. I found the problem in my full sketch. I had a global object called crc32! The same name as the function name with problem. Sorry for bothering you and thank you @earlephilhower for your help. Closing this issue...

closed 2020-10-12 by kamran00f

lclem, 2022-11-06:
wow this is crazy, such issue should be get caught at compile time!

mcspr, 2022-11-06:
> I had the same issue... wow this is crazy, such an issue should be get caught at compile time!

Do you have another example, or it is still this crc32 issue?
Is this a problem with Arduino IDE or PlatformIO (as above)?
Do we have some kind of link-order issue so crc32 from local build overrides crc32 from framework?

lclem, 2022-11-06:
I had this on Arduino IDE.