esp8266/Arduino #7654: ethernet library issue with W5500

infrafast, 2020-10-15:
# Basic Infos

- [ x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ x] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12
- Core Version:			2.6.3 and also 2.7
- Development Env:		Arduino IDE 1.5
- Operating System:		Windows

### Settings in IDE

- Module:  			Wemos D1 mini r2
- Flash Mode:			
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		
- Flash Frequency:	[40Mhz]
- CPU Frequency:		80Mhz
- Upload Using:		both OTA and SERIAL
- Upload Speed:   115200

### Problem Description

I am facing an issue with trying to implement a WP5500 an ESP12 module
The business case is that I need a very small scale device having  both WIFI and ethernet connectity because of mechanical and other constraints.

As far as I know I should use the the Ethernet library from Arduinom which I tried. 
However looks like the implementation of this library for the ESP8266 is a bit behind other board as it does't allow to for instance use CS pin (using init method) and lacks a few other method like EthernetLinkStatus

Configuration as following:

- ESP8266 IDE package: 2.6.3 (also tested with latest 2.7.x)
- ESP8266 model: D1 mini pro noname (wemos like)
- pinout as following:
D0	GPIO16	4	
D1	GPIO5	20	
D2	GPIO4	19	CS
D3	GPIO0	18	
D4	DGPIO2	19	
D5	GPIO14	5	SCLK
D6	CPIO12	6	MISO
D7	GPIO13	7	MOSI
D8	GPIO15	16	 
I kept D5,D6,D7 as the standard SPI config (https://arduino-esp8266.readthedocs.io/en/latest/libraries.html#spi) and only use D2 as CS

Header of ethernet.h from last github repo: https://github.com/arduino-libraries/Ethernet
	static int maintain();
	static EthernetLinkStatus linkStatus();
	static EthernetHardwareStatus hardwareStatus();

	static void begin(...);
	static void init(uint8_t sspin = 10);

	static void MACAddress(uint8_t *mac_address);
	static IPAddress localIP();
	static IPAddress subnetMask();
	static IPAddress gatewayIP();
	static IPAddress dnsServerIP() { return _dnsServerAddress; }

	void setMACAddress(const uint8_t *mac_address);
	void setLocalIP(const IPAddress local_ip);
	void setSubnetMask(const IPAddress subnet);
	void setGatewayIP(const IPAddress gateway);
	void setDnsServerIP(const IPAddress dns_server) { _dnsServerAddress = dns_server; }
	void setRetransmissionTimeout(uint16_t milliseconds);
	void setRetransmissionCount(uint8_t num);

	etc...

Header of ethernet.h from ESP8266 board (for arduino IDE 1.5):
  
  only:
  
  void begin(...);
  int maintain();
  IPAddress localIP();
  IPAddress subnetMask();
  IPAddress gatewayIP();
  IPAddress dnsServerIP();

 I tried to :
 - use the ESP8266 native library from the package "as is" but the program is just pending at 
	if (Ethernet.begin(mac) == 0)
	and never continues...(no reset no errors)
 
 - force usage of the most recent library but when compiling I get followinging error message. I could try to but I am a bit afraid that by doing that I will pull a things that will never end:
 W5100.setSubnetMask(subnet._address.bytes);

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <Ethernet.h>

byte mac[] = {
  0x00, 0xAA, 0xBB, 0xCC, 0xDE, 0x02
};

void setup() {
if (Ethernet.begin(mac) == 0)
}

void loop() {
 // void
}

```

### Debug Messages

no messages
Debug messages go here
```

d-a-v, 2020-10-15:
Could you try [git version](https://arduino-esp8266.readthedocs.io/en/2.7.4_a/installing.html#using-git-version) of this core ?
Or installable [beta (0.0.1) version](https://d-a-v.github.io/esp8266/Arduino/index.html) ?

Both include https://github.com/arduino-libraries/Ethernet version.
Alternatively, you could try #6680

infrafast, 2020-10-15:
thanks a lot for your fast reply dav, I am giving it a try now, I may need time to debug and make it work before I feedback you.
just one thing  already observed:  ESPhttpUpdate.update protoype has changed...as my code don't compile anymore: it require now a WifiClient as input. Is there any available doc  or any hint on that new implementation ? Cheers

d-a-v, 2020-10-29:
I can see in your code that you use D2 as SlaveSelect/ChipSelect gpio.
Although there is no mention of SS/CS pin for the TFT screen in your code, it says D1 in logs.
It also says "27"(MHz?) as SPI frequency.

In Ethernet code, we have:
```
// Safe for all chips
#define SPI_ETHERNET_SETTINGS SPISettings(14000000, MSBFIRST, SPI_MODE0)`
```

Frequency in ethernet is initialized once. Maybe you should try and lower your TFT SPI frequency initialization (27->14?).

Anyway
- This is not the place to ask about external drivers or libraries (TFT)
- Ethernet was historically copied and locally adapted, and is now replaced by a submodule to have up-to-date drivers (more boards, debugged code). It is not our code anymore. As submodule, we can't update their sources without submitting a pull-request on their repository.
- You may also try with #6680, that we control, in which SPI is externally configured, and which offers better integration with libraries involving network.

jandrassy, 2020-10-29:
Ethernet library uses SPI transactions and applies the settings before every use of the bus

d-a-v, 2020-10-29:
@jandrassy ah right !
I was looking only in w5100.cpp in which mainly `setSS()` .. `resetSS()` is used.
Then @infrafast I personally have no idea about what can be wrong between TFT lib and Arduino's Ethernet lib...
... wait, you have this in your logs:
```
12:25:51.784 -> Transactions = No  <-- must be yes
12:25:51.784 -> SPI overlap  = No  <-- what is it ?
```

infrafast, 2020-10-29:
Thank you both, the author of TFT-espi provided good feedback here: [](https://github.com/Bodmer/TFT_eSPI/issues/811)
I will perform additionnal test with 6680 and keep you posted here

d-a-v, 2020-10-29:
You may add some `yield()` in your code, that should help FW internals including WiFi processing.

earlephilhower, 2020-12-24:
#6680 is merged and Ethernet is now a submodule.  Closing this as I believe all is fixed and 2 months w/o updates.

closed 2020-12-24 by earlephilhower