esp8266/Arduino #7660: OTA update reliability

smurfix, 2020-10-17:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					ESP-01
- Core Version:			as installed by platformio
- Development Env:		Platformio
- Operating System:		Debian

### Settings in IDE

- Flash Mode:			qio
- Flash Size: 		1MB
- Reset Method:		n/a
- Upload Using:		OTA

### Problem Description

I'm running into strange OTA reliability issues.

* Flashing w/ serial: success
* Flashing via HTTP using Mozilla: success
* Flashing via TCP port 8266 using `espota.py`: fails
* Flashing via HTTP using `curl -F firmware=@file.bin IP/update`: fails

**fails** means that the ESP-01 flashes something and reboots, but then either doesn't boot or crashes when it tries to set up Wifi.

The above is 100% reproducible.

kickshawprogrammer, 2021-01-10:
For what it’s worth, I’m having this problem too. ESP-01, visual studio code (not sure of Arduino extension version), Arduino IDE 1.8.10 (I think), ESP library (not sure if that’s what it’s called) version 2.7.4 

Flash via serial always works. OTA while in serial programmer works more reliably than OTA in application’s location (my attic) but neither case is it anywhere near 100%. 

Lights blink. ESP-01 resets (sometimes) update didn’t take. Source code is available if needed. Maybe I’m doing something wrong?

juggledad, 2021-01-28:
Don't know if this will help (new to OTA) but I just read this:

> The only disadvantage to OTA is that you have to explicitly add the code for it to every sketch you upload. You also need a flash chip that is twice the size of your sketch, so it won't work for 512kB boards. (It has to download the new sketch while still running the old code.)

kickshawprogrammer, 2021-02-02:
Thanks for the info. I had already come across that bit of information tho. Anyway, it appears that my issue was power related. After adding a 100uF capacitor across vcc  and ground, my reliability rate shut up dramatically. For the ones that do fail, it was always an error code 6 (steam read error), where I just restart the update process again.

closed 2021-03-23 by earlephilhower