esp8266/Arduino #7663: ESP-01 Webserver Issue

Motarded603, 2020-10-18:
Hey all,
I'm in the works of making a WiFi IR remote and app. I currently have a NodeMCU running a webserver that works perfectly paired with a custom Android app and it all works flawlessly.

However, I'm trying to make another setup with an ESP-01 with the same webserver arduino code I made, and the webserver is running when plugged into my computer and I can connect to it. My issue is when I have the ESP-01 powered by a 3.3V/5V Power Supply Module, the webserver never launches. I checked to make sure it's supplying 3.3v to the ESP-01, which it is. Any other thoughts of why the webserver won't work when not connected to my computer?

closed 2020-10-19 by devyte

Motarded603, 2020-10-19:
@devyte why did you closed this if no one has given any advice? 
Did I post this to the wrong forum? If so, could you point me to the right one? Thanks.

devyte, 2020-10-20:
Correct: wrong forum. This is an issue tracker, not meant for general help requests. 
The reason, and the direction, is in the issue template, which you saw and ignored when opening this. You should ask at e. g. esp8266.com or stackoverflow.

Motarded603, 2020-10-20:
Tried stackoverflow, they shut me down. I tried esp8266 and they also shut me down. I'm tired of running around forums and everyone shutting me down. I just don't understand where to ask this question. I'm not trying to be a dick, just getting annoyed and frustrated because I'm being told "try this forum" and being told the same thing.

devyte, 2020-10-20:
As a maintainer here I can firmly say this is the wrong place. Such discussions belong at a community forum, and esp8266.com and stackoverflow are precisely that. I don't know how you asked there, maybe wrong section, or wrong wording, or wrong person answering, but that's where you need to ask.

jandrassy, 2020-10-20:
@devyte, stackoverflow is about codding. 
Arduino Stack Exchange or Electronics Stack Exchange are the right 'stack' sites for this.

@Motarded603 , where on esp8266 forum is your post?

JiriBilek, 2020-10-20:
@Motarded603 , first make sure it is not a power supply issue. The ESP is quite hungry, it can easily draw hundreds of mA when using radio.

Motarded603, 2020-10-20:
@JiriBilek Thanks for the reply. 
I'm using a [Power Supply module](https://www.amazon.com/JBtek-Breadboard-Supply-Arduino-Solderless/dp/B010UJFVTU/ref=asc_df_B010UJFVTU/?tag=hyprod-20&linkCode=df0&hvadid=194024095585&hvpos=&hvnetw=g&hvrand=7132559912894113584&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9002305&hvtargid=pla-314833700048&psc=1), which is supplying 3.3v steady, but I wasn't able to check the current as the multimeter I was using didn't support DC current, only AC. If it's the power supply, what's the best way in powering these boards? I have a few [micro usb header boards](https://www.amazon.com/gp/product/B0183KF7TM/ref=ppx_od_dt_b_asin_title_s01?ie=UTF8&psc=1) and waiting on [5v to 3.3v converters](https://www.amazon.com/gp/product/B08CDMZMDN/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) that I was going to use to hopefully power it as well. I only used the power supply module for a breadboard test.

Motarded603, 2020-10-20:
@jandrassy Not sure where my post ended up, I think a mod deleted it or something. The post was completely the same as this one, just in IoT forums.

JiriBilek, 2020-10-20:
@Motarded603 The power supply you are using is ok. I am using it frequently, too. You could try to add 10uF tantalum capacitor on the 3v3 line (near the ESP module or on the module) but I doubt it'll help. Sorry.