esp8266/Arduino #7665: Present boards in alphabetical order in "Board manager"

mats-nk, 2020-10-21:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:			All
- Core Version:			2020-10-21
- Development Env:		Arduino IDE
- Operating System:		All

### Settings in IDE

"Board manager"

### Problem Description

I have found it hard to always scan the list of boards in the board manager for the board I'm currently working with since they are not in alphabetical order.

So I split the "Arduino/tools/boards.txt.py" into one file per board and then the start and end info into separate files.

boards.zip contains all the "one file per board".
[boards.zip](https://github.com/esp8266/Arduino/files/5416432/boards.zip)

build.txt ->build.bat builds a new version of boards.txt.py with the boards in alphabetical order.
[build.txt](https://github.com/esp8266/Arduino/files/5416446/build.txt)

I would appreciate if the boards could be presented in an alphabetical order highly.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

### Debug Messages

d-a-v, 2020-10-21:
Why don't you make a pull request with an updated version of `boards.txt.py` in which you have alphabetically sorted the board description blocks instead ?

mhightower83, 2020-10-21:
@mats-nk I have a problem with searching any long list. My solution was to add and use the [--filter option on boards.txt.py](https://arduino-esp8266.readthedocs.io/en/latest/faq/a05-board-generator.html?highlight=filter#how-to-create-an-abridged-boards-txt-file). This option will also place the boards in the order they appear in the filter file. 

So you can dump a list of the boards with:
```
./tools/boards.txt.py --boardnames >favorites.txt
```
arrange and delete boards as needed. This would also allow you to group boards in any order. Then:
```
./tools/boards.txt.py --boardsgen --filter favorites.txt
```
I don't remember how I positioned the generic boards. They are special and always added even if you leave them out of the `favorites.txt`.

mhightower83, 2020-10-21:
@mats-nk Well, I just got back to my system and looked at this again. :frowning_face: The boards.txt is ordered to match the `favorites.txt` file; however, the version of Arduino IDE 1.8.9 that I am now using doesn't honor the order when it presents the list. I am sure this use to work. Otherwise, I would not have added it to the doc. :frowning_face:

mats-nk, 2020-10-22:
Neat that was a nice way, I'm not a python guy so I didn¨t see that one in the ´boards.txt.py´ .

Then I would suggest that the boards.txt that is built and delivered through the "json" is built with the flag `--boardnames`.

mats-nk, 2020-10-22:
And a thought, is it possible to make `boards.txt.py` more generic, example to split out the boards from it and place all board definition in a separate file. In that way it could also be used in the "Arduino" project, or any other hardware boards.

mats-nk, 2020-10-22:
@mhightower83 thx for that info, I have never heard of favorites.txt and searching for it was not easy, but I found:

[https://arduino-esp8266.readthedocs.io/en/latest/faq/a05-board-generator.html](https://arduino-esp8266.readthedocs.io/en/latest/faq/a05-board-generator.html)

But that doc needs some proof reading, like `AppDataLocalArduino15packagesesp8266hardwareesp82662.4.2for PlatformIO it is in .platformiopackagesframework-arduinoespressif8266.` . I think I can guess what it is, but without \ and / not so easy

mats-nk, 2020-10-22:
So I tried the above suggestions with 

> ./tools/boards.txt.py --boardnames >favorites.txt
>arrange and delete boards as needed. This would also allow you to group boards in any order. Then:
>./tools/boards.txt.py --boardsgen --filter favorites.txt

And it works, I get the subset that I specified in favorites.txt + "Generic ESP8285 Module", but the sorting I had to do manually in `favorites.txt`.

I just specified :

```
# Available board names. Delete or comment out the boards you do not need:
d1_mini              # WeMos D1 R2 & mini
generic              # Generic ESP8266 Module
```

d-a-v, 2020-10-22:
Sorting could be automated

mats-nk, 2020-10-22:
@d-a-v Sorting would be highly appreciated from me.

Is it hard to split out the board definitions in a separate file, like boards.def, because then boards.txt.py could be used with the AVR MCU and other MCUs.

d-a-v, 2020-10-23:
@mhightower83 it seems the order given by `boards.txt` is preserved in the Arduino IDE (I tested with 1.8.14).
@mats-nk 
Sorting can be done like this:
```
./tools/boards.txt.py --boardnames | sort > favorites.txt
```
(it should also work under windows, I just discovered `sort` is there too).
Does this work with you ? Should it be enough to document this ?

mats-nk, 2020-10-23:
As I have repeated earlier, I think it would be appreciated to have the boards sorted in alphabetical order in the release, NOT something that you have to do manually after finding something in the doc and have installed python.

Lets have the average user in focus.

**And thanks**, I didn't know that `sort` worked in the Windows cli, Linux bash is more my tool.

d-a-v, 2020-10-23:
#7670 is a proposal for automating board sorting, leaving generic boards always first.
[link to new default boards.txt](https://github.com/esp8266/Arduino/blob/5a96d80c9a1481ce89db93627bff924e4addb414/boards.txt)

Is that OK ?

PR #7670 (2020-10-24): boards.txt: sort by displayed name
fixes #7665
closed 2020-10-24 by earlephilhower