esp8266/Arduino #7667: Nodemcu Esp8266 , Send HTTP Request returns -1 , " connection Fails "

EngRanya, 2020-10-23:
Hi All
Please I need Help 
I try to send HTTP Request but it fails 
HTTP Code returns -1 
Although the URL worked successfully on Postman and the Wifi successfully connected
I'm using WIFI manager and it works sucessfully connected to the network 

Note : mysite2020 is an example , but my host worked well on POSTMAN and data posted in Database
Here is my code 

```

#include <OneWire.h>
#include <DallasTemperature.h>
#include <ESP8266WiFi.h>
#include <WiFiManager.h>
#include <ESP8266HTTPClient.h>
#include <Wire.h>

IPAddress staticIP750_100(192,168,1,16);
IPAddress gateway750_100(192,168,1,1);
IPAddress subnet750_100(255,255,255,0);

   HTTPClient http; 

#define ONE_WIRE_BUS 2 // DS18B20 on NodeMCU pin D4 

OneWire oneWire(ONE_WIRE_BUS);
DallasTemperature DS18B20(&oneWire);

float temp_0;
  const char* host = "http://mysite2020.info";
  

void setup() 
{
 

  Serial.begin(115200);
   WiFiManager wifimanger ;

  DS18B20.begin();

Wire.begin(D2, D1)
 
 
  WiFi.mode(WIFI_STA);

  wifimanger.autoConnect("Inlet Device","12345");
  
  while ((!(WiFi.status() == WL_CONNECTED))){
    delay(300);
    Serial.println("...");

  }

  Serial.println("WiFi connected");  

  WiFi.config(staticIP750_100, gateway750_100, subnet750_100);
  WiFi.hostname("Inlet Device") ;
  
delay(3000);
Serial.println("Welcome To Device No : 1");

  Serial.println("IP address: ");
    Serial.println(WiFi.localIP());

 
    Serial.println("Your Gateway is : ");

    Serial.println((WiFi.gatewayIP().toString().c_str()));

  Serial.println("Your Hostname is : ");

  Serial.println((WiFi.hostname()));

delay(3000);

   Serial.println("Welcome to my  Project!");

 delay(3000);

}
void loop() {
  
Serial.println(host); // Use WiFiClient class to create TCP connections 
WiFiClient client; 
const int httpPort = 80; 
if (!client.connect(host, httpPort)) 
{ 
Serial.println("connection failed"); 
return; 
} 
Serial.print("Requesting URL: "); 

sendtemp ();
}

void sendtemp ()
{

 temp_0=23.26;
 String  url = "http://mysite2020.info/Api/insert_mssqlserver.php?Reading=" + String(temp_0);
 
  Serial.println(url);
    http.begin(url); 
http.addHeader("Content-Type", "application/x-www-form-urlencoded"); 
auto httpCode = http.GET(); 
Serial.println(httpCode); //Print HTTP return code 
String payload = http.getString(); 
Serial.println(payload); //Print request response payload 
http.end(); //Close connection Serial.println(); 
Serial.println("closing connection");

  delay(500);

}
```

EngRanya, 2020-10-23:
any help ??

jandrassy, 2020-10-23:
this is not the right place for this

earlephilhower, 2020-10-23:
As @jandrassy and the issue template you ignored requested, please use https://esp8266.com or https://gitter.im/esp8266/Arduino for general questions like this.

closed 2020-10-23 by earlephilhower