esp8266/Arduino #7677: When using the ssd1306 and pcf8574 chips, the pcf8574 communication failure will occur if i2c communication is used at the same time

linqingfeng, 2020-10-25:
Hi,

When using the ssd1306 and pcf8574 chips, the pcf8574 communication failure will occur if i2c communication is used at the same time.
And this error only appears after upgrading to version 2.7.0 or higher. The code that caused the error is in the following source code. If you replace the code back to version 2.6.3, there is no problem.

/esp8266/Arduino/blob/master/cores/esp8266/core_esp8266_si2c.cpp

v2.6.3
```
void ICACHE_RAM_ATTR Twi::busywait(unsigned char v)
{
    unsigned int i;
#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Wunused-but-set-variable"
    unsigned int reg;
    for (i = 0; i < v; i++)
    {
        reg = GPI;
    }
    (void)reg;
#pragma GCC diagnostic pop
}
```

d-a-v, 2020-10-25:
@Tech-TX

T-RK, 2020-10-25:
Hello

First, there are a few things you should be sure of to use the i2c bus for different devices.
One of them is to check that the i2c bus is idle. In the previous version it was working because it checked and then started to write data. In some cases, we will use it without checking, but if you add delay (1) as an alternative, the problem will be solved.
In the meantime, make sure that both devices have entered their DeviceIDs correctly.

Best Regards

Jason2866, 2020-10-25:
There is no general problem. I use a ssd1306 and the pcf8574 with Tasmota. Works well.
Tasmota uses core 2.7.4 (with PWM PR 7231).

Tech-TX, 2020-10-28:
You must be doing something odd,  as I've used both ssd1306 display and port expander on the same bus.  I'll need more than a vague description to figure out what you're trying to do, and what's breaking.  OLED can be SPI or I2C,  for starters.

earlephilhower, 2020-10-28:
Given the multiple reports of success with the chip from others, the lack of MCVE, etc., I'm closing this as it doesn't seem to be core related.  @linqingfeng you can use https://esp8266.com or https://gitter.im/esp8266/Arduino for help with your specific project.

closed 2020-10-28 by earlephilhower

Tech-TX, 2020-10-31:
His problem description makes no sense: the ESP8266 isn't multi-core, so there's no such thing as two different processes running at the same time, it's one or the other.

@linqingfeng Probably relevant, same exact parts: https://github.com/esp8266/Arduino/issues/7327
Writing a clock speed of 400KHz used to actually produce 313KHz, and now it's 363KHz.
Simple solution: change the clock speed in the Adafruit library to work with your PCF8574.  Looking at their current library, it's an optional constructor for calling their library (line 129 of the .h file).  100KHz is guaranteed, anything above that you're lucky if it works.

I looked around and see a problem with the Adafruit library.  They're blindly switching the bus up to 400KHz (regardless of what it was) and then blindly setting it back to 100KHz when they're done.  That's a Bad Idea if you talk to anyone that's done a lot of work in I2C.  In general, "_the bus runs at the speed of the slowest device_".  If you put a 400KHz SSD1306 display on the same bus as a 100KHz PCF8574, then erratic operation can occur when/if the slower speed device misinterprets the high speed address/data.  I tried something like that ~ 25 years ago and it didn't work then, I had to drop the bus speed or split the bus.  If  that's what linqingfeng was attempting then all bets are off, as the bus timing changed significantly when I modified that busywait.  400KHz was **quite a bit** slower before, and is closer to true now.  That alone could be causing the issue he/she saw, as both Philips/NXP and TI only spec their parts to 100KHz.  They **might** run at 200 or 250K, but if they were reliable at 400KHz then the manufacturer(s) would have rated them for Full Speed operation.

Jason2866, 2020-10-31:
@Tech-TX  That would explain it. We do NOT use a library for the pcf8574. We use our own code in Tasmota. Our experience many Adafruit libraries have quirks when using together with other devices and drivers...

alirezaimi, 2021-06-23:
Hi
Any solution for this ?