esp8266/Arduino #7678: ESP8266 WiFiClientSecure connection drops immediately on GET, error not decoded

dalbert2, 2020-10-27:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [N/A] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:		        [ESP-12E]
- Core Version:			[Core 5.0.1 Espressif8266 2.6.2]
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[AIThinker]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [921600] (serial upload only)

### Problem Description

I am experiencing a problem with the ESP8266 WiFiClientSecure where it connects to my site, but drops the connection the moment the GET request is sent.  The exact same code will work for other sites (e.g. www.google.com/index.html) but not this specific site.  I have tested using fingerprint, trusted root certificate, and inscure modes; all have the same result: successful SSL connection, but connection lost the moment any HTTP GET is sent.  The site works properly when tested with openssl (openssl s_client -crlf -connect tenetics.com:443) or from any browser.  

The SSL error returned (296) is not decoded successfully so debugging is difficult.   Any guidance would be appreciated; the site is open so you can issue the Additional error handling is requested to provide useful feedback as to why a connection is dropped.  More documentation will help too (I will be happy to write it once this is solved). 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <WiFiClientSecure.h>

char ssid[] = "myssid";
char psk[]  = "mykey";

void ICACHE_FLASH_ATTR setup()
{	
	Serial.begin(115200);

	// Start WiFI network
	WiFi.mode(WIFI_STA);
	delay(2000);
	WiFi.begin(ssid,psk); 
    while (WiFi.status() != WL_CONNECTED) {
    	delay(100);
  	}
  	delay(1000);

	WiFiClientSecure client;
	client.setInsecure();

	HTTPClient http;
	// works
	//http.begin(client, "https://www.google.com/index.html");

	// doesn't work but works with: openssl s_client -crlf -connect tenetics.com:443
	http.begin(client, "https://www.tenetics.com/index.html");

	int httpResult = http.GET();
	if (httpResult == HTTP_CODE_OK) {
		Serial.println(http.getString());
	} else {
		Serial.printf("HTTP error: %d\r\n", httpResult);
	}
	char sslErrorMsg[80];
	int sslError = client.getLastSSLError(sslErrorMsg, sizeof(sslErrorMsg));
	if (sslError) {
		Serial.printf("SSL error: %d: %s\r\n", sslError, sslErrorMsg);
	}
	http.end();
}

void ICACHE_FLASH_ATTR loop()
{
	yield();
}

```

### Debug Messages

```
 BSSL:Connected!
          [HTTP-Client] connected to www.tenetics.com:443
                                                         [HTTP-Client] sending request header
             -----
                  GET /index.html HTTP/1.1
Host: www.tenetics.com
User-Agent: ESP8266HTTPClient
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
Connection: keep-alive
Content-Length: 0

-----
     [HTTP-Client][returnError] error(-5): connection lost
                                                          HTTP error: -5
SSL error: 296: Unknown error code.
[HTTP-Client][end] tcp is closed
```

earlephilhower, 2020-10-27:
Please enable full SSL debugging in the menu and capture the output.  It's pretty chatty and may point to the exact issue in the logs.

earlephilhower, 2020-10-27:
Also, if you're really on core 2.6.2, can you please upgrade to the latest release, 2.7.4.  It should be available thru the normal channels and has quite a few bugfixes from 2.6.x.

dalbert2, 2020-10-27:
I'll give it a try; 2.6.2 is the newest core that's available through platformio.
I'll try overwriting the 2.6.2 files with 2.7.4.  
Thanks again!

earlephilhower, 2020-10-27:
Looks like it connects and then the other end times out.  Are you running at 160MHZ?  OTW, the SSL handshake may take too long and GOOG will drop the connection.

earlephilhower, 2020-10-27:
Also, if 2.6.2 is really the latest PIO release we need to figure out why that is and fix it...2.7.x came out months ago.

dalbert2, 2020-10-27:
I'm running at 160MHz and I don't think it's that sort of problem because I can connect to many other sites (google, yahoo, etc.).  My site isn't just broken because I can connect to it with browsers and from the CLI with openssl.
So I think this is an incompatibility between BearSSL and my site (running apache2), but the unsupported error code makes it hard to figure out.  

As you suggest, the disconnect is instant once the server receives the full GET request.  If I hold off on sending the final \r\n, it will keep the connection open until it receives it, but then the disconnect is instant before any response is received.

earlephilhower, 2020-10-27:
I'm not sure what the exact error is w/SSL, but from a quick glance I see that when bit 8 is set (256) then BSSL is saying the remote side sent a SSL fatal alert:
https://bearssl.org/gitweb/?p=BearSSL;a=blob;f=inc/bearssl_ssl.h;h=e91df47556905512f8c8850a27e280c384d8b149;hb=dda1f8a0c46e15b4a235163470ff700b2f13dcc5#l219

If it's your own site, can you enable debugging on the webserver and see if there's anything in the logs.  Once the SSL handshake finishes (`BSSL:Connected!`) comms are very simple using AES encrypted packets.  So if it was a SSL-specific thing, I'd expect the handshake to fail and not after connection.

Another thing we can add is HTTP client debugging (you can just enable everything in the debug menu) which would verify that it's not a HTTP-level issue vs. a SSL one.

dalbert2, 2020-10-27:
Good eyes!  I saw the error codes but they appeared to be an enum rather than a bitmap and I didn't look closely enough; the 256 is added to the error code.  Unfortunately that still leaves me with an error code of 40 which is also un-mapped.

I'm sorry for the dumb question, you've mentioned the debug menu several times, and I've spent quite a bit of time looking for it, can you give me a pointer?  From what I've seen, all debugging in PlatformIO is enabled by build flag defines in platformio.ini

dalbert2, 2020-10-27:
From the apache error log:
'''
[Tue Oct 27 21:26:45.477215 2020] [ssl:error] [pid 10333] [client 72.81.142.2:61447] AH02261: Re-negotiation handshake failed
[Tue Oct 27 21:26:45.477300 2020] [ssl:error] [pid 10333] SSL Library Error: error:14094153:SSL routines:ssl3_read_bytes:no renegotiation
'''

earlephilhower, 2020-10-27:
> ```
> [Tue Oct 27 21:26:45.477215 2020] [ssl:error] [pid 10333] [client 72.81.142.2:61447] AH02261: Re-negotiation handshake failed
> [Tue Oct 27 21:26:45.477300 2020] [ssl:error] [pid 10333] SSL Library Error: error:14094153:SSL routines:ssl3_read_bytes:no renegotiation
> ```

I think that's your culprit.  SSL3 was deprecated many years ago because it was busted:  https://tools.ietf.org/html/rfc7568 .  BearSSL doesn't support it, either.

If that's the case and your server is still configured to try SSLv3, this is expected behavior...

dalbert2, 2020-10-27:
I think the SSL3 is coming from the client (ESP8266); the server only supports TLS 1.x (not SSL3).  I've tested and cannot connect to it using ssl3 and the ssllabs.com report also shows ssl3 is disabled.

earlephilhower, 2020-10-27:
BearSSL has never had SSL3 support, and it can't negotiate it ever.  It was written well after the SSL3 deprecation.  So if your apache isn't configured to use it, then maybe it's a false lead.  I think it's time to start digging thru the BSSL sources https://bearssl.org/gitweb/?p=BearSSL;a=summary and see where the undocumented error level `40` is coming from...

Also, if you have wireshark, it's quick work to verify the actual TLS level and cipher used with it, just for sanity's sake.  I don't think it can decode anything to do w/renegotiations, though.

dalbert2, 2020-10-27:
There is an error code 40: BR_ERR_X509_EXTRA_ELEMENT (Decoding error: extraneous element.)

dalbert2, 2020-10-27:
Note that in WiFiClientSecureBearSSL.cpp, method getLastSSLError() should probably handle the 256 and 512 values that can apparently be added to other error enumeration codes.

earlephilhower, 2020-10-27:
Agreed about the errors.  I didn't think of looking in the X509 decoder, either.

So that would lead me to believe there's something fishy with the cert exchange.  I'm a little fuzzy on multi-tenant SSL, is your server shared with multiple SNI names (hence the initial handshake and then another handshake)?

dalbert2, 2020-10-27:
When using the fingerprint mode instead of insecure, there's a little more info (and it confirms the error 40):

```
[HTTP-Client][begin] url: https://www.tenetics.com/index.html
                                                             [HTTP-Client][begin] host: www.tenetics.com port: 443 url: /index.html
                                 [HTTP-Client][begin] httpsFingerprint: E2:05:15:7B:B5:39:AB:C0:AE:C4:03:F9:48:CF:BD:74:41:30:57:89
                                 [HTTP-Client][sendRequest] type: 'GET' redirCount: 0
                                                                                     [hostByName] request IP for: www.tenetics.com
                                [hostByName] Host: www.tenetics.com IP: 104.131.223.32
                                                                                      State:    sending Client Hello (1)
                      State:    receiving Server Hello (2)
                                                          State:        receiving Certificate (11)
                                                                                                  === CERTIFICATE ISSUED TO ===
                             Common Name (CN):          tenetics.com
                                                                    Organization (O):           Tenetics, LLC
           Organizational Unit (OU):    Ceres
                                             Location (L):                      Columbia
                                                                                        Country (C):                      US
                          State (ST):                   Maryland
                                                                Basic Constraints:              critical, CA:FALSE, pathlen:10000
                               Key Usage:                       critical, Digital Signature, Key Encipherment
           Subject Alt Name:            tenetics.com www.tenetics.com
                                                                      === CERTIFICATE ISSUED BY ===
 Common Name (CN):              Sectigo RSA Extended Validation Secure Server CA
                                                                                Organization (O):         Sectigo Limited
                       Location (L):                    Salford
                                                               Country (C):                     GB
                                                                                                  State (ST):                     Greater Manchester
                                                  Not Before:                   Tue Jul  9 00:00:00 2019
      Not After:                        Thu Jul 22 23:59:59 2021
                                                                RSA bitsize:                    2048
  Sig Type:                     SHA256
                                      === CERTIFICATE ISSUED TO ===
                                                                   Common Name (CN):            Sectigo RSA Extended Validation Secure Server CA
                                              Organization (O):         Sectigo Limited
                                                                                       Location (L):                      Salford
                               Country (C):                     GB
                                                                  State (ST):                   Greater Manchester
                Basic Constraints:              critical, CA:TRUE, pathlen:0
                                                                            Key Usage:            critical, Digital Signature, Key Cert Sign, CRL Sign
                                                    === CERTIFICATE ISSUED BY ===
                                                                                 Common Name (CN):                USERTrust RSA Certification Authority
                                                     Organization (O):          The USERTRUST Network
   Location (L):                        Jersey City
                                                   Country (C):                 US
                                                                                  State (ST):             New Jersey
                  Not Before:                   Fri Nov  2 00:00:00 2018
                                                                        Not After:                Tue Dec 31 23:59:59 2030
                        RSA bitsize:                    2048
                                                            Sig Type:                   SHA384
                                                                                              === CERTIFICATE ISSUED TO ===
                         Common Name (CN):              USERTrust RSA Certification Authority
                                                                                             Organization (O):            The USERTRUST Network
                                             Location (L):                      Jersey City
                                                                                           Country (C):                   US
                          State (ST):                   New Jersey
                                                                  Basic Constraints:            critical, CA:TRUE, pathlen:10000
[... cut ...]
Content-Length: 0

-----
     Alert: handshake failure
                             Error: SSL error 40
                                                Alert: unexpected message
                                                                         Alert: close notify
                                                                                            [HTTP-Client][returnError] error(-5): connection lost
                                               HTTP error: -5
[HTTP-Client][end] tcp is closed
```

dalbert2, 2020-10-27:
For completeness, the same error takes place when using the rootCA certificate for verification:
CODE:
```
// Sectigo RSA Organization Validation Secure Server CA PEM
const char trustRoot[] PROGMEM = R"EOF(
-----BEGIN CERTIFICATE-----
MIIF3jCCA8agAwIBAgIQAf1tMPyjylGoG7xkDjUDLTANBgkqhkiG9w0BAQwFADCB
iDELMAkGA1UEBhMCVVMxEzARBgNVBAgTCk5ldyBKZXJzZXkxFDASBgNVBAcTC0pl
cnNleSBDaXR5MR4wHAYDVQQKExVUaGUgVVNFUlRSVVNUIE5ldHdvcmsxLjAsBgNV
BAMTJVVTRVJUcnVzdCBSU0EgQ2VydGlmaWNhdGlvbiBBdXRob3JpdHkwHhcNMTAw
MjAxMDAwMDAwWhcNMzgwMTE4MjM1OTU5WjCBiDELMAkGA1UEBhMCVVMxEzARBgNV
BAgTCk5ldyBKZXJzZXkxFDASBgNVBAcTC0plcnNleSBDaXR5MR4wHAYDVQQKExVU
aGUgVVNFUlRSVVNUIE5ldHdvcmsxLjAsBgNVBAMTJVVTRVJUcnVzdCBSU0EgQ2Vy
dGlmaWNhdGlvbiBBdXRob3JpdHkwggIiMA0GCSqGSIb3DQEBAQUAA4ICDwAwggIK
AoICAQCAEmUXNg7D2wiz0KxXDXbtzSfTTK1Qg2HiqiBNCS1kCdzOiZ/MPans9s/B
3PHTsdZ7NygRK0faOca8Ohm0X6a9fZ2jY0K2dvKpOyuR+OJv0OwWIJAJPuLodMkY
tJHUYmTbf6MG8YgYapAiPLz+E/CHFHv25B+O1ORRxhFnRghRy4YUVD+8M/5+bJz/
Fp0YvVGONaanZshyZ9shZrHUm3gDwFA66Mzw3LyeTP6vBZY1H1dat//O+T23LLb2
VN3I5xI6Ta5MirdcmrS3ID3KfyI0rn47aGYBROcBTkZTmzNg95S+UzeQc0PzMsNT
79uq/nROacdrjGCT3sTHDN/hMq7MkztReJVni+49Vv4M0GkPGw/zJSZrM233bkf6
c0Plfg6lZrEpfDKEY1WJxA3Bk1QwGROs0303p+tdOmw1XNtB1xLaqUkL39iAigmT
Yo61Zs8liM2EuLE/pDkP2QKe6xJMlXzzawWpXhaDzLhn4ugTncxbgtNMs+1b/97l
c6wjOy0AvzVVdAlJ2ElYGn+SNuZRkg7zJn0cTRe8yexDJtC/QV9AqURE9JnnV4ee
UB9XVKg+/XRjL7FQZQnmWEIuQxpMtPAlR1n6BB6T1CZGSlCBst6+eLf8ZxXhyVeE
Hg9j1uliutZfVS7qXMYoCAQlObgOK6nyTJccBz8NUvXt7y+CDwIDAQABo0IwQDAd
BgNVHQ4EFgQUU3m/WqorSs9UgOHYm8Cd8rIDZsswDgYDVR0PAQH/BAQDAgEGMA8G
A1UdEwEB/wQFMAMBAf8wDQYJKoZIhvcNAQEMBQADggIBAFzUfA3P9wF9QZllDHPF
Up/L+M+ZBn8b2kMVn54CVVeWFPFSPCeHlCjtHzoBN6J2/FNQwISbxmtOuowhT6KO
VWKR82kV2LyI48SqC/3vqOlLVSoGIG1VeCkZ7l8wXEskEVX/JJpuXior7gtNn3/3
ATiUFJVDBwn7YKnuHKsSjKCaXqeYalltiz8I+8jRRa8YFWSQEg9zKC7F4iRO/Fjs
8PRF/iKz6y+O0tlFYQXBl2+odnKPi4w2r78NBc5xjeambx9spnFixdjQg3IM8WcR
iQycE0xyNN+81XHfqnHd4blsjDwSXWXavVcStkNr/+XeTWYRUc+ZruwXtuhxkYze
Sf7dNXGiFSeUHM9h4ya7b6NnJSFd5t0dCy5oGzuCr+yDZ4XUmFF0sbmZgIn/f3gZ
XHlKYC6SQK5MNyosycdiyA5d9zZbyuAlJQG03RoHnHcAP9Dc1ew91Pq7P8yF1m9/
qS3fuQL39ZeatTXaw2ewh0qpKJ4jjv9cJ2vhsE/zB+4ALtRZh8tSQZXq9EfX7mRB
VXyNWQKV3WKdwrnuWih0hKWbt5DHDAff9Yk2dDLWKMGwsAvgnEzDHNb842m1R0aB
L6KCq9NjRHDEjf8tM7qtj3u1cIiuPhnPQCjY/MiQu12ZIvVS5ljFH4gxQ+6IHdfG
jjxDah2nGN59PRbxYvnKkKj9
-----END CERTIFICATE-----
)EOF";
X509List rootCert(trustRoot);

void sendGetCert() {
	WiFiClientSecure client;
	client.setX509Time(1603839517); // set to a reasonable current time...normally using NTP
	client.setTrustAnchors(&rootCert);

	HTTPClient http;
	http.begin(client, "https://www.tenetics.com/index.html");
	int httpResult = http.GET();
	if (httpResult == HTTP_CODE_OK) {
		Serial.println(http.getString());
	} else {
		Serial.printf("HTTP error: %d\r\n", httpResult);
	}
	char sslErrorMsg[80];
	int sslError = client.getLastSSLError(sslErrorMsg, sizeof(sslErrorMsg));
	if (sslError) {
		Serial.printf("SSL error: %d: %s\r\n", sslError, sslErrorMsg);
	}
	http.end();
}
```

OUTPUT:
```
[HTTP-Client][begin] url: https://www.tenetics.com/index.html
                                                             [HTTP-Client][begin] host: www.tenetics.com port: 443 url: /index.html
                                 [HTTP-Client][sendRequest] type: 'GET' redirCount: 0
                                                                                     [hostByName] request IP for: www.tenetics.com
                                [hostByName] Host: www.tenetics.com IP: 104.131.223.32
                                                                                      BSSL:_connectSSL: start connection
                      BSSL:Connected!
                                     [HTTP-Client] connected to www.tenetics.com:443
                                                                                    [HTTP-Client] sending request header
                      -----
                           GET /index.html HTTP/1.1
Host: www.tenetics.com
User-Agent: ESP8266HTTPClient
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
Connection: keep-alive
Content-Length: 0

-----
     [HTTP-Client][returnError] error(-5): connection lost
                                                          HTTP error: -5
SSL error: 296: Unknown error code.
[HTTP-Client][end] tcp is closed
```

dalbert2, 2020-10-27:
I've submitted query to Thomas Pornin (author of BearSSL) to see if he has any ideas; I dug around in the X509 code, but it is not for the faint of heart.

dalbert2, 2020-10-28:
I seem to have solved the issue.  In my apache ssl configuration, when I moved the following two lines out of the Directory sections and into the global (VirtualHost) section, the renegotiation handshake failure stopped and my HTTP requests were processed properly.
```
        SSLVerifyClient optional
        SSLVerifyDepth 1
```

closed 2020-12-22 by earlephilhower