esp8266/Arduino #7683: Exception 29 (StoreProhibited) on LittleFS.openDir("a")

claudiomattera, 2020-10-29:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:             [ESP-12]
- Core Version:         [77bd71ec2987849491ec5d6d889eb77bad882f67 (latest commit on 2020-10-29)]
- Development Env:      [Make]
- Operating System:     [Linux]

### Settings in IDE

- Module:           [Nodemcu v2]
- Flash Mode:       [?]
- Flash Size:       [4MB]
- lwip Variant:     [v2 Higher Bandwidth]
- Reset Method:     [?]
- Flash Frequency:  [?]
- CPU Frequency:    [?]
- Upload Using:     [SERIAL]
- Upload Speed:     [115200]

Note: I am not entirely sure about some of these fields, I filled them up to the best of my knowledge.

### Problem Description

Hello everyone,

I want to use LittleFS to work with directories, but I get an exception 29 (StoreProhibitedCause) as soon as I try to open one.

I am using [makeEspArduino](https://github.com/plerup/makeEspArduino) to compile and build my sketch (I included the `config.mk` file and the sketch below).
I use the commands:

~~~~
espmake erase_flash
espmake flash_fs
espmake run
~~~~

I tried this sketch on two different NodeMCUv2 units from two different vendors and I get the same results.
No wires are attached to any of the pins.

I tried core version 2.7.4 and 2.6.0, and the latest commit available on master, and I get the same results.

However, I also tried to compile and flash the sketch using a different toolchain (an old [Esp8266-Arduino-Makefile](https://github.com/thunderace/Esp8266-Arduino-Makefile) I used to use), and it works perfectly.
So it does not look like a hardware problem (using that toolchain I can also run more complex sketches where I read and write files in a directory).

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <Arduino.h>
#include <LittleFS.h>

void setup(void)
{
    Serial.begin(115200);

    LittleFS.begin();
    Serial.println("Formatting");
    Serial.println(LittleFS.format());
    Serial.println("Formatted");

    Serial.println("Creating dir");
    Serial.println(LittleFS.mkdir("a"));
    Serial.println("Created dir");

    Serial.println("Opening dir"); // <- This is the last output I get
    Dir directory = LittleFS.openDir("a");
    Serial.println("Opened dir");

    LittleFS.end();
}

void loop(void)
{
}
```

~~~~
# File config.mk
THIS_DIR := $(realpath $(dir $(realpath $(lastword $(MAKEFILE_LIST)))))
ROOT := $(THIS_DIR)/..
LIBS = $(ESP_LIBS)/LittleFS

SKETCH = $(THIS_DIR)/dir-test.cpp

UPLOAD_PORT = /dev/ttyUSB0
UPLOAD_SPEED = 115200

# Enable high-performance network stack with IPv6 support
LWIP_VARIANT = hb6f

FS_TYPE = littlefs

BUILD_EXTRA_FLAGS = -DARDUINO_ARCH_ESP8266
~~~~

~~~~
# Makefile for Esp8266-Arduino-Makefile
ARDUINO_VARIANT = nodemcuv2
SERIAL_PORT = /dev/ttyUSB0
SERIAL_BAUD = 115200
LOG_SERIAL_TO_FILE=yes

include ../Esp8266-Arduino-Makefile/espXArduino.mk
~~~~

### Debug Messages

```
Formatting
1
Formatted
Creating dir
1
Created dir
Opening dir

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (29):
epc1=0x40201d0b epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000020 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffc80 end: 3fffffc0 offset: 0190
3ffffe10:  3ffefc84 00000400 3ffffedc 40201c84
3ffffe20:  00000008 3ffeff94 00000001 00000040
3ffffe30:  3ffeff95 00000000 3ffffe90 40201eac
3ffffe40:  3ffeff94 00000000 3ffffe90 00200401
3ffffe50:  3ffffe00 00000000 00000004 00000000
3ffffe60:  3fffff78 3ffffedc 3ffefc84 00000000
3ffffe70:  00000008 3ffefc84 00000000 40202b6c
3ffffe80:  500ffc1b 00010002 00000078 00000079
3ffffe90:  3ffeff94 402078c0 00000040 40206687
3ffffea0:  3fffff78 3ffefc84 3ffeff94 40207912
3ffffeb0:  00000004 00000080 500ffc1b 3ffee9f8
3ffffec0:  402088ec 3fffff1f 3ffee9f8 402088f8
3ffffed0:  402088ec 3fffff1f 3ffee9f8 40208002
3ffffee0:  00000001 00000061 00000000 00000000
3ffffef0:  00000000 00000000 00000000 00000000
3fffff00:  00000000 00000000 00000000 00000000
3fffff10:  00000000 00000000 00000000 00000000
3fffff20:  00000000 00000000 00000000 00000000
3fffff30:  00000000 00000000 00000000 00000000
3fffff40:  00000000 00000000 00000000 00000000
3fffff50:  00000000 00000000 00000000 00000000
3fffff60:  00000000 00000000 00000000 00000000
3fffff70:  00000000 00000000 00000000 00000000
3fffff80:  00000000 00000000 00000000 00000000
3fffff90:  00000000 00000000 00000000 00000000
3fffffa0:  00000000 00000000 00000000 00000000
3fffffb0:  00000000 00000000 00000000 00000000
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3640, room 16
```

If I decode the stack trace with `espmake crash` I get this:

~~~~
Exception (29):
StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores

=== Stack trace ===

lfs_dir_find
  /home/claudio/projects/micro-controllers/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:1102
lfs_dir_get
  /home/claudio/projects/micro-controllers/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:582
lfs_dir_getinfo
  /home/claudio/projects/micro-controllers/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:1048
lfs_dir_open
  /home/claudio/projects/micro-controllers/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:2011
__gnu_cxx::new_allocator<std::_Sp_counted_ptr_inplace<lfs_dir, std::allocator<lfs_dir>, (__gnu_cxx::_Lock_policy)0> >::allocate(unsigned int, void const*)
  /home/claudio/projects/micro-controllers/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/10.2.0/ext/new_allocator.h:115
operator new(unsigned int)
  /home/claudio/projects/micro-controllers/esp8266/cores/esp8266/abi.cpp:41
littlefs_impl::LittleFSImpl::openDir(char const*)
  /home/claudio/projects/micro-controllers/esp8266/libraries/LittleFS/src/LittleFS.cpp:122
HardwareSerial::write(unsigned char const*, unsigned int)
  /home/claudio/projects/micro-controllers/esp8266/cores/esp8266/HardwareSerial.h:164
HardwareSerial::write(unsigned char const*, unsigned int)
  /home/claudio/projects/micro-controllers/esp8266/cores/esp8266/HardwareSerial.h:165
HardwareSerial::write(unsigned char const*, unsigned int)
  /home/claudio/projects/micro-controllers/esp8266/cores/esp8266/HardwareSerial.h:164
settimeofday
  /home/claudio/projects/micro-controllers/esp8266/cores/esp8266/time.cpp:244
~~~~

d-a-v, 2020-10-29:
What about this:
```
#include <Arduino.h>
#include <LittleFS.h>

void setup(void)
{
    Serial.begin(115200);

    LittleFS.begin();
    Serial.println("Formatting");
    Serial.println(LittleFS.format());
    Serial.println("Formatted");

    Serial.println("Creating dir");
    Serial.println(LittleFS.mkdir("a"));
    Serial.println("Created dir");

    {
        Serial.println("Opening dir"); // <- This is the last output I get
        Dir directory = LittleFS.openDir("a");
        Serial.println("Opened dir");
    }

    LittleFS.end();
}

void loop(void)
{
}
```

earlephilhower, 2020-10-29:
Please try with git head (being sure to update the toolchain).  I'm not able to hit any issue with the current G++ v10.2 and git head when I run your sketch:

````
Formatting
1
Formatted
Creating dir
1
Created dir
Opening dir
Opened dir
````

claudiomattera, 2020-11-01:
@d-a-v 
I get the same results.

@earlephilhower 
I tried deleting everything, cloning the repositories to the latest commits on master branch from scratch, and reinstalling the toolchain (`cd tools; python3 get.py` gcc 10.2.0), but I get the same results.

earlephilhower, 2020-11-01:
Very strange.  2.7 and git head have wildly different compilers (gcc 4.8 vs gcc 10.2), and you are getting the same exact crash from both.  Could PIO not actually be using the toolchain you got with `get.py`?

earlephilhower, 2020-11-01:
Also, what is your flash config (i.e. "4MB, 1MB OTA, 2MB FS" or "4MB, 0 OTA, 3MB FS")?

claudiomattera, 2020-11-02:
`espmake VERBOSE=1` shows that the correct compiler is used.

I tried various flash configurations (4MB (FS:2MB OTA:~1019KB), 4MB (FS:3MB OTA:~512KB) and 4MB (FS:1MB OTA:~1019KB)) but always get the same problem.

I will try to find a different machine and see if I can replicate the problem there.
Maybe the issue is somewhere on my system.

claudiomattera, 2020-11-03:
I got a Raspberry Pi and installed ESP8266 Arduino core (latest commit), the toolchain, and makeEspArduino from scratch.
I still get the very same results.

earlephilhower, 2020-11-03:
Sorry, I suggest trying the same MCVE using the Arduino IDE.  We've not received any similar reports on LittleFS from other users, so I don't really know what to tell you at this point.

There may be different options ESPmake is using which are busted now.  Here's the GCC command lines being used in the current repo:
````
/usr/local/bin/arduino-1.8.8/arduino-builder -dump-prefs -logger=machine -hardware /usr/local/bin/arduino-1.8.8/hardware -hardware /home/earle/.arduino15/packages -hardware /home/earle/Arduino/hardware -tools /usr/local/bin/arduino-1.8.8/tools-builder -tools /usr/local/bin/arduino-1.8.8/hardware/tools/avr -tools /home/earle/.arduino15/packages -built-in-libraries /usr/local/bin/arduino-1.8.8/libraries -libraries /home/earle/Arduino/libraries -fqbn=esp8266com:esp8266:d1_mini:xtal=80,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,eesz=4M1M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -ide-version=10808 -build-path /tmp/arduino_build_335163 -warnings=none -build-cache /tmp/arduino_cache_571336 -prefs=build.warn_data_percentage=75 -verbose /home/earle/Arduino/sketch_oct27a/sketch_oct27a.ino
/usr/local/bin/arduino-1.8.8/arduino-builder -compile -logger=machine -hardware /usr/local/bin/arduino-1.8.8/hardware -hardware /home/earle/.arduino15/packages -hardware /home/earle/Arduino/hardware -tools /usr/local/bin/arduino-1.8.8/tools-builder -tools /usr/local/bin/arduino-1.8.8/hardware/tools/avr -tools /home/earle/.arduino15/packages -built-in-libraries /usr/local/bin/arduino-1.8.8/libraries -libraries /home/earle/Arduino/libraries -fqbn=esp8266com:esp8266:d1_mini:xtal=80,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,eesz=4M1M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -ide-version=10808 -build-path /tmp/arduino_build_335163 -warnings=none -build-cache /tmp/arduino_cache_571336 -prefs=build.warn_data_percentage=75 -verbose /home/earle/Arduino/sketch_oct27a/sketch_oct27a.ino
Using board 'd1_mini' from platform in folder: /home/earle/Arduino/hardware/esp8266com/esp8266
Using core 'esp8266' from platform in folder: /home/earle/Arduino/hardware/esp8266com/esp8266
/home/earle/Arduino/hardware/esp8266com/esp8266/tools/python3/python3 /home/earle/Arduino/hardware/esp8266com/esp8266/tools/makecorever.py --build_path /tmp/arduino_build_335163 --platform_path /home/earle/Arduino/hardware/esp8266com/esp8266 --version unix-3.0.0-dev
Detecting libraries used...
/home/earle/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/include -I/home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/lwip2/include -I/home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/libc/xtensa-lx106-elf/include -I/tmp/arduino_build_335163/core -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10808 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_WEMOS_D1MINI\"" -DFLASHMODE_DIO -DESP8266 -I/home/earle/Arduino/hardware/esp8266com/esp8266/cores/esp8266 -I/home/earle/Arduino/hardware/esp8266com/esp8266/variants/d1_mini /tmp/arduino_build_335163/sketch/sketch_oct27a.ino.cpp -o /dev/null
....
Linking everything together...
/home/earle/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -CC -E -P -DVTABLES_IN_FLASH /home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/ld/eagle.app.v6.common.ld.h -o /tmp/arduino_build_335163/local.eagle.app.v6.common.ld
/home/earle/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -fno-exceptions -Wl,-Map -Wl,/tmp/arduino_build_335163/sketch_oct27a.ino.map -g -w -Os -nostdlib -Wl,--no-check-sections -u app_entry -u _printf_float -u _scanf_float -Wl,-static -L/home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/lib -L/home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/lib/NONOSDK22x_190703 -L/home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/ld -L/home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/libc/xtensa-lx106-elf/lib -Teagle.flash.4m1m.ld -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read -o /tmp/arduino_build_335163/sketch_oct27a.ino.elf -Wl,--start-group /tmp/arduino_build_335163/sketch/sketch_oct27a.ino.cpp.o /tmp/arduino_build_335163/libraries/LittleFS/LittleFS.a /tmp/arduino_build_335163/core/core.a -lhal -lphy -lpp -lnet80211 -llwip2-536-feat -lwpa -lcrypto -lmain -lwps -lbearssl -lespnow -lsmartconfig -lairkiss -lwpa2 -lstdc++ -lm -lc -lgcc -Wl,--end-group -L/tmp/arduino_build_335163
/home/earle/Arduino/hardware/esp8266com/esp8266/tools/python3/python3 /home/earle/Arduino/hardware/esp8266com/esp8266/tools/elf2bin.py --eboot /home/earle/Arduino/hardware/esp8266com/esp8266/bootloaders/eboot/eboot.elf --app /tmp/arduino_build_335163/sketch_oct27a.ino.elf --flash_mode dio --flash_freq 40 --flash_size 4M --path /home/earle/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin --out /tmp/arduino_build_335163/sketch_oct27a.ino.bin
Creating BIN file "/tmp/arduino_build_335163/sketch_oct27a.ino.bin" using "/home/earle/Arduino/hardware/esp8266com/esp8266/bootloaders/eboot/eboot.elf" and "/tmp/arduino_build_335163/sketch_oct27a.ino.elf"
[... cut ...]
BSS    : 25096 )         - zeroed variables      (global, static) in RAM/HEAP 
Using library LittleFS at version 0.1.0 in folder: /home/earle/Arduino/hardware/esp8266com/esp8266/libraries/LittleFS 
/home/earle/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-size -A /tmp/arduino_build_335163/sketch_oct27a.ino.elf
Sketch uses 290340 bytes (27%) of program storage space. Maximum is 1044464 bytes.
Global variables use 27616 bytes (33%) of dynamic memory, leaving 54304 bytes for local variables. Maximum is 81920 bytes.
````

devyte, 2020-11-03:
This is a support request to build a sketch with make. I'd love to have that supported directly here in our core, but we currently do not. 
I suggest discussing at the repo where that is implemented, as referenced in the op. 
Closing due to not a core issue.

closed 2020-11-03 by devyte