esp8266/Arduino #7684: CertStoreBearSSL.cpp is only file that pervents compile with -D FS_NO_GLOBALS

isadora-6th, 2020-10-29:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				NodeMCU-V1.0 (Esp8266)
- Core Version:			espressif8266@2.6.2 , framework-arduinoespressif8266@3.20704.0 (newest platformio can download by version number)
- Development Env:		Platformio
- Operating System:		Windows

### Settings in IDE

- Module:  			Nodemcu
- Flash Mode:			dio
- Flash Size: 		        4MB
- lwip Variant:                  v2 Lower Memory
- Reset Method:	        nodemcu
- Flash Frequency:	        40Mhz
- CPU Frequency:		160MHz
- Upload Using:		SERIAL
- Upload Speed:              other

### Problem Description

Older project version used SDFat as lib needed for work.  
I updated ESP8266 core and package, and currentrly using ESP8266SdFat
Older lib had no defined namespace for it, so i need to prevent different FS in global scope.

To prevent issue with different FS difinitons i used this flag
`-D FS_NO_GLOBALS`  
So all code compiled ok except 1 file`CertStoreBearSSL.cpp`
located at path  
`C:\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src\CertStoreBearSSL.cpp`
pathcing file by adding `fs::` before all FS stuff usage solved issue, all compiled fine.

```ini

[env:nodemcuv2]
platform = espressif8266@2.6.2
platform_packages = 
    platformio/framework-arduinoespressif8266@3.20704.0
board = nodemcuv2
framework = arduino
build_flags = 
    -D FS_NO_GLOBALS
lib_deps =
    ESP8266WiFi
    SPI
    ESP8266SdFat

```

### Debug Messages

```
Processing nodemcuv2 (platform: espressif8266@2.6.2; board: nodemcuv2; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (2.6.2) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
[... 294 lines cut ...]
      ta->dn.len = sizeof(ci.sha256);

      return ta;
    }
  }
  index.close();
  return nullptr;
}

void CertStore::freeHashedTA(void *ctx, const br_x509_trust_anchor *ta) {
  CertStore *cs = static_cast<CertStore*>(ctx);
  (void) ta; // Unused
  delete cs->_x509;
  cs->_x509 = nullptr;
}

}

  ```
  
</details>

I think I should had make a Pull Request but never done any before, 
For now i will try create patcher cmdlet to solve this for myself.  
(due to C++ can't just unuse namespace)

earlephilhower, 2020-10-29:
> Core Version: espressif8266@2.6.2 , framework-arduinoespressif8266@3.20704.0 (newest platformio can download by version number)

@ivankravets , sorry to drag you in here but it seems the last PIO release for the ESP8266 Arduino is very old.  Do you need to run something on your end to pick up the new releases, or did we break something you were using to detect new releases?  We're on 2.7.4 now...

@isadora-6th, doing a pull request is pretty simple if you're familiar with git.  See https://docs.github.com/en/free-pro-team@latest/desktop/contributing-and-collaborating-using-github-desktop/creating-an-issue-or-pull-request .  Basically you'll fork the repo, clone it locally, make your changes to a branch, push the branch to your fork on GH, and then on the GW website for the repo homepage you'll have a "Create pull request" button you can use to get one in.

If that's too much, let us know and we can do it.  But, since you have a fix, it'd be nice to get your name attached to it.  

We should also have a NO_GLOBALS build somewhere in CI to catch this kind of thing, too.

isadora-6th, 2020-10-30:
About new releases, 2.6.2 is platform version
framework-arduinoespressif8266 3.20704.0 (2.7.4)
2.7.4 as i understod arduino user-friendly version instead of 3.20704.0

As I can see from this part
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (2.6.2) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)

Made a pull request, thank you for instructions, they are very clear  
https://github.com/esp8266/Arduino/pull/7685

closed 2020-10-30 by isadora-6th