esp8266/Arduino #7687: Example HelloServer can't connect, but ESP-Link can

rtek1000, 2020-11-01:
Hi,

I would like to know what happened to the library, I used an old version, and today I updated it and it stopped working. I have two NodeMCU cards and both are the same. I tested with the ESP-Link firmware and connected correctly.

The Scan example can find the network. AP mode works. Only station mode does not work.

No other example in station mode (STA) worked.

Serial monitor shows only:

> ................................................................................................................................................................................................................

Ref.: https://github.com/jeelabs/esp-link

Tech-TX, 2020-11-02:
I may have it, but I don't know if the program is fully working.  I duplicated your problem and found a 'fix', although the fix doesn't make any sense.  Both methods should work, but the one in the example won't connect for me.

Replace:
```cpp
#ifndef STASSID
#define STASSID "your-ssid"
#define STAPSK  "your-password"
#endif

const char* ssid = STASSID;
const char* password = STAPSK;
```
with
```cpp
/*#ifndef STASSID
#define STASSID "your-ssid"
#define STAPSK  "your-password"
#endif*/

const char* ssid     = "your-ssid";
const char* password = "your-password";
```
That simple change gets a connection for me.  I don't see any hidden non-printing characters buried in the source file, but something is wonky there.

devyte, 2020-11-02:
The only reasonable way that could have any impact is if STASSID and STAPSK were being defined externally. In that case, your ssid and psk will never be set as you would expect, because the external defines have priority. 
I suggest reverting to the code that fails to connect and printing out STASSID and STAPSK.

rtek1000, 2020-11-02:
Hi,

It looks like this code **has a problem with the DHCP** client:

Serial monitor (after add Serial.setDebugOutput(true); ):

> mode : sta(a0:20:a6:xx:xx:xx)
> add if0
> 
> .....scandone
> state: 0 -> 2 (b0)
> .state: 2 -> 3 (0)
> state: 3 -> 5 (10)
> add 0
> aid 4
> cnt 
> 
> connected with ., channel 6
> dhcp client start...
> ....................pm open,type:2 0
> .......................................................................................................................................

```C++
  Serial.begin(115200);
  Serial.setDebugOutput(true);
```

If I set static IP it works.

```C++
  IPAddress staticIP(192, 168, 1, 74);
  IPAddress gateway(192, 168, 1, 1);
  IPAddress subnet(255, 255, 255, 0);
  IPAddress dns(192, 168, 1, 1);

  if (WiFi.config(staticIP, gateway, subnet, dns, dns) == false) {
    Serial.println("Configuration failed.");
  }

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
```

Serial monitor:

> mode : sta(a0:20:a6:xx:xx:xx) + softAP(a2:20:a6:xx:xx:xx)
> add if0
> bcn 0
> del if1
> mode : sta(a0:20:a6:xx:xx:xx)
> 
> .....scandone
> state: 0 -> 2 (b0)
> .state: 2 -> 3 (0)
> state: 3 -> 5 (10)
> add 0
> aid 4
> cnt 
> 
> connected with ., channel 6
> ip:192.168.1.74,mask:255.255.255.0,gw:192.168.1.1
> ip:192.168.1.74,mask:255.255.255.0,gw:192.168.1.1
> .
> Connected to .
> IP address: 192.168.1.74
> MDNS responder started
> HTTP server started
> pm open,type:2 0

rtek1000, 2020-11-02:
I tested with Arduino IDE 1.8.9, 1.8.10 and 1.8.13.

I also tested since version> 2.4.2 of the ESP8266 board and the same problem also occurred.

I already restarted my router. I think I will have to use only static IP.

rtek1000, 2020-11-02:
It really was DHCP, I activated the Router on my smartphone and the ESP8266 was able to connect to it.

Thank you.

> bcn 0
> del if1
> usl
> mode : sta(a0:20:a6:xx:xx:xx)
> add if0
> 
> .....scandone
> state: 0 -> 2 (b0)
> .state: 2 -> 3 (0)
> state: 3 -> 5 (10)
> add 0
> aid 1
> cnt 
> 
> connected with MyRouter, channel 1
> dhcp client start...
> ip:192.168.43.19,mask:255.255.255.0,gw:192.168.43.1
> .
> Connected to MyRouter
> IP address: 192.168.43.19
> MDNS responder started
> HTTP server started
> pm open,type:2 0

Tech-TX, 2020-11-02:
I'm running DHCP and it connects after the changes I have above.

When I printed out STASSID it was correct immediately before the connection was attempted (before I commented out the IFNDEF chunk).  I didn't print STAPSK yet, but since the IFNDEF tests STASSID, STAPSK shouldn't have been a leftover from code elsewhere as it's explicitly defined here.

Tech-TX, 2020-11-07:
Never mind, my issues were solely because I fat-fingered my SSID, and then looked at it and said "Yep, that's it..."

One suggestion: before the WiFi.mode(WIFI_STA) on line 45 add:
  **WiFi.persistent(false);**

That should/might keep it from trying to connect to your previous connection.  I'm fairly sure it ignores the stored connection.  

Alternately, try Tools -> Erase Flash: -> Sketch + WiFi Settings before you compile, which ought to accomplish the same thing.

rtek1000, 2020-11-08:
Hi,

I tried to add the **_WiFi.persistent(false);_** instruction, but the DHCP client has not yet worked.

I found another reference on the router's configuration, but it also didn't work after switching from TKPI / AES to AES encryption.

ESP8266 NOT GETTING DHCP IP ADDRESS FROM CISCO AP:
https://www.esp8266.com/viewtopic.php?p=74498

Still working with static IP only.

Thank you.

d-a-v, 2020-11-08:
If the esp works with a static address and is able to communicate with other devices connected to same the cisco AP, then the issue should not be related with WPA TKIP or AES. These are link layer protocols while IP (with a static address or from DHCP) are network layer protocols.

One could try to flash an RTOS SDK example (different and updated link layer firmware) and see if DHCP client is working with such a cisco AP.