esp8266/Arduino #7688: HTTPClient library only allows Basic authorization

guneemwelloeux, 2020-11-02:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12F
- Core Version:			2.7.4
- Development Env:		Arduino IDE
- Operating System:		MacOS

### Settings in IDE

- Module:  			Wemos D1 R2
- Flash Mode:			qio
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		ck
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   921600

### Problem Description

After checking the source code of the library, it seems that the only way to set the `Authorization` header is through the use of either of the 2 `setAuthorization` methods (adding the header manually through the `addHeader` method does nothing for this header).
However, both of these methods internally set a base64 encoded field, which is then used as `Basic` type authentication.

As far as I can see, there is no way to set a `Bearer` token to authenticate requests.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <ESP8266HTTPClient.h>

const char* ssid = "ssid";
const char* password = "pass";
HTTPClient client;

void setup() {
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
 
  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }
}

void loop() {
  client.begin("http://localhost:5678");
  client.setAuthorization("user", "pass"); // Will set _base64Authorization to base64encode(user+ ':' + pass)
  client.setAuthorization("base64encoded_String"); // Will set _base64Authorization to the arg value
  client.GET(); // Will set the Authorization header to "Basic " + _base64Authorization
}

```

### Debug Messages

No debug messages

earlephilhower, 2020-11-03:
Yes, that is the case.  Only basic authentication is supported now.  It might be possible to add the headers manually by extending the HTTPClient class to allow user-supplied headers, but I'm not familiar w/Bearer authentication so there might be some gotcha even then.

If this feature is important to you, please consider having a try at making a PR implementing it.  We can assist in the PR or via the IRC-like https://gitter.im/esp8266/Arduino .

Zakaria16, 2020-12-29:
You can use the addHeader method to set other authorization.
Looking at the the source code, the addHeader method allows authorization field to be set only if the **_base64Authorization length is zero**, as shown below:
```c
  if (!name.equalsIgnoreCase(F("Connection")) &&
        !name.equalsIgnoreCase(F("User-Agent")) &&
        !name.equalsIgnoreCase(F("Host")) &&
        !(name.equalsIgnoreCase(F("Authorization")) && _base64Authorization.length()))
```

so to set a custom authorization you can do something like this:
```c
//this will make _base64Authorization.length == 0 
client.setAuthorization("");
client.addHeader("Authorization","Bearer bearer_token");
```

mateusscheper, 2021-01-18:
Hey guys,

Sorry for disturbing someone else's post, but I'm having the same problem.
I need to set the Authorization header as "Bearer", plus some custom headers.

I tried to copy/paste the lib so I could change the code and use it, but my headers are not being set. I think it's because of platformio, I'm not sure (I'm new to this).

Would it be possible to create a new update() like this?
```
HTTPUpdateResult ESP8266HTTPUpdate::update(WiFiClient& client, const String& url, const std::map<String,String>& headers)
{
    HTTPClient http;
    for (const auto &header : headers) {
        if (header.first.equals("Authorization"))
            http.setAuthorization("");

        http.addHeader(header.first, header.second);
    }

    http.begin(client, url);
    return this->handleUpdate(http, "", false);
}
```

I'm trying to fix this platformio conflict I'm facing and I can't test this piece of code :( 
Also, I'm not sure how to make a PR (never made one).

AlaaElnagar, 2021-01-31:
same problem here any help link ?!!1

tomash, 2021-04-08:
It is possible to add custom header in HTTPClient, I've followed snippet in this thread and got bearer auth to work properly: https://forum.arduino.cc/index.php?topic=702610.0

GamerClassN7, 2021-12-07:
Hello is possible to set header in latest wersion or atleas pass custome httpsclient instance ?