esp8266/Arduino #7690: Some POSIX time zones are not handled correctly

rjwats, 2020-11-02:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12]
- Core Version:			[2.7.4]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

It looks as if some POSIX time zone strings may be being interpreted incorrectly when configured using:

```cpp
configTime(const char* tz, const char* server1, const char* server2, const char* server3)
```
or

```cpp
setTZ(const char* tz)
```

The following examples have been taken from [TZ.h](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/TZ.h):

The following are some of the values which demonstrate the issue:

```cpp
#define TZ_America_Sao_Paulo	PSTR("<-03>3") // Offset by +03:00 instead of -03:00
#define TZ_Africa_Casablanca	PSTR("<+01>-1") // Offset by -01:00 instead of +01:00
#define TZ_America_Asuncion	PSTR("<-04>4<-03>,M10.1.0/0,M3.4.0/0") // Offset by +4:00 instead of -03:00
```

This pattern appears consistent for all time zones which are defined in this format.

By contrast the following (also taken from [TZ.h](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/TZ.h)) **do not** exhibit this issue:

```cpp
#define TZ_America_Los_Angeles	PSTR("PST8PDT,M3.2.0,M11.1.0") // Offset by -08:00
#define TZ_Europe_Rome	PSTR("CET-1CEST,M3.5.0,M10.5.0/3") // Offset by +01:00
#define TZ_Europe_Kaliningrad	PSTR("EET-2") // Offset by +02:00
```

All of the examples above were tested today @ ~1604351488 unix timestamp.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#ifndef STASSID
#define STASSID "VM723402-2G"
#define STAPSK  "gmwscxnu"
#endif

#include <TZ.h>

#define MYTZ TZ_America_Sao_Paulo

#include <ESP8266WiFi.h>
#include <coredecls.h>
#include <PolledTimeout.h>

#include <time.h>
#include <sys/time.h>

extern "C" int clock_gettime(clockid_t unused, struct timespec *tp);

static time_t now;
static esp8266::polledTimeout::periodicMs showTimeNow(5000);

String toISOString(tm* time) {
  char time_string[25];
  strftime(time_string, 25, "%FT%T", time);
  return String(time_string);
}

void showTime() {
  now = time(nullptr);
  Serial.print("gmtime:     ");
  Serial.println(toISOString(gmtime(&now)));
  Serial.print("localtime:  ");
  Serial.println(toISOString(localtime(&now)));
}

void time_is_set_scheduled() {
  showTime();
}

void setup() {
  Serial.begin(115200);
  Serial.println("Starting...");

  settimeofday_cb(time_is_set_scheduled);

  // Set the timezone
  configTime(MYTZ, "pool.ntp.org");

  // start network
  WiFi.persistent(false);
  WiFi.mode(WIFI_STA);
  WiFi.begin(STASSID, STAPSK);
}

void loop() {
  if (showTimeNow) {
    showTime();
  }
}

```

### Debug Messages

One example, when using TZ_America_Sao_Paulo, local time should be offset -03:00, instead it is offset +03:00:

```
gmtime:     2020-11-02T20:36:51
localtime:  2020-11-02T23:36:51
gmtime:     2020-11-02T20:36:56
localtime:  2020-11-02T23:36:56
gmtime:     2020-11-02T20:37:01
localtime:  2020-11-02T23:37:01
gmtime:     2020-11-02T20:37:06
localtime:  2020-11-02T23:37:06
```

d-a-v, 2020-11-03:
Thanks for the report and MCVE.
This is reproducible.
I don't think it could be on newlib but we are using their API and we make no difference from working TZ and Sao Paulo.
I have so far no idea where it comes from.

d-a-v, 2020-11-10:
This is fixed by #7699

rjwats, 2020-11-10:
Nice, thanks 👍

PR #7699 (2020-11-10): Help newlib TZ parser
Timezones coded with `<±nn>±nn<±nn>[±nn][,...]` are incorrectly parsed by newlib's TZ parser.
In these timezones, `<±nn>` is used because there is no matching name like in other timezones.
Example:
- `GMT0BST` for London, (GMT is known, BST is `British Summer Time`).
- `<+1030>-10:30<+11>-11` for Lord Howe Island.

These names GMT,BST,CET,... are meant to be human readable but are useless in our use case.
Replacing `<±nn>` occurences by `UNK`nown allows newlib's TZ parser to nicely interpret all timezones.
`UNK-10:30UNK-11` is then correctly parsed.

Ultimately, newlib sh [... cut]
PR #7702 (2020-11-12): Fix newlib to support <+-nn> timezone names
Undoes the change in #7699 and fixed #7690 root cause.

Newlib did not support timezone names of the form "<[+-]?[0-9]+>" and
would parse the offset using the <name>.

Fix newlib tzset parser with
  https://github.com/earlephilhower/newlib-xtensa/pull/14
and undo the UNK changes used as an expedient workaround.
commit 8375faa5 (2020-11-12): Fix newlib to support <+-nn> timezone names (#7702)

Undoes the change in #7699 and fixed #7690 root cause.

Newlib did not support timezone names of the form "<[+-]?[0-9]+>" and
would parse the offset using the <name>.

Fix newlib tzset parser with
  https://github.com/earlephilhower/newlib-xtensa/pull/14
  https://github.com/earlephilhower/newlib-xtensa/pull/15
and undo the UNK changes used as an expedient workaround.
closed 2020-11-10 by d-a-v