esp8266/Arduino #7692: http.getString() return blank

igorfelluga, 2020-11-04:
Hi
I have a simple code that read a json and display result on lcd.
After some month I resumed it but I think that some update (Arduino CLI o other) give me a problem with http.getString() that return empty string.
To do a simple test I did an empty project with only wifi connection http request.
Here my code:
```#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

const char* ssid = "FUNCOOL";
const char* password = "1807200505092011";

WiFiClient client;
HTTPClient http;

void setup() {

  // put your setup code here, to run once:
  Serial.begin(115200);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.println("Connecting...");
  }

}

void loop() {
  if (WiFi.status() == WL_CONNECTED) {

    http.begin(client, "http://192.168.1.252/user/counter.json");
    int httpCode = http.GET();
    Serial.println("Connected");
    Serial.println(httpCode);
    //Check the returning code
    if (httpCode == 200) {
      Serial.println("Connected");
      String payload = http.getString();
      Serial.println("http string");
      Serial.println(payload);
      Serial.println("http string end");

    } else {
      Serial.println("Gateway Problem");
    }
    http.end();   //Close connection
  } else {
    Serial.println("NOT Connected! Check WiFi");
  }
  // Delay
  delay(1000);

}```

when I trace it get 
```
Connecting...
Connecting...
Connected
200
Connected
http string

http string end
Connected
200
Connected
http string

http string end
Connected
200
Connected
http string
```
Has you can see the payload is empty/blank.
Any suggestion for me?
Thanks

Tech-TX, 2020-11-06:
This is an area for issues with the Arduino/esp8266 core files and included libraries.  The issue template that you've ignored clearly states that "how do I..."  questions are for the arduino.cc or esp8266.com forums,  not here. 
The glitter is also open to generic "how do I"  questions.

igorfelluga, 2020-11-07:
I'm in right place
I have tested with older version and with 2.7.1 everything work.
With 2.7.4 I get the blank payload.

d-a-v, 2020-11-07:
Your code works well with current master.
What does `curl -D - http://192.168.1.252/user/counter.json` give ?

igorfelluga, 2020-11-08:
I get
`
HTTP/1.1 200 OK
Connection: close
Content-Type: application/json
Cache-Control: no-cache

{
"logged": 0,
"domus": "000000000000",
"life": 0,
"rt_stat": 2,
"dataora": 1604830496,
"instant": ["1.23kW","","","","","","","","","","","","","","","","","","","","","","","","","","","","0W","0W",""]
}

`

d-a-v, 2020-11-08:
There is no `Content-Length` in the header. It is usually recommended but [not mandatory](https://stackoverflow.com/questions/15991173/is-the-content-length-header-required-for-a-http-1-0-response), after checking it is supposed to be handled in our code.
There were recent changes on ability to reuse a connection. This is maybe a so-far-unseen side effect.
We need to check.
Ou of curiosity what is the server ?

igorfelluga, 2020-11-09:
it's a gateway 
https://pro.comelitgroup.com/en-gb/product/1456

d-a-v, 2020-11-09:
I reproduced the issue and while trying to fix I realized it already is in #6979.
You can try with this pull request or you can also try the [installable alpha release (v0.0.2)](https://d-a-v.github.io/esp8266/Arduino).

closed 2020-11-20 by earlephilhower

Endc89, 2022-01-30:
Hi,
I have a similar behavior to that described in this issue. 

I'm sending some GET requests and reading the response with http.getString(). For all the requests I get 200 as a respose code but for some of them http.getString() return blank.

Testing requests with curl everything works properly. The big difference between the requests is the payload size, I noticed that if the payload is large http.getString() return blank. At the moment I don't know exactly after what size this behavior occurs and if really the problem is the size.

Testing requests with curl I have also identified the size of the payloads:
- requests with payloads of ~600 characters and ~2500 characters -> http.getString() ok
- requests with payloads of ~5000 characters and ~9500 characters -> http.getString() blank
 
Is this possible? Any suggestions?

At the moment i'm not using `Content-Length` because i don't know the dimensions of the answers.
I'm using Release 3.0.2

Thanks

d-a-v, 2022-01-30:
If you can get a 9500 bytes long answer maybe this protocol should be revised ?
It'd be the best option, otherwise there are some paths:
- use a reference string if you currently don't, it will prevent from duplicating a large object
```cpp
const String& payload = http.getString();
```
- ask core to reserve data in second heap (option must be enabled from tools menu)
```cpp
ESP.setIramHeap();
const String& payload = http.getString();
ESP.resetHeap();
``` 
  (check [example](https://github.com/esp8266/Arduino/blob/master/libraries/esp8266/examples/irammem/irammem.ino))
- or stream data to a file (in flash or sdcard) (wear-leveling is the enemy here)

bergerb-com, 2023-02-24:
I know, this is Off-Topic to the issue, but to anybody finding this via google.. don't .end() your client before you do .getString()... Yeah it won't return anything that way.. don't be stupid like me

fedu, 2023-10-22:
For those who ended up here, when `.getString()` can't handle the body size of your GET request - this is the answer you're looking for:
```C
  // String payload = http.getString();
  String payload = "";
  while (stream->available()) {
    char c = stream->read();
    payload += c;
  }
```

sercona, 2023-12-31:
> For those who ended up here, when `.getString()` can't handle the body size of your GE

interesting that this solution works when my esp8266 does a GET from an external website, but when one esp tries to serve that same file (json string) to another esp, the stream pointer solution returns nothing; whereas the getString() way DOES work.  really weird.  trying to understand why this is.

PR #7691 (2020-11-20): Update to ESP8266HTTPClient.cpp for no Content-Length
Response bodies are ignored when _transferEncoding == HTTPC_TE_IDENTITY and there is no Content-Length header.  The added code here attempts to fix that issue by writing the response body based upon how much content is available to be read.

*edit*:
Fixes #7692