esp8266/Arduino #7693: Further transparent integration of compressed updates?

sticilface, 2020-11-06:
This is to discuss expanding compressed OTA binaries.  

My understanding so far is that .bin files must be compressed on the command line, and then signed manually (if so inclined), before being uploaded to the sketch.  The sketch must have defined `ATOMIC_FS_UPDATE` for the bootloader to write the compressed FS (i'm curious how this works for say FS sizes of 3Mb which when compressed gives a file greater than max OTA size (which for 3Mb FS on a 4Mb flash is 512Kb). 

I would like to propose. 

- A mechanism to report compression functionality in the boot loader (ie a #DEFINE) 
- A mechanism whereby the OTA updater class can report back to espota.py that it supports compression
- espota.py can then send compressed binaries instead.  

advantages:  
- no user input required, fully transparent, uploads will be much faster for everyone without the user having to do anything at all. 

disadvantages
- possible extra handshake during OTA

Any thoughts?

d-a-v, 2020-12-30:
> I would like to propose.

Do you already have something to test ?

sticilface, 2021-01-11:
Sorry for the delay. I am afraid I do not have any work to contribute. I have no python skills! More importantly I'm currently fairly occupied with the second wave here in the Uk as an icu dr!

d-a-v, 2021-01-12:
> no user input required, fully transparent, uploads will be much faster for everyone without the user having to do anything at all.

That's a fair request

>  I'm currently fairly occupied with the second wave here in the Uk as an icu dr!

!! That's good to know.
I don't really know what to wish you, but do take the best from it if that can help.