esp8266/Arduino #7694: client.read() always return 255 if buffer size is bigger 256

enjoyneering, 2020-11-06:
#### Platform

- Hardware:			[ESP-12]
- Core Version:		[2.7.4]
- Development Env:	[Arduino IDE]
- Operating System:	[Windows]

### Settings in IDE

- Module:  		[Generic ESP8266 Module]
- Flash Mode:		[qio]
- Flash Size: 		[4MB]
- lwip Variant:  	 [v2 Higher Bandwidth]
- Reset Method:	[ck|nodemcu]
- Flash Frequency:	[80Mhz]
- CPU Frequency:	[80Mhz]
- Upload Using:	[SERIAL]
- Upload Speed:   	[115200 (serial upload only)

### Problem Description

Hi I am using a overhead buffer to copy data from the server and then copy it to my own class. The problem is in uint8_t. As long as the size of the overhead buffer is 255 or less, I have no problem, but if I make it 256, client.read (buffer, 256); always return 255. As far as I understand WiFiClient.h, the client buffer is TCP MSS 1460 // lwip1.4. Maybe it is because I am using the lwip v2 Higher Bandwidth?

P.S. The foo calss has buffer size is uint16_t.

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>

#define BUFFER_SIZE   256

foo            myClass;
WiFiClient client;

uint8_t  readBytes;
uint8_t  overheadBuffer[BUFFER_SIZE] = {0};

void setup() {
client.connect(address, port);
client.print(myGETRequest);
}

void loop() {
  if (client.available() > 0)
  {
    readBytes = client.read(overheadBuffer, BUFFER_SIZE); //read to overheadBuffer, up to overheadBuffer size

  for (uint16_t i = 0; i < BUFFER_SIZE; I++)
  {
    Serial.println(overheadBuffer++); //if BUFFER_SIZE > 255 Serial.println() = 255!!!!
  }

    myClass.writeData(overheadBuffer, readBytes);     //if BUFFER_SIZE > 255 myClass.writeData received 255!!!
  }
}

```

d-a-v, 2020-11-06:
`::read()` returns the number of read bytes. If you ask for N but receive less than N, then you need to call `::read()` again to get the following data.

```
remain = BUFFER_SIZE;
done = 0;
while (done < BUFFER_SIZE)
{
    readBytes = client.read(overheadBuffer + done, remain);
    done += readBytes;
    remain -= readBytes;
}
```

enjoyneering, 2020-11-06:
Thank you for unswer. Oh sorry I made a mistake in while copying the code, but that doesn't explain why myClass.writeData(overheadBuffer, **readBytes**); gets array filled with 255, if array size is bigger 255? corrected code:

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>

#define BUFFER_SIZE   256

foo        myClass;
WiFiClient client;

uint8_t  readBytes = 0;
uint8_t  overheadBuffer[BUFFER_SIZE] = {0};

void setup() {
client.connect(address, port);
client.print(myGETRequest);
}

void loop() {
  if (client.available() > 0)
  {
    readBytes = client.read(overheadBuffer, BUFFER_SIZE); //read to overheadBuffer, up to overheadBuffer size

    for (uint16_t i = 0; i < readBytes; i++)
    {
      Serial.println(overheadBuffer++); //if BUFFER_SIZE > 255 Serial.println() = 255!!!!
    }

    myClass.writeData(overheadBuffer, readBytes);     //if BUFFER_SIZE > 255 myClass.writeData received 255!!!
  }
}

```

jandrassy, 2020-11-06:
> 
> 
> Thank you for unswer. Oh sorry I made a mistake in while copying the code, but that doesn't explain why myClass.writeData(overheadBuffer, **readBytes**); gets array filled with 255, if array size is bigger 255?
> 
why does it matter? internally it works with 255 byte large buffer so it returns so much at once. if you have a larger buffer call it once more with the new start position and remaining size.

enjoyneering, 2020-11-06:
If I understand you correctly, client.read() has an upper limit of 225 bytes. Right?

The way I'm doing it right now I cannot afford another call to client.read (). I receive an MP3, 320kbps audio stream. The second call takes too long and the VS1050 hardware FIFO goes empty.

d-a-v, 2020-11-06:
@enjoyneering You may try @160MHz and you must use lwIP option "larger bandwidth" to speed network up.
But you need to call read or write as many time as it takes to get all your data transferred.

enjoyneering, 2020-11-06:
It looks like I'm already using the highest IwIP - **v2 Higher Bandwidth**. Also cranked flash speed and mode to 80MHz/QIO. Do not want to increase CPU speed for now. There must be a programm way to avoid the VS1053 buffer flaw.

If I understand correctly, client.read() has an upper limit of 225 bytes. Right?

d-a-v, 2020-11-06:
> If I understand correctly, client.read() has an upper limit of 255 bytes. Right?

As far as lwIP is concerned, there's no such limit.
Signatures are:
```
virtual int Client::read(uint8_t *buf, size_t size)
virtual int WiFiClient::read(uint8_t *buf, size_t size) override
```

in "v2 higher bandwidth", segment size is max 1460 bytes, so maximum bandwidth is better at maximum packet rate.
But effective bandwidth depends on many things: from environmental state (radio link, rssi), through what are the involved protocols, to way of coding on both side of the tcp link. They all have an effect on the effective packet rate.

enjoyneering, 2020-11-06:
Now it is all clear to me. Thank you d-a-v for the answer.

One last question before closing the ticket. Is there any tool in the ESP8266 Arduino Core to tell the server over TCP that I can receive bigger packets?

d-a-v, 2020-11-06:
> Is there any tool in the ESP8266 Arduino Core to tell the server over TCP that I can receive bigger packets?

In no decent TCP implementation you can do that. TCP is designed to handle things in the background.
MSS is the limit for maximum TCP packet size. In "higher bandwidth" it's 1460 and that's the maximum value defined by standards (we can't deal with 9KB jumbo packets).
If you want to control what's in a packet you need to use UDP but you'd have to manage yourself with the flow control (which is the purpose of TCP).

enjoyneering, 2020-11-06:
Seems like the simplest solution is a ring buffer for 1-5 seconds of audio before I start feeding the VS1053 2KB FIFO. When the packet rate drops to 255 or less, I still have ~5 secons.

Thank you d-a-v for help.

closed 2020-11-06 by enjoyneering