esp8266/Arduino #7695: Error Connecting to esp8266

Duui3111, 2020-11-06:
#### Platform

- Hardware:			ESP-12E
- Core Version:			[latest git hash or date]
- Development Env:		Arduino IDE
- Operating System:		Window 10

### Settings in IDE

- Module:  			Nodemcu
- Flash Mode:			[qio|dio|other]
- Flash Size:                      4MB
- lwip Variant:                   v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	        40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:               115200

### Problem Description

the Problem i have is i cant see the port of my esp8266 nodeMCU ESP-12E and my 2 one that i have.

## what i did

1. i added the url `https://arduino.esp8266.com/stable/package_esp8266com_index.json` in settings 
2. i opened board manager i installed esp2688 library
3. i pluged my esp8266

## what i tried to do to fix the issue 

1. i have installed the drivers for my esp8266 
2. to use a different pc 
3. and i tried t use 3 different usb cables 
4. a different esp8266  

```cpp

#include <Arduino.h>

void setup()
{
    Serial.begin(9600);
    Serial.print("start");
   pinMode(2, OUTPUT);
}

void loop()
{
  Serial.print("loop");
  digitalWrite(2, HIGH);
  delay(1000);
  digitalWrite(2, LOW);
  delay(1000);
}
```

### Debug Messages

```
Arduino: 1.8.13 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Executable segment sizes:

IROM   : 232872          - code in flash         (default or ICACHE_FLASH_ATTR) 

IRAM   : 26892   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 

DATA   : 1248  )         - initialized variables (global, static) in RAM/HEAP 

RODATA : 716   ) / 81920 - constants             (global, static) in RAM/HEAP 

BSS    : 24936 )         - zeroed variables      (global, static) in RAM/HEAP 

Sketch uses 261728 bytes (25%) of program storage space. Maximum is 1044464 bytes.

Global variables use 26900 bytes (32%) of dynamic memory, leaving 55020 bytes for local variables. Maximum is 81920 bytes.

esptool.py v2.8

Serial port COM3

Traceback (most recent call last):

  File "C:\Users\usr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4/tools/upload.py", line 65, in <module>

    esptool.main(cmdline)

  File "C:/Users/usr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 2889, in main

    esp = chip_class(each_port, initial_baud, args.trace)

  File "C:/Users/usr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 237, in __init__

    self._port = serial.serial_for_url(port)

  File "C:/Users/usr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/pyserial\serial\__init__.py", line 88, in serial_for_url

    instance.open()

  File "C:/Users/usr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/pyserial\serial\serialwin32.py", line 62, in open

    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))

serial.serialutil.SerialException: could not open port 'COM3': FileNotFoundError(2, 'The system cannot find the file specified.', None, 2)

serial.serialutil.SerialException: could not open port 'COM3': FileNotFoundError(2, 'The system cannot find the file specified.', None, 2)

Error opening serial port 'COM3'. (Port not found)

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

earlephilhower, 2020-11-07:
> Error opening serial port 'COM3'. (Port not found)

This means the serial port on your board went away.  Could be USB port power or USB cable related, could be bad clone boards.

Either way, this is not the appropriate spot as this is a core issue tracker.  Use https://esp8266.com or https://gitter.im/esp8266/Arduino for assistance.

closed 2020-11-07 by earlephilhower

Nxjk, 2022-04-01:
> #### Platform
> * Hardware:			ESP-12E
> * Core Version:			2.7.4
> * Development Env:		Arduino IDE
> * Operating System:		Window 10
> 
> ### Settings in IDE
> * Module:  			Nodemcu
> * Flash Mode:			[qio|dio|other]
> * Flash Size:                      4MB
> * lwip Variant:                   v2 Lower Memory
> * Reset Method:		nodemcu
> * Flash Frequency:	        40Mhz
> * CPU Frequency:		80Mhz
> * Upload Using:		SERIAL
> * Upload Speed:               115200
> 
> ### Problem Description
> the Problem i have is i cant see the port of my esp8266 nodeMCU ESP-12E and my 2 one that i have.
> 
> ## what i did
> 1. i added the url `https://arduino.esp8266.com/stable/package_esp8266com_index.json` in settings
> 2. i opened board manager i installed esp2688 library
> 3. i pluged my esp8266
> 
> ## what i tried to do to fix the issue
> 1. i have installed the drivers for my esp8266
> 2. to use a different pc
> 3. and i tried t use 3 different usb cables
> 4. a different esp8266
> 
> ```c++
> #include <Arduino.h>
> 
> void setup()
> {
>     Serial.begin(9600);
>     Serial.print("start");
>    pinMode(2, OUTPUT);
> }
> 
> void loop()
> {
>   Serial.print("loop");
>   digitalWrite(2, HIGH);
>   delay(1000);
>   digitalWrite(2, LOW);
>   delay(1000);
> }
> ```
> 
> ### Debug Messages
> ```
> Arduino: 1.8.13 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"
> 
> Executable segment sizes:
> 
> IROM   : 232872          - code in flash         (default or ICACHE_FLASH_ATTR) 
> 
> IRAM   : 26892   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
> 
> DATA   : 1248  )         - initialized variables (global, static) in RAM/HEAP 
> 
> RODATA : 716   ) / 81920 - constants             (global, static) in RAM/HEAP 
> 
> BSS    : 24936 )         - zeroed variables      (global, static) in RAM/HEAP 
> 
> Sketch uses 261728 bytes (25%) of program storage space. Maximum is 1044464 bytes.
> 
> Global variables use 26900 bytes (32%) of dynamic memory, leaving 55020 bytes for local variables. Maximum is 81920 bytes.
> 
> esptool.py v2.8
> 
> Serial port COM3
> 
> Traceback (most recent call last):
> 
>   File "C:\Users\usr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4/tools/upload.py", line 65, in <module>
> 
>     esptool.main(cmdline)
> 
>   File "C:/Users/usr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 2889, in main
> 
>     esp = chip_class(each_port, initial_baud, args.trace)
> 
>   File "C:/Users/usr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 237, in __init__
> 
>     self._port = serial.serial_for_url(port)
> 
>   File "C:/Users/usr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/pyserial\serial\__init__.py", line 88, in serial_for_url
> 
>     instance.open()
> 
>   File "C:/Users/usr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/pyserial\serial\serialwin32.py", line 62, in open
> 
>     raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
> 
> serial.serialutil.SerialException: could not open port 'COM3': FileNotFoundError(2, 'The system cannot find the file specified.', None, 2)
> 
> serial.serialutil.SerialException: could not open port 'COM3': FileNotFoundError(2, 'The system cannot find the file specified.', None, 2)
> 
> 
> 
> Error opening serial port 'COM3'. (Port not found)
> 
> This report would have more information with
> "Show verbose output during compilation"
> option enabled in File -> Preferences.
> ```

Did you find a solution for this I have the same problem? can you confirm that it the board that is bad.
Hey , thanks.