esp8266/Arduino #7698: commit #7611 breaks compiler optimisation

5chufti, 2020-11-09:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:			[]
- Core Version:			[latest git]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dout]
- Flash Size: 		[1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

referenced commmit breaks use of any optimisation above -Os
at link time there are curious missing symbols ...

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```
void setup() {
  Serial.begin(74880); 
  pinMode(LED_BUILTIN, OUTPUT);
  while (!Serial) ; // wait for Arduino Serial Monitor
}

void loop() {
    digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN));
    delay(500);
}
```

### Debug Messages

```
Linking everything together...
"D:\\arduino\\portable\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\3.0.0-gnu12/bin/xtensa-lx106-elf-gcc" -CC -E -P -DVTABLES_IN_FLASH "D:\\arduino\\portable\\packages\\esp8266\\hardware\\esp8266\\3.0.0/tools/sdk/ld/eagle.app.v6.common.ld.h" -o "C:\\Users\\me\\AppData\\Local\\Temp\\arduino_build_74451/local.eagle.app.v6.common.ld"
"D:\\arduino\\portable\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\3.0.0-gnu12/bin/xtensa-lx106-elf-gcc" -fno-exceptions -Wl,-Map "-Wl,C:\\Users\\me\\AppData\\Local\\Temp\\arduino_build_74451/sketch_nov09a.ino.map" -O3 -g -Wall -Wextra -nostdlib -Wl,--no-check-sections -u app_entry -u _printf_float -u _scanf_float -Wl,-static "-LD:\\arduino\\portable\\packages\\esp8266\\hardware\\esp8266\\3.0.0/tools/sdk/lib" "-LD:\\arduino\\portable\\packages\\esp8266\\hardware\\esp8266\\3.0.0/tools/sdk/lib/NONOSDK22x_191122" "-LD:\\arduino\\portable\\packages\\esp8266\\hardware\\esp8266\\3.0.0/tools/sdk/ld" "-LD:\\arduino\\portable\\packages\\esp8266\\hardware\\esp8266\\3.0.0/tools/sdk/libc/xtensa-lx106-elf/lib" -Teagle.flash.1m.ld -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read -o "C:\\Users\\me\\AppData\\Local\\Temp\\arduino_build_74451/sketch_nov09a.ino.elf" -Wl,--start-group "C:\\Users\\me\\AppData\\Local\\Temp\\arduino_build_74451\\sketch\\sketch_nov09a.ino.cpp.o" "C:\\Users\\me\\AppData\\Local\\Temp\\arduino_build_74451\\core\\core.a" -lhal -lphy -lpp -lnet80211 -llwip2-536 -lwpa -lcrypto -lmain -lwps -lbearssl -lespnow -lsmartconfig -lairkiss -lwpa2 -lstdc++ -lm -lc -lgcc -Wl,--end-group "-LC:\\Users\\me\\AppData\\Local\\Temp\\arduino_build_74451"
d:/arduino/portable/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.0-gnu12/bin/../lib/gcc/xtensa-lx106-elf/10.1.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\me\AppData\Local\Temp\arduino_build_74451\core\core.a(Stream.cpp.o):(.text._ZN6Stream10readStringEv+0x4): undefined reference to `_ZN6String4initEv'
d:/arduino/portable/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.0-gnu12/bin/../lib/gcc/xtensa-lx106-elf/10.1.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\me\AppData\Local\Temp\arduino_build_74451\core\core.a(Stream.cpp.o):(.text._ZN6Stream10readStringEv+0x1b): undefined reference to `_ZN6String4initEv'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board Generic ESP8266 Module.
```

mcspr, 2020-11-09:
Does it build after removing `inline` keyword here?
https://github.com/esp8266/Arduino/blob/5d2563eee98ae116dc3d4b6bc14225efb3c85098/cores/esp8266/WString.cpp#L129

5chufti, 2020-11-09:
yes, removing the "inline" seems to fix the issue.

earlephilhower, 2020-11-20:
@5chufti can you please verify if #7553 (already merged) w/o any changes fixes your issue and report back?

5chufti, 2020-11-22:
Hi, issue seems resolved with current git version.
So closing issue ...

closed 2020-11-22 by 5chufti